lwfinger/rtl8188eu #275: Realtek FTP logins for various drivers + v5.3.9

kimocoder, 2019-01-29:
I've just probed through Realtek FTP's and found a decent new base,
The rtl8188eus has a fairly new release of v5.3.9_28540.20180627

Official changelog lays https://github.com/kimocoder/rtl8188eus/blob/v5.3.9/RTL8188EUS%20ReleaseNotes.pdf

FTP Server : ftp3.realtek.com
a)Account:RTL8192C Password: 3ubP7C (8188CUS/CTV/8192CU/8192CE)
b)Account:RTL8192D Password: a6dWF2 (8192DU)
c)Account:RTL8723 Password: F4v9Ws (8723AS/AS-VAU)
d)Account:RTL8188E Password: M6b3Er (8188EUS/ETV/8189ES)
e)Account:RTL8812A Password: J2e8Fg (8811AU/8812AU)
f)Account:RTL8192E Password: N5c6Sx (8192EU)
g)Account: RTL8723B Password: G3q6Wp (RTL8723B)

Hope some may be of interest? Anyhow, keep up your amazing work!

closed 2019-12-21 by kimocoder

epman100, 2020-04-06:
as of April 2020, it appears many of these account logins & passwords to the Realtek FTP server have stopped working and generate 530 authentication failed error messages.

CGarces, 2020-04-16:
Ping @kimocoder 
How you get the version v5.7.0_34085.20200313 for rtl8812au?

epman100, 2021-07-03:
a follow-up, it appears the ftp3.realtek.com site is recently dead and is no longer in service

lwfinger, 2021-07-03:
That site has been dead for quite a while.