lwfinger/rtl8188eu #278: Packet injection and monitor mode

Thalhammer, 2019-02-24:
Maybe I'm just too dumb, but whatever I do monitor mode does not work.
On my notebook I got monitor to work one time, but injection did not work and on a raspberry pi none of those work.
Is this just a driver problem or is the firmware unable to inject arbitrary packages and monitor packages ?
Or is this just a setup problem.

kimocoder, 2019-05-24:
@Thalhammer this driver doesn't support frame injection.
However, sorry for spamming @lwfinger -- but my driver does, found [here](https://github.com/aircrack-ng/rtl8188eus)

lwfinger, 2019-05-24:
Thanks for the update. Now all the hackers can leave me alone!!

closed 2019-05-24 by lwfinger

kimocoder, 2019-05-24:
At least a whitehat one. I admire your work, Larry. both you and Johannes!
Keep up the amazing work, and again.. sorry to interrupt. Love :1st_place_medal: