lwfinger/rtl8188eu #280: System freeze in hotspot mode with Qualcoom Atheros QCA9377

Quiwy, 2019-03-28:
Hi,
I'm using the rtl8188eu driver as the provider of the EMB-4500 motherboard changed recently the wifi chipset.
I have no problem when using the driver except in one case where the whole system freezes.

The only information I have on serial port is:
` mmc0: Timeout waiting for hardware interrupt.
INFO: rcu_preempt_state detected stall on CPU 0 (t=6000 jiffies)` 

and I can do nothing else than restarting the motherboard.

I'm using the 3.0.35 kernel and have no other choice than using it.

The problem occurs when connecting with a Qualcoom Atheros QCA9377 Wireless Network Adapter. I've tried with other computers (with intel chipset) or with phone and I have no problem.

If you have any info of how to fix this issue or need more log I would be happy to provide it.

lwfinger, 2019-03-28:
The missing interrupt is strange because the USB sussystem handles all interripts.

What wifi chip was used before they went to the 8188eu?

Which branch of this repo are you using? Have any of the configuration variables been changed?

What kernel is being used on the box having the QCA9377. It seems likely that it is using some option that was not present in 3.0.35. Of course, the 8188eu should return an error, not freeze.

Please provide any information being written to the dmesg log. If needed, would it be possible for you to capture the over-the-air packets from the exchange between the QCA9377 and the RTL8188EU?

Quiwy, 2019-03-29:
The wifi chip used before was a RTL8188CTV with  rtl8192cu driver.
And now it's a RTL8188ETV.

I'm using master branch of the repo, I can try with another one if you think it will fix the problem.
Not sure of which configuration variables you are talking about. We use same config file for hostpad.

```
#####` hostapd configuration file ##############################################

interface=wlan0
ssid=XXXXXXX
wpa_passphrase=XXXXXXX

ignore_broadcast_ssid=1

#wep_default_key=0
#wep_key0=XXXXXXX
#wep_key1="XXXXXXX"
#wep_key2=XXXXXXX
#wep_key3="XXXXXXX"

wpa=2

wpa_key_mgmt=WPA-PSK
wpa_pairwise=CCMP
wpa_group_rekey=3600

##### default configuration #######################################

driver=rtl871xdrv
ctrl_interface=/var/run/hostapd
eap_server=1
channel=6
beacon_int=100
hw_mode=g
ieee80211n=1
wme_enabled=1
ht_capab=[SHORT-GI-20][SHORT-GI-40][HT40+]
supported_rates=60 90 120 180 240 360 480 540
basic_rates=60 120 240
max_num_sta=8
ap_max_inactivity=300

logger_syslog=-1
logger_syslog_level=2
logger_stdout=-1
logger_stdout_level=2

dump_file=/var/tmp/hostapd.dump

wps_state=0

device_name=XXXX
manufacturer=XXXX
model_name=XXXX
model_number=XXXX
```

The box having the QCA9377 chip is a PC running with windows. 
Dmesg log before crash:
```
root@freescale ~$ dmesg
. 1581.05 BogoMIPS (lpj=7905280)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 512
CPU: Testing write buffer coherency: ok
hw perfevents: enabled with ARMv7 Cortex-A9 PMU driver, 7 counters available
CPU1: Booted secondary processor
CPU2: Booted secondary processor
CPU3: Booted secondary processor
[... 298 lines cut ...]
GobiSerial 2-1.6:1.1: GobiSerial converter detected
usb 2-1.6: GobiSerial converter now attached to ttyUSB0
GobiNet: 2018-06-22/SWI_2.52
GobiSerial 2-1.6:1.3: GobiSerial converter detected
usb 2-1.6: GobiSerial converter now attached to ttyUSB1
GobiSerial 2-1.6:1.2: GobiSerial converter detected
usb 2-1.6: GobiSerial converter now attached to ttyUSB2
usbcore: registered new interface driver GobiSerial
GobiSerial: 2018-06-22/SWI_2.33
GobiNet 2-1.6:1.0: eth1: register 'GobiNet' at usb-fsl-ehci.1-1.6, GobiNet Ethernet Device, 2a:a7:79:37:3c:35
USB Speed : USB 2.0
usbcore: registered new interface driver GobiNet
tun: Universal TUN/TAP device driver, 1.6
tun: (C) 1999-2004 Max Krasnyansky <maxk@qualcomm.com>
creating qcqmi0
Ethernet mode
flexcan imx6q-flexcan.0: writing ctrl=0x0329208d
flexcan imx6q-flexcan.1: writing ctrl=0x03292085
R8188EU: Firmware Version 11, SubVersion 1, Signature 0x88e1
MAC Address = 28:f3:66:f9:40:d8
R8188EU: INFO assoc success
eth0: Freescale FEC PHY driver [Generic PHY] (mii_bus:phy_addr=1:01, irq=-1)
```

Don't really know what you mean by "capture the over-the-air packets from the exchange between the QCA9377 and the RTL8188EU".  You can find a capture done with Wireshark, but I'm not really sure this is what you need. [capture.zip](https://github.com/lwfinger/rtl8188eu/files/3023257/capture.zip)

lwfinger, 2019-03-29:
I left that comment about packet capture intentionally vague. That way the abilities of my correspondent are effective screened. I have not yet looked at those packets.

I would like you to try branch v5.2.2.4. It builds with support for native cfg80211/nl80211. Your hostapd config file would need to use "driver=nl80211" rather than "driver=rtl871xdrv". Otherwise it should be the same.

Quiwy, 2019-04-01:
Can't compile with this branch:

```
drivers/net/wireless/rtl8188eu/os_dep/ioctl_cfg80211.c: In function 'rtw_cfg80211_indicate_sta_assoc':
drivers/net/wireless/rtl8188eu/os_dep/ioctl_cfg80211.c:3395:18: error: 'STATION_INFO_ASSOC_REQ_IES' undeclared (first use in this function)
drivers/net/wireless/rtl8188eu/os_dep/ioctl_cfg80211.c:3395:18: note: each undeclared identifier is reported only once for each function it appears in
drivers/net/wireless/rtl8188eu/os_dep/ioctl_cfg80211.c:3396:8: error: 'struct station_info' has no member named 'assoc_req_ies'
drivers/net/wireless/rtl8188eu/os_dep/ioctl_cfg80211.c:3397:8: error: 'struct station_info' has no member named 'assoc_req_ies_len'
scripts/Makefile.build:305: recipe for target 'drivers/net/wireless/rtl8188eu/os_dep/ioctl_cfg80211.o' failed
make[4]: *** [drivers/net/wireless/rtl8188eu/os_dep/ioctl_cfg80211.o] Error 1
```

lwfinger, 2019-04-01:
It seems that the v5.2.2.4 driver does not build with that old a kernel. The assoc_req_ies* members were added in 3.2. Unfortunately, it will not be easy to backport the changes.

I have not yet had a chance to look at your wireshark output yet.

Quiwy, 2019-04-02:
Ok thanks, will try to update kernel but not sure it will work with the motherboard. I'll come back to you when I have something new.