lwfinger/rtl8188eu #281: incorporate latest rtl8188eus v5.3.9 code

rpetrovski, 2019-04-15:
Hello, now that the recent version of rtl8188eus code from realtek is available [here](https://github.com/kimocoder/rtl8188eus/) is it possible to have it incorporated in linux kernel? What would be the problems with just copying it as is?

My vested interest is that current linux driver, which comes with kernel, is not able to connect in a stable manner. The current lwfinger master, although connects well, ends up with ridiculously low transfer rates. 100x less than I get with kernel one when it works. The 5.3.9 solves both issues but then it seems nobody is working on having it distributed as part of kernel.

My device is an outdoor Alfa dongle 0bda:8179, so I can't easily replace it with something else.

Cheers.
R.

lwfinger, 2019-04-15:
The problem with copying it "as is" is that it would never pass review. A "standard" Realtek driver requires from 3-6 months of work to get it close to that state. To convince you, merely look at all the code that depends on CONFIG_PLATFORM_WINDOWS. Do you really think that any Linux rewiewer would allow that code in the official kernel?

Have you tested the v5.2.2.4 branch of the repo?
 
I am willing to add the new driver as a separate branch of this repo, but unless someone comes up with serious cash, I would not consider modifying it so as to get it into the kernel.

rpetrovski, 2019-04-16:
I will hopefully have an opportunity to test v5.2.2.4 over the weekends. Hopefully it works.

If it does not:

I'm not familiar with the general rules the reviewers expect the code to satisfy, so bear with me please. If we remove all the code that is not relevant to hardware functioning under linux kernel environment with something like [coan](http://coan2.sourceforge.net/coan_man_1.html), what would be the next major issue?

I am assuming not everybody will like to have their devices to be handled by the updated driver without extensive testing. Is there an possibility to have the realtek 5.3.9 code initially handle just my dongle and then let the other hardware switch to it if/as needed?

rpetrovski, 2019-04-21:
v5.2.2.4 seems to work great at the moment. Thank you. 

it gives me 
RTW: ERROR rtw_check_bcn_info: cur beacon key
every few minutes in dmesg but without any apparent adverse effects. The connectivity and transfer rates seem to be excellent.

What will it take for 5.2.2.4 to get into kernel?

Roman.

lwfinger, 2019-04-21:
I could not see where that message is coming from. Once it is found, it can likely be eliminated as it is not an error.

I did some editing of this source with the goal of using it to replace r8188eu. It now passes most of the Sparse tests, but there are a lot of \errors from scripts/checkpatch.pl in your kernel source directory that would need to be fixed. One of the easy ones is to replace the license/copyright block at the beginning of each file. It now looks like

```
/******************************************************************************`
 *
 * Copyright(c) 2007 - 2011 Realtek Corporation. All rights reserved.
 *
 * This program is free software; you can redistribute it and/or modify it
 * under the terms of version 2 of the GNU General Public License as
 * published by the Free Software Foundation.
 *
 * This program is distributed in the hope that it will be useful, but WITHOUT
 * ANY WARRANTY; without even the implied warranty of MERCHANTABILITY or
 * FITNESS FOR A PARTICULAR PURPOSE. See the GNU General Public License for
 * more details.
 *
 * You should have received a copy of the GNU General Public License along with
 * this program; if not, write to the Free Software Foundation, Inc.,
 * 51 Franklin Street, Fifth Floor, Boston, MA 02110, USA
 *
 *
 ******************************************************************************/
```
For .c files, that should be replaced by

```
// SPDX-License-Identifier: GPL-2.0
/* Copyright(c) 2007 - 2011 Realtek Corporation. All rights reserved. */

```

For .h files, use

```
/* SPDX-License-Identifier: GPL-2.0 */
/* Copyright(c) 2007 - 2011 Realtek Corporation. All rights reserved. */

```
No, I do not know why the two forms are different.

All of this stuff needs to be cleaned up before the code can go into the kernel.

rpetrovski, 2019-05-10:
Ok, after some thinking, I'm dropping the issue. Will shop for non-usb dongle like Ubiquiti Bullet or some such. Cheers.

closed 2019-05-10 by rpetrovski