lwfinger/rtl8188eu #287: Error for wireless request "Set Mode" (8B06) :     SET failed on device wla0 ; No such device.

snowbreze, 2019-07-04:
Please help me with this problem.

iwconfig
lo        no wireless extensions.

wlan1     IEEE 802.11  ESSID:"Luis"  
          Mode:Managed  Frequency:2.457 GHz  Access Point: 7C:8B:CA:46:87:AD   
          Bit Rate=1 Mb/s   Tx-Power=20 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:on
          Link Quality=32/70  Signal level=-78 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:120  Invalid misc:25   Missed beacon:0

eth0      no wireless extensions.

wlan0     IEEE 802.11bgn  ESSID:"Luis"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:2.457 GHz  Access Point: 7C:8B:CA:46:87:AD   
          Bit Rate:72.2 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:****-****-****-****-****-****-****-****   Security mode:open
          Power Management:off
          Link Quality=51/100  Signal level=-56 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

root@kaliPC:~# ifconfig wlan0 down
root@kaliPC:~# iwconfig wlan0 mode monitor
Error for wireless request "Set Mode" (8B06) :
    SET failed on device wlan0 ; Invalid argument.
root@kaliPC:~# airmon-ng check kill

Killing these processes:

  PID Name
 6543 wpa_supplicant

root@kaliPC:~# iwconfig wlan0 mode monitor
Error for wireless request "Set Mode" (8B06) :
    SET failed on device wlan0 ; Invalid argument.
root@kaliPC:~# airmon-ng start wlan0

PHY	Interface	Driver		Chipset

null	wlan0		r8188eu		TP-Link TL-WN722N v2
phy0	wlan1		ath10k_pci	Qualcomm Atheros QCA9377 802.11ac Wireless Network Adapter (rev 31)

root@kaliPC:~# iwconfig
lo        no wireless extensions.

wlan1     IEEE 802.11  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=20 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:on
          
eth0      no wireless extensions.

wlan0     unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.457 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=0/100  Signal level=0 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

root@kaliPC:~# ifconfig wlan0 down
root@kaliPC:~# iwconfig wlan0 mode monitor
Error for wireless request "Set Mode" (8B06) :
    SET failed on device wlan0 ; Invalid argument.
root@kaliPC:~# airmon-ng check kill

Killing these processes:

  PID Name
 6543 wpa_supplicant

root@kaliPC:~# iwconfig wlan0 mode monitor
Error for wireless request "Set Mode" (8B06) :
    SET failed on device wlan0 ; Invalid argument.
root@kaliPC:~# airmon-ng start wlan0

PHY	Interface	Driver		Chipset

null	wlan0		r8188eu		TP-Link TL-WN722N v2
phy0	wlan1		ath10k_pci	Qualcomm Atheros QCA9377 802.11ac Wireless Network Adapter (rev 31)

root@kaliPC:~# iwconfig
lo        no wireless extensions.

wlan1     IEEE 802.11  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=20 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:on
          
eth0      no wireless extensions.

wlan0     unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.457 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=0/100  Signal level=0 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

lwfinger, 2019-07-05:
The driver does not support monitor mode. It was never coded by Realtek.

Your system also does not have device wla0. It is wlan0.

It is possible that branch v5.2.2.4 will support monitor mode.

snowbreze, 2019-07-05:
So what can i do to use my adaptor in monitor Mode, i need to bought a New one?

lwfinger, 2019-07-05:
Try the other branch. If that does not work, you will either need to code in monitor mode, hire someone to do it, or buy a new adapter.

closed 2019-07-05 by lwfinger

SHARMAG-cyber, 2020-09-04:
[root@kali]─[/home/carnage]
└──╼ #sudo iwconfig
lo        no wireless extensions.

eth0      no wireless extensions.

wlan0     unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.442 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

┌─[root@kali]─[/home/carnage]
└──╼ #ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.150.128  netmask 255.255.255.0  broadcast 192.168.150.255
        inet6 fe80::20c:29ff:fe13:82e1  prefixlen 64  scopeid 0x20<link>
        ether 00:0c:29:13:82:e1  txqueuelen 1000  (Ethernet)
        RX packets 139  bytes 9502 (9.2 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 37  bytes 2914 (2.8 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 19  base 0x2000  

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 18  bytes 790 (790.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 18  bytes 790 (790.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether d0:37:45:bd:dd:df  txqueuelen 1000  (Ethernet)
        RX packets 49  bytes 8413 (8.2 KiB)
        RX errors 0  dropped 12  overruns 0  frame 0
        TX packets 52  bytes 6696 (6.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

┌─[root@kali]─[/home/carnage]
└──╼ #sudo ifconfig wlan0 down                                                                                                       
┌─[root@kali]─[/home/carnage]
└──╼ #sudo iwconfig wlan0 mode monitor                                                                                               
Error for wireless request "Set Mode" (8B06) :
    SET failed on device wlan0 ; Invalid argument.

lwfinger, 2020-09-04:
It has been stated at least 100 times - thr driver does not support monitor mode!!!!!!

bmj970, 2021-10-03:
when you set up your adapter to monitor mode don't use 
# ifconfig wlan0 down
only follow given commands 
# airmon-ng check kill
your machine will lose internet connection
then, 
# iwconfig wlan0 mode monitor
it's done

dardoschuster, 2024-03-29:
Same problem with tp-link AC600 Archer T2U Plus where can i find the driver to put it in monitor mode ???

lwfinger, 2024-03-29:
The driver rtl8xxxu in modern mainline kernels should do monitor mode, otherwise, I have no suggerstions for you. Moniror mode is not high on the developer's priorities.