lwfinger/rtl8188eu #298: does it suport TP-LINK TL-WN722N v2 driver?

brunoCreator, 2019-10-26:
I have tried https://github.com/kimocoder/rtl8188eus I can get in to monitor mode but doesn't work with a reaver, this error appears [X] ERROR: pcap_activate status -1. https://github.com/t6x/reaver-wps-fork-t6x/issues/265 this issue report sad to try https://github.com/lwfinger/rtl8188eu, doesn't work, when I try ifconfig wlan0 down and iwconfig wlan0 mode monitor appear Killed on the terminal and the terminal stop working.

Sqxxe, 2020-02-11:
I'm here trying to find the same info, got it into monitor mode yet airgeddon nor reaver are working, have you found anything? I may assume that the tl-wn722n v2 is just not monitor mode supportive.

lwfinger, 2020-02-11:
It is probably the driver. If you are using the master branch, that driver is brain dead.

closed 2020-02-11 by lwfinger