lwfinger/rtl8188eu #299: OpenWrt Kernel 4.14.151

TheM4rc0, 2019-11-03:
Hello)
The driver is not loaded, where can the error be?

[rtl8188eufw.zip](https://github.com/lwfinger/rtl8188eu/files/3801100/rtl8188eufw.zip)

lsusb
`Bus 001 Device 002: ID 2357:010c TP-Link TL-WN722N v2`

**cat /sys/kernel/debug/usb/devices**
```
T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 4.14
S:  Manufacturer=Linux 4.14.151 ehci_hcd
S:  Product=EHCI Host Controller
S:  SerialNumber=1b000000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=2357 ProdID=010c Rev= 0.00
S:  Manufacturer=Realtek
S:  Product=802.11n NIC
S:  SerialNumber=00E04C0001
C:* #Ifs= 1 Cfg#= 1 Atr=a0 MxPwr=500mA
I:* If#= 0 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E:  Ad=81(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=03(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms

```

lwfinger, 2019-11-03:
You put the driver in the wrong directory. Use 'sudo make install' the way READ.md says!!!

closed 2019-11-03 by lwfinger

TheM4rc0, 2019-11-03:
> You put the driver in the wrong directory. Use 'sudo make install' the way READ.md says!!!

Hello I can not do make install on the Openwrt system.

@echo "blacklist r8188eu"> /etc/modprobe.d/50-8188eu.conf <--- This folder does not exist either, there is only /etc/modules.d! Where should I create the blacklist config?

The driver comes in there! there is no folder /kernel/drivers/staging/rtl8188eu/r8188eu.ko
```
root@OpenWrt:~# ls -a /lib/modules/4.14.151
.                          hwmon.ko                   nf_conntrack_ipv4.ko       nf_reject_ipv4.ko          xt_REDIRECT.ko
..                         ip6_tables.ko              nf_conntrack_ipv6.ko       nf_reject_ipv6.ko          xt_TCPMSS.ko
ath.ko                     ip6t_REJECT.ko             nf_conntrack_rtcache.ko    nls_base.ko                xt_comment.ko
ath10k_core.ko             ip6table_filter.ko         nf_defrag_ipv4.ko          ppp_async.ko               xt_conntrack.ko
ath10k_pci.ko              ip6table_mangle.ko         nf_defrag_ipv6.ko          ppp_generic.ko             xt_limit.ko
ath9k.ko                   ip_tables.ko               nf_flow_table.ko           pppoe.ko                   xt_mac.ko
ath9k_common.ko            ipt_MASQUERADE.ko          nf_flow_table_hw.ko        pppox.ko                   xt_mark.ko
ath9k_hw.ko                ipt_REJECT.ko              nf_log_common.ko           slhc.ko                    xt_multiport.ko
cfg80211.ko                iptable_filter.ko          nf_log_ipv4.ko             usb-common.ko              xt_nat.ko
compat.ko                  iptable_mangle.ko          nf_log_ipv6.ko             usbcore.ko                 xt_state.ko
crc-ccitt.ko               iptable_nat.ko             nf_nat.ko                  x_tables.ko                xt_tcpudp.ko
ehci-hcd.ko                ledtrig-usbport.ko         nf_nat_ipv4.ko             xt_CT.ko                   xt_time.ko
ehci-platform.ko           mac80211.ko                nf_nat_masquerade_ipv4.ko  xt_FLOWOFFLOAD.ko
gpio-button-hotplug.ko     nf_conntrack.ko            nf_nat_redirect.ko         xt_LOG.ko
```

TheM4rc0, 2019-11-04:
root@OpenWrt:~# modinfo r8188eu
module:         /lib/modules/4.9.184/r8188eu.ko
license:        GPL
depends:        usbcore

lwfinger, 2019-11-04:
I know nothing of openWRT. You are on your own.

TheM4rc0, 2019-11-04:
> I know nothing of openWRT. You are on your own.

https://github.com/lwfinger/rtl8188eu/issues/266#issuecomment-443289701

https://github.com/lwfinger/rtl8188eu/issues/266

He also had the same problem, but I'm not very good at it. Add with depends in the kernel!