lwfinger/rtl8188eu #300: Kernel panic (5.4.2)

lpapp, 2019-12-06:
It seems that the current code causes kernel panic when plugging the device in. Would you be able to advise me how to proceed from here?
[image]
[image]
[image]

lwfinger, 2019-12-06:
DO NOT EVER SEND ME PRIVATE E-MAIL UNLESS I ASK FOR IT.

I only learned of this crash this morning even though the fundamental cause has been present since kernel 5.3. For the most part, I check the drivers to see if they build, not that they run.

Your crash pictures are better than nothing, but they do not show me the cause. I had to find my 8188eu device and look at the output of dmesg before the kernel had a panic. That showed me the fix, which has been pushed to the repo. The driver now works for me.

closed 2019-12-06 by lwfinger

lpapp, 2019-12-06:
Thanks! I was going to make a video for you. I could not capture anything in the rolling kernel panic screen. All the important information had rolled out the latest screen. And I could not scroll back.

lpapp, 2019-12-06:
By the way, are you planning to port all these changes in this repo back to the mainline kernel? Then, we would not have to go to external places like this and use dkms, etc.

lwfinger, 2019-12-06:
Absolutely not. To get the driver into a shape that it would be accepted into the kernel would take almost a year of full-time work, and no one has offered to support such an undertaking!

lpapp, 2019-12-06:
Oh, I see. That is quite a lot of work for just one driver out of the many, many. :)

Thank you for maintaining this at least!