lwfinger/rtl8188eu #302: Fail compile in Centos 8, Kernel 4.18.0-80.11.2.el8_0.x86_64

joaomfrebelo, 2019-12-21:
I have the kernel source installed and i had the the last git sourced, the errror:

/root/wi-fi/rtl8188eu/os_dep/ioctl_linux.c:7814:3: error: ‘struct iw_handler_def’ has no member named ‘private’
  .private = rtw_private_handler,
   ^~~~~~~
/root/wi-fi/rtl8188eu/os_dep/ioctl_linux.c:7814:13: error: initialization of ‘struct iw_statistics * (*)(struct net_device *)’ from incompatible pointer type ‘int (**)(struct net_device *, struct iw_request_info *, union iwreq_data *, char *)’ [-Werror=incompatible-pointer-types]
  .private = rtw_private_handler,
             ^~~~~~~~~~~~~~~~~~~
/root/wi-fi/rtl8188eu/os_dep/ioctl_linux.c:7814:13: note: (near initialization for ‘rtw_handlers_def.get_wireless_stats’)
/root/wi-fi/rtl8188eu/os_dep/ioctl_linux.c:7815:3: error: ‘struct iw_handler_def’ has no member named ‘private_args’
  .private_args = (struct iw_priv_args *)rtw_private_args,
   ^~~~~~~~~~~~
/root/wi-fi/rtl8188eu/os_dep/ioctl_linux.c:7815:18: warning: excess elements in struct initializer
  .private_args = (struct iw_priv_args *)rtw_private_args,
                  ^
/root/wi-fi/rtl8188eu/os_dep/ioctl_linux.c:7815:18: note: (near initialization for ‘rtw_handlers_def’)
/root/wi-fi/rtl8188eu/os_dep/ioctl_linux.c:7816:3: error: ‘struct iw_handler_def’ has no member named ‘num_private’
  .num_private = sizeof(rtw_private_handler) / sizeof(iw_handler),
   ^~~~~~~~~~~
/root/wi-fi/rtl8188eu/os_dep/ioctl_linux.c:7816:17: warning: excess elements in struct initializer
  .num_private = sizeof(rtw_private_handler) / sizeof(iw_handler),
                 ^~~~~~
/root/wi-fi/rtl8188eu/os_dep/ioctl_linux.c:7816:17: note: (near initialization for ‘rtw_handlers_def’)
/root/wi-fi/rtl8188eu/os_dep/ioctl_linux.c:7817:3: error: ‘struct iw_handler_def’ has no member named ‘num_private_args’
  .num_private_args = sizeof(rtw_private_args) / sizeof(struct iw_priv_args),
   ^~~~~~~~~~~~~~~~
/root/wi-fi/rtl8188eu/os_dep/ioctl_linux.c:7817:22: warning: excess elements in struct initializer
  .num_private_args = sizeof(rtw_private_args) / sizeof(struct iw_priv_args),
                      ^~~~~~
/root/wi-fi/rtl8188eu/os_dep/ioctl_linux.c:7817:22: note: (near initialization for ‘rtw_handlers_def’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:316: /root/wi-fi/rtl8188eu/os_dep/ioctl_linux.o] Error 1
make[1]: *** [Makefile:1528: _module_/root/wi-fi/rtl8188eu] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.18.0-80.11.2.el8_0.x86_64'
make: *** [Makefile:155: modules] Error 2

lwfinger, 2019-12-22:
I do not have a 4.18.0 kernel; however both the master and v5.2.2.4 build with kernels 4.12 and 5.5. In other words, I cannot duplicate your problem, thus I have no idea what is causing it. Sorry, but I am not able to help.

Giffen-good, 2019-12-31:
I'm currently observing the same error when I attempt to make from latest master branch

lwfinger, 2019-12-31:
I reiterate that the master branch builds here. What likely has happened is that your distro has backported some kernel API changes. When that happens, a test of the macro LINUX_VERSION_CODE fails to give the correct API the way it does with an unaltered kernel. Ever since Ubuntu started this behavior, I have refused to support it, thus you will need to determine the exact place in the source where it happens and diddle with the codes that say things like
`#if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)`
Because your distro has broken the ability to detect that this particular change happens with kernel v4.15, you will need to try changing the 15 to 14, 13, 12, 11,... As nearly every distro has begun this madness, I refuse to support any of these asinine backports.

closed 2019-12-31 by lwfinger

hyuan-kamuda, 2020-07-13:
@lwfinger @joaomfrebelo @Giffen-good I search got this issue post , I met same issue , with CentOS 8(2004) which is with kernel 4.18.0-193     -  can anyone shield me some light how you solve this issue ?

lwfinger, 2020-07-13:
Read my previous comment. Your distro has screwed you and you will need to fix it yourself!

hyuan-kamuda, 2020-07-14:
> Read my previous comment. Your distro has screwed you and you will need to fix it yourself!

Yes, I know it is Redhat/centos changed kernel code result in compiling fail, could you please tell me more specific(maybe with an example in your code) how to change the macro definition so can pass compiling? Thanks lot.

hyuan-kamuda, 2020-07-14:
> Read my previous comment. Your distro has screwed you and you will need to fix it yourself!

Ps, I checked there's already /lib/firmware/rtlwifi/rtl8188eufw.bin after installing CentOS 8(2004), I can see rtl8188eufw.bin under /lib/firmware/rtlwifi.   Should I still pull code from here to compile/build/install RTL8188EUFW again? 

With CentOS8, by running lsmod, it has no rtl8188eu module loaded at all.
I tried lsusb, system can find "Realtek Semiconductor Corp.RTL8188EUS 802.11n Wireless Network Adaptor".
I tried to use tail -f /var/log/messages to see what messages have when I plug out /in TL-WN725N USB wifi dongle, I can see :

USB disconnect, device number 4
new high-speed USB device number 5 using ehci-pci
new USB device found, idVendor=0bda, idProduct=8179, bcdDevice= 0.00
New USB device strings: Mfr=1, Product=2,SerialNumber=3
Product: 802.11n NIC
Manuafacturer: Realtek
SerialNumber: 00E04C0001

using ifconfig -a to see if system can recognize wlan adaptor, it has nothing about wlan in command output.

Then I tried using insmod /lib/firmware/rtlwifi/rtl8188eufw.bin to manually load module, it shows error of:
insmod: ERROR: could not insert module /lib/firmware/rtlwifi/rtl8188eufw.bin: Invalid module format

Could you kindly point to me if CentOS8(2004) has already supported RTL8188EU wifi adapter driver or not? If yes, why system can't recognize the hardware and load module automatically? Shall I manually build/compile/install it from: https://github.com/lwfinger/rtl8188eu ?

Thanks!

hyuan-kamuda, 2020-07-14:
> I reiterate that the master branch builds here. What likely has happened is that your distro has backported some kernel API changes. When that happens, a test of the macro LINUX_VERSION_CODE fails [...]

in your code files, there're below places which use #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)`, by your suggestions , should I change these places（all of them) with "#if LINUX_VERSION_CODE >= KERNEL_VERSION(4, **14** 0)“  to see if compiling is ok or not, if not, change all of them to "#if LINUX_VERSION_CODE >= KERNEL_VERSION(4, **13**, 0)`....... until a right "#if LINUX_VERSION_CODE >= KERNEL_VERSION(4, **n**, 0)” tested???

rtw_led.c (core) line 29 : #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)
BlinkTimerCallback in rtw_led.c (core) : #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)
InitLed871x in rtw_led.c (core) : #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)
rtw_led.h (include) line 182 : #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)
rtw_p2p.c (core) line 1747 : #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)
reset_ch_sitesurvey_timer_process in rtw_p2p.c (core) : #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)
rtw_p2p.c (core) line 1769 : #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)
reset_ch_sitesurvey_timer_process2 in rtw_p2p.c (core) : #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)
rtw_p2p.c (core) line 1791 : #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)
restore_p2p_state_timer_process in rtw_p2p.c (core) : #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)
rtw_p2p.c (core) line 1810 : #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)
pre_tx_scan_timer_process in rtw_p2p.c (core) : #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)
rtw_p2p.c (core) line 1848 : #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)
find_phase_timer_process in rtw_p2p.c (core) : #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)
rtw_init_wifidirect_timers in rtw_p2p.c (core) : #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)
rtw_p2p_enable in rtw_p2p.c (core) : #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)
rtw_pwrctrl.c (core) line 211 : #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)
pwr_state_check_handler in rtw_pwrctrl.c (core) : #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)
rtw_init_pwrctrl_priv in rtw_pwrctrl.c (core) : #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)
rtw_recv.c (core) line 44 : #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)
_rtw_init_recv_priv in rtw_recv.c (core) : #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)
rtw_recv.c (core) line 2193 : #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)
rtw_signal_stat_timer_hdl in rtw_recv.c (core) : #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 15, 0)

lwfinger, 2020-07-14:
That driver has been in the kernel since version 3.12!! All you should need to do is "plug it in"! In the kernel, the module is r8188eu. I know it works - I am using it at the moment.

There is a fundamental difference between firmware and a driver. Firmware is the code that runs on the CPU of an embedded device such as a router, your refrigerator, or a wireless chip. A driver is the code that runs on your computer to drive that device. Frequently, a particular wireless device requires external firmware. In that case, the driver is responsible for loading the firmware onto the wireless device in the process of starting. With this driver, the firmware is in /lib/modules/rtlwifi/rtl8188eufw.bin. That file should be provided by your distro.

Reboot your system, plug in the device, and look at 'dmesg | less'. Near the end, you should find something like "Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_A_CUT_1T1R_RomVer(0)". If it is not working post the lines that follow.

For the moment, forget trying to build an external module. That step seems to be a stretch for your Linux skill set.

hyuan-kamuda, 2020-07-14:
well, thanks your help anyway!
The existing rtl8288eu driver latest one is for centos7 :
https://centos.pkgs.org/7/elrepo-x86_64/kmod-8188eu-5.2.2.4-2.20190907git.el7_8.elrepo.x86_64.rpm.html

tried to install it with centos8 , it is failed because kernel is not compatible, so have to debug your code and see how can make it work with CentOS customized kernel of 4.18.0-193 although i am not familiar with kernel driver development. Will learn.  Thanks

lwfinger, 2020-07-14:
Well, what do you mean? I gave you exactly what you should do to make it work with the driver in the kernel. Walk before you try to run.

hyuan-kamuda, 2020-07-14:
I mean, becuase there's no existing driver built for centos lk 4.18.0-193  --- all existing driver rpm will be failed to install since kernel version is not compatible, I have only two choices:

1. Request others to build for centos 8/lk 4.18.0.193, like go to ELrepo.org to ask this, and wait for.

2. Try to build it with 4.18.0-193 and fix the compiling issues reported here by myself.

hyuan-kamuda, 2020-07-15:
Hi 

I am writing down what I changed to allow your driver code can be built with CentOS 8.2(2004) , linux kernel version is 4.18.0-193 ( CentOS 8 customized kernel base on upstream 4.18.0 ), hoping can help later readers to reduce efforts on enabling it, you may want to help to review it as well:

#1, in this project's Makefile , adding below

_EXTRA_CFLAGS +=DCONFIG_WEXT_PRIV
EXTRA_CFLAGS +=DCONFIG_WIRELESS_EXT_

#2, in os_dep/os_intfs.c

static u16 rtw_select_queue(struct net_device *dev, struct sk_buff *skb
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 2, 0))
			    ,struct net_device *sb_dev
#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 19, 0))
			    ,struct net_device *sb_dev
                            ,select_queue_fallback_t fallback
#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 14, 0))
			    ,void *unused
                            ,select_queue_fallback_t fallback
#elif (LINUX_VERSION_CODE == KERNEL_VERSION(3, 13, 0))
			    , void *accel
#endif
)

change above #elif (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 19, 0))

_to #elif (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 18, 0))_

#3, in os_dep/rtw_android.c: 

int rtw_android_priv_cmd(struct net_device *net, struct ifreq *ifr, int cmd)
{
	int ret = 0;
	char *command = NULL;
	int cmd_num;
	int bytes_written = 0;
	struct android_wifi_priv_cmd priv_cmd;

	rtw_lock_suspend();
	if (!ifr->ifr_data) {
		ret = -EINVAL;
		goto exit;
	}
	if (copy_from_user(&priv_cmd, ifr->ifr_data,
			   sizeof(struct android_wifi_priv_cmd))) {
		ret = -EFAULT;
		goto exit;
	}
	command = kmalloc(priv_cmd.total_len, GFP_KERNEL);
	if (!command) {
		DBG_88E("%s: failed to allocate memory\n", __func__);
		ret = -ENOMEM;
		goto exit;
	}
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 0, 0))
	if (!access_ok(priv_cmd.buf, priv_cmd.total_len)) {
#else
	if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)) {
#endif

change above 

#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 0, 0))
	if (!access_ok(priv_cmd.buf, priv_cmd.total_len)) {
#else
	if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)) {
#endif

to 

_#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 18, 0))
	if (!access_ok(priv_cmd.buf, priv_cmd.total_len)) {
#else
	if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)) {
#endif_

lwfinger, 2020-07-16:
To use the driver in branch "master", you need to have CONFIG_WIRELESS_EXT=y in your kernel configuration.  That will supply those missing globals.

If your kernel generation had included CONFIG_STAGING=y, and CONFIG_R8188EU=m, then you could have "just plugged the device in". I was not aware that Centos used such brain-dead kernels.

When you publish changes needed to compile, they are a lot more readable when you use the patch form. Generate the file with 'git diff > diff.patch.

omid455, 2020-07-17:
> Read my previous comment. Your distro has screwed you and you will need to fix it yourself!

Thanks for being a great help!!!! I don't know who put you in charge of this, what an arrogant person you are! fix your code before making distros change theirs. Instead of helping people you are just bashing them with your selfish and arrogant remarks! "walk before you run?" well done! "I refuse to support?" who do you think you are? wow!

hyuan-kamuda, 2020-07-17:
> To use the driver in branch "master", you need to have CONFIG_WIRELESS_EXT=y in your kernel configuration. That will supply those missing globals.

see me kernel compile configuration by make menuconfig below screen snapshoot, still print :

[ 346.976147] 8188eu: Unknown symbol iwe_stream_add_event (err 0)
[ 346.976244] 8188eu: Unknown symbol wireless_send_event (err 0)
[ 346.976283] 8188eu: Unknown symbol iwe_stream_add_point (err 0)

when loading module, (either by insmod command or by plugin device). 

[image]
[image]

lwfinger, 2020-07-17:
Well you can just stumble around. You need me worse than I need you, but that bridge is thoroughly burned.

SidMorad, 2020-07-31:
@hyuan-kamuda if upgrading kernel is an option for you? [Upgrading to 5.x](https://computingforgeeks.com/how-to-install-latest-kernel-on-centos-linux) help me to bypass this issue.

enigma131, 2021-04-09:
I've solved the problem for Almalinux 8.3, / Centos 8.3. / Redhat 8.3  (latest updated kernel 4.18)
You need to take the branch v4.1.8_9499
Then doing the modifications : 
os_dep/rtw_android.c:#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 18, 0))  //  ex 5,0   @line 359
os_dep/os_intfs.c:#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 18, 0))      // ex 4,19  @line 419

And after it compiles and work.

@lwfinger : can you stick the solution because many people are searching about. This is a closed tread and except a web searching engine nobody comes here.

lwfinger, 2021-04-09:
I have made those changes and push them. Ordinarily, I would not make changes that are specific to certain distro's kernels. Doing so is a slippery slope and keeping up with all the people that backport API changes would be a full-time job. In this case, moving changes from 5.0 or 4.19 to 4.18 probably will not cause a lot of problems. I reserve the right to analyze each such request,

enigma131, 2021-04-10:
Thanks , cross fingers for other distros, RHEL 8 series with kernel 4.18 will be maintained until 2029 that's why some people like me are looking for alternative of CentOs that will be killed by RedHat by end 2021. AlmaLinux is the first alternative of replacement and is 100 % compatible and functional.
I have see an other modification for set_rf in osdep_service.c , for what issue is it made ?