lwfinger/rtl8188eu #303: Realtek 502.11N RTL8188etv

jacqpret, 2019-12-24:
Good Afternoon, purchased a Realtek 502.11N RTL8188etv USB Network Adapter,does the driver support monitor mode & frame injection to utilise in Kali.
Thank You

kimocoder, 2019-12-24:
My rtl8188eus is already in Kali, simply "apt-get install realtek-rtl8188eus-dkms"

Monitor mode & frame injection supported, also MESH and som other stuff.

jacqpret, 2019-12-25:
Thank you very much, will try it immediately

jacqpret, 2019-12-25:
apt-get install realtek-rtl8188eus-dkms
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  realtek-rtl8188eus-dkms
0 upgraded, 1 newly installed, 0 to remove and 59 not upgraded.
1 not fully installed or removed.
Need to get 1,491 kB of archives.
After this operation, 12.0 MB of additional disk space will be used.
Get:1 https://mirrors.ocf.berkeley.edu/kali kali-rolling/contrib amd64 realtek-rtl8188eus-dkms all 5.3.9~20191129-0kali1 [1,491 kB]
Fetched 1,491 kB in 7s (202 kB/s)                                              
Selecting previously unselected package realtek-rtl8188eus-dkms.
(Reading database ... 469799 files and directories currently installed.)
Preparing to unpack .../realtek-rtl8188eus-dkms_5.3.9~20191129-0kali1_all.deb ...
Unpacking realtek-rtl8188eus-dkms (5.3.9~20191129-0kali1) ...
Setting up ndiswrapper-dkms (1.57-1) ...
Removing old ndiswrapper-1.57 DKMS files...

------------------------------
Deleting module version: 1.57
completely from the DKMS tree.
------------------------------
Done.
Loading new ndiswrapper-1.57 DKMS files...
Building for 5.3.0-kali3-amd64
Building initial module for 5.3.0-kali3-amd64
Error! Bad return status for module build on kernel: 5.3.0-kali3-amd64 (x86_64)
Consult [/var/lib/dkms/ndiswrapper/1.57/build/make.log]for more information.
dpkg: error processing package ndiswrapper-dkms (--configure):
 installed ndiswrapper-dkms package post-installation script subprocess returned error exit status 10
Setting up realtek-rtl8188eus-dkms (5.3.9~20191129-0kali1) ...
Loading new realtek-rtl8188eus-5.3.9~20191129 DKMS files...
Building for 5.3.0-kali3-amd64
Building initial module for 5.3.0-kali3-amd64
Done.

8188eu.ko:
Running module version sanity check.
Error! Module version v5.3.9_28540.20180627 for 8188eu.ko
is not newer than what is already found in kernel 5.3.0-kali3-amd64 (v5.3.9_28540.20180627).
You may override by specifying --force.

depmod.......

DKMS: install completed.
Errors were encountered while processing:
 ndiswrapper-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

kimocoder, 2019-12-25:
ndiswrapper? why use ndiswrapper?

jacqpret, 2019-12-25:
Thank you very much Christian, tried this but could not install due to errors, could you please assist.

Regards

Jacques Pretorius

jacqpret, 2019-12-25:
Will uninstall and try again

kimocoder, 2019-12-25:
It's tested and working, by me, Debian & Kali.
Sorry for once hijacking your repo, @lwfinger finger

If DKMS doesn't work for you, whatever ndiswrapper is..

Do "updatedb" and "locate 8188eu.ko"
"rm -rf" delete those files ..

Create "/etc/modprobe d/blacklist.conf" with content "blacklist r8188eu" to Block Larry's kernel built driver.

And git clone the repo
"git clone https://github.com/aircrack-ng/rtl8188eus -b v5.3.9" make and make install

This issue may be closed.

jacqpret, 2019-12-25:
Sorry to sound stupid, how do I go about to do  this step

Create "/etc/modprobe d/blacklist.conf" with content "blacklist r8188eu" to Block Larry's kernel built driver.

jacqpret, 2019-12-25:
Tried to uninstall apt-get remove ndiswrapper. apt-get purge --auto-remove ndiswrapper, no luck.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ndiswrapper

root@kali   ~/Downloads/ndiswrapper-1.62  apt-get purge ndiswrapperndiswrapper-dkms:
  Installed: 1.57-1
  Candidate: 1.57-1
  Version table:
 *** 1.57-1 500
        500 http://old.kali.org/kali moto/main amd64 Packages
        100 /var/lib/dpkg/status

lwfinger, 2019-12-25:
ndiswrapper is a very old driver that wrapped around Windows drivers so that they could be used with Linux. As far as I know, it no longer works. Besides, it was quite likely to crash the kernel.

closed 2019-12-25 by lwfinger

jacqpret, 2019-12-25:
Managed to uninstall netwrapper and blacklist file.
when II run make I get the following output. Think I should make a clean install of Kali and try all over.

root@kali   ~/rtl8188eus     v5.2.2.4  make       720  04:24:58  
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.3.0-kali3-amd64/build M=/root/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-kali3-amd64'
  CC [M]  /root/rtl8188eus/core/rtw_cmd.o
  CC [M]  /root/rtl8188eus/core/rtw_security.o
  CC [M]  /root/rtl8188eus/core/rtw_debug.o
  CC [M]  /root/rtl8188eus/core/rtw_io.o
  CC [M]  /root/rtl8188eus/core/rtw_ioctl_query.o
  CC [M]  /root/rtl8188eus/core/rtw_ioctl_set.o
  CC [M]  /root/rtl8188eus/core/rtw_ieee80211.o
  CC [M]  /root/rtl8188eus/core/rtw_mlme.o
  CC [M]  /root/rtl8188eus/core/rtw_mlme_ext.o
/root/rtl8188eus/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/root/rtl8188eus/core/rtw_mlme_ext.c:1441:6: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1441 |   if (check_fwstate(pmlmepriv, WIFI_AP_STATE) == _TRUE)
      |      ^
/root/rtl8188eus/core/rtw_mlme_ext.c:1446:2: note: here
 1446 |  case WIFI_ASSOCREQ:
      |  ^~~~
  CC [M]  /root/rtl8188eus/core/rtw_mi.o
  CC [M]  /root/rtl8188eus/core/rtw_wlan_util.o
  CC [M]  /root/rtl8188eus/core/rtw_vht.o
  CC [M]  /root/rtl8188eus/core/rtw_pwrctrl.o
  CC [M]  /root/rtl8188eus/core/rtw_rf.o
  CC [M]  /root/rtl8188eus/core/rtw_recv.o
  CC [M]  /root/rtl8188eus/core/rtw_sta_mgt.o
  CC [M]  /root/rtl8188eus/core/rtw_ap.o
/root/rtl8188eus/core/rtw_ap.c: In function ‘rtw_add_bcn_ie’:
/root/rtl8188eus/core/rtw_ap.c:193:9: warning: ‘ielen’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  193 |   ielen += 2;
      |   ~~~~~~^~~~
  CC [M]  /root/rtl8188eus/core/rtw_xmit.o
  CC [M]  /root/rtl8188eus/core/rtw_p2p.o
  CC [M]  /root/rtl8188eus/core/rtw_tdls.o
  CC [M]  /root/rtl8188eus/core/rtw_br_ext.o
  CC [M]  /root/rtl8188eus/core/rtw_iol.o
  CC [M]  /root/rtl8188eus/core/rtw_sreset.o
  CC [M]  /root/rtl8188eus/core/rtw_btcoex_wifionly.o
  CC [M]  /root/rtl8188eus/core/rtw_btcoex.o
  CC [M]  /root/rtl8188eus/core/rtw_beamforming.o
  CC [M]  /root/rtl8188eus/core/rtw_odm.o
  CC [M]  /root/rtl8188eus/core/efuse/rtw_efuse.o
  CC [M]  /root/rtl8188eus/os_dep/osdep_service.o
  CC [M]  /root/rtl8188eus/os_dep/linux/os_intfs.o
/root/rtl8188eus/os_dep/linux/os_intfs.c:1240:22: error: initialization of ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *)’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *)’} from incompatible pointer type ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} [-Werror=incompatible-pointer-types]
 1240 |  .ndo_select_queue = rtw_select_queue,
      |                      ^~~~~~~~~~~~~~~~
/root/rtl8188eus/os_dep/linux/os_intfs.c:1240:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-5.3.0-kali3-common/scripts/Makefile.build:286: /root/rtl8188eus/os_dep/linux/os_intfs.o] Error 1
make[2]: *** [/usr/src/linux-headers-5.3.0-kali3-common/Makefile:1639: _module_/root/rtl8188eus] Error 2
make[1]: *** [/usr/src/linux-headers-5.3.0-kali3-common/Makefile:179: sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-kali3-amd64'
make: *** [Makefile:1911: modules] Error 2

kimocoder, 2019-12-26:
v5.2.2.4 isnt't my driver.. I'm out of this. Read some README's or something