lwfinger/rtl8188eu #306: Dkms error says I have to use git clone and not the zip, so it fails to compile, but I am already using git clone!

pitsi, 2020-01-13:
So, I am trying to make this thing work on debian stable 10 (kernel 4.19), via dkms
`Bus 007 Device 002: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter`

As described in the main page:
I cloned the repo and 
```
# git clone https://github.com/lwfinger/rtl8188eu
Cloning into 'rtl8188eu'...
remote: Enumerating objects: 16, done.
remote: Counting objects: 100% (16/16), done.
remote: Compressing objects: 100% (16/16), done.
remote: Total 12745 (delta 3), reused 2 (delta 0), pack-reused 12729
Receiving objects: 100% (12745/12745), 16.10 MiB | 831.00 KiB/s, done.
Resolving deltas: 100% (9669/9669), done.
```
I added it to dkms
```
# dkms add ./rtl8188eu

Creating symlink /var/lib/dkms/8188eu/1.0/source ->
                 /usr/src/8188eu-1.0

DKMS: add completed.
```
and then I asked dkms to build it, but it failed
```
# dkms build 8188eu/1.0

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=4.19.0-6-amd64...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.0-6-amd64 (x86_64)
Consult /var/lib/dkms/8188eu/1.0/build/make.log for more information.
```
And, according to the log, I had to use git clone, not the zip, which is exactly what I did!
```
# cat /var/lib/dkms/8188eu/1.0/build/make.log 
DKMS make.log for 8188eu-1.0 for kernel 4.19.0-6-amd64 (x86_64)
Mon 13 Jan 2020 02:06:10 PM EET
Directory .git does not exist indicating that you downloaded the source as a zip file. Only the 'git clone' method is now supported.
make: *** [Makefile:152: test] Error 1
```
Any help will be appreciated.

p.s. I want to avoid the make option to install it, because then I will have to do it again on every kernel update, that is why I prefered dkms in the fisrt place.

lwfinger, 2020-01-13:
I do not use dkms, and cannot provide any help. If you 'cd ~/rtl8188eu' and 'make', what happens?

pitsi, 2020-01-13:
It builds as it should, but I want to avoid that tactic.

lwfinger, 2020-01-13:
Then you need to fix your dkms.conf. I think it is pointing to the wrong directory.

closed 2020-01-13 by lwfinger

pitsi, 2020-01-28:
Due to a major health issue (pneumonia), I answer 15+ days later. Sorry for the delay.

Do you mean this dkms.conf file? This does not seem to mention the .git folder. Also, the same issue is described on #310  
https://github.com/lwfinger/rtl8188eu/blob/master/dkms.conf

I will give it one more try once I feel better, hopefully by the end of the week.

pieniacy, 2020-01-28:
I can confirm the issue, I have reproduced it, I have a fix (see #310).

I believe saying "your dkms.conf" is not right:
1. This file is in the repo so it should be maintained.
2. Readme says _The module can also be installed with DKMS._ which is not true at the moment, dkms way should be fixed or such note deleted.

lwfinger, 2020-01-28:
It was not my idea to put the dkms.conf file there. I did that to satisfy some other users; however, I stated at that time that I would not and could not maintain it.

I will remove the file and the readme comment!

Never forget that I do this as a service to the community. If there are many more demanding requests like yours, I will be inclined to delete the entire repository!

pitsi, 2020-01-31:
First of all, we all appreciate the work you do for the community.

Let me explain how things are from my side, so as to give you a better idea why I am really frustrated.
I have the above card which is supposed to work with kernel's r8188eu. The driver shows up in dmesg when I plug the card and on lsmod as well. It's hwid is even listed in the driver's modinfo
```
# modinfo r8188eu | grep 8179
alias:          usb:v07B8p8179d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8179d*dc*dsc*dp*ic*isc*ip*in*
```
But it does not work, as `iwlist scan` returns `wlan0 No scan results`, every single time. And, assuming that debian 10's 4.19 is "old", the same happens with the backported 5.4, so all I can assume is that the stock driver is problematic with this card. And this is how my quest for a better driver begins. 

I found your github repo on a mirror of wikidevi and I had tried 2 other github repos of similar drivers before yours. Both were built via dkms, but none of them worked. Yours does work, although I get the above message from iwlist scan sometimes. I do not use wicd, nm or any ther similar app.
Btw, if I use your driver, do I have to blacklist the kernel one in order to use yours only?

p.s. I am still in bad shape after the pneumonia, so my hopes for more checking by the end of this week have are now 0. I hope I will be better next week, but in the end, I may have to resort to windows to make the card work as it should. Please don't be offended. Thank you in advance