lwfinger/rtl8188eu #307: for the love of god

cyb3rc0lby, 2020-01-24:
how do i clone this? it's truly awful to force people into this terrible github shit.

rhempel, 2020-01-24:
Just put on your big boy pants and learn git like thousands of other programmers that use it and depend on it every day.

cyb3rc0lby, 2020-01-24:
ok my bad i had a typo in the url. 2020 and we still cant paste into an xterm?

lwfinger, 2020-01-24:
And I'm supposed to feel sorry for you. Just think about it - hundreds of people have read the instructions, copied the actual lines that have been given to them, and built the drivers. I guess you must be one in a thousand!

In an xterm, SHIFT-CTRL-C is like CTRL-C in a GUI, and SHIFT-CTRL-V is the same as CTRL-V in a GUI.

Shacklebolt13, 2020-04-09:
well.. git clone didn't work for me either, some -3, -5 code errors while inflating..
i just altered the sh file to not quit after showing MSG. lol :)

closed 2020-04-10 by lwfinger