lwfinger/rtl8188eu #310: Installation via dkms is broken

pieniacy, 2020-01-25:
Because of the check for presence of `.git` the dkms way now does not work, reproduce with:
```
git clone https://github.com/lwfinger/rtl8188eu
sudo dkms add ./rtl8188eu
sudo dkms build 8188eu/1.0
```
Then it fails, because dkms does not copy `.git` directory to the build directory. After manually copying `.git` I can successfully build with `sudo dkms build 8188eu/1.0`.

pitsi, 2020-01-28:
May I ask how did you "manually copy .git"?

pieniacy, 2020-01-28:
Sure, here you go:
tldr: `sudo cp -r ./rtl8188eu/.git /usr/src/8188eu-1.0/`

More info: (notice the location when it says "Creating symlink")
```
sr@sr:~/demo $ sudo dkms add ./rtl8188eu

Creating symlink /var/lib/dkms/8188eu/1.0/source ->
                 /usr/src/8188eu-1.0

DKMS: add completed.
sr@sr:~/demo $ sudo dkms build 8188eu/1.0

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
'make' all KVER=4.4.0-166-generic....(bad exit status: 2)
ERROR (dkms apport): binary package for 8188eu: 1.0 not found
Error! Bad return status for module build on kernel: 4.4.0-166-generic (x86_64)
Consult /var/lib/dkms/8188eu/1.0/build/make.log for more information.
sr@sr:~/demo $ cat /var/lib/dkms/8188eu/1.0/build/make.log
DKMS make.log for 8188eu-1.0 for kernel 4.4.0-166-generic (x86_64)
Tue Jan 28 07:22:24 EST 2020
Directory .git does not exist indicating that you downloaded the source as a zip file. Only the 'git clone' method is now supported.
Makefile:152: recipe for target 'test' failed
make: *** [test] Error 1
sr@sr:~/demo $ sudo cp -r ./rtl8188eu/.git /usr/src/8188eu-1.0/
sr@sr:~/demo $ sudo dkms build 8188eu/1.0

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
'make' all KVER=4.4.0-166-generic............
cleaning build area....

DKMS: build completed.
```

I don't know the reason, one possible explanation is dkms is issuing something equivalent to
`cp -r ./rtl8188eu/* /usr/src/8188eu-1.0/`
but such copy does not copy dot files ("hidden" files) and hence `.git` is not copied properly. Also, it might be a feature: dkms might not want to have `.git` directory in its source directories.

Either way, as of now repo's readme instructions are wrong (but fix is not difficult).

fgimenezm, 2020-02-01:
It's not DKMS who complains about .git missing, it's the makefile:
https://github.com/lwfinger/rtl8188eu/blob/master/Makefile#L152

That was added here: https://github.com/lwfinger/rtl8188eu/commit/4930848febecd66a1c4b0737954a3c98984e9ea6

DKMS file had this line:
```
MAKE="'make' KVER=${kernelver} all"
```
changing it to this should circumvent the issue:
```
MAKE="'make' KVER=${kernelver} modules"
```

I guess anywone willing to maintain and support dkms can open a PR to add it back, or create and maintain a dkms-enabled fork.

lwfinger, 2020-02-01:
Please do the fork so that I can get rid of this PITA.

closed 2020-02-01 by lwfinger