lwfinger/rtl8188eu #313: Enable SW encryption

ZupoLlask, 2020-02-09:
Hi!

I'm using Linux Kernel 5.4.18 loaded with rtl8188eu module v4.1.4_6773.20130222, using an aarch64 (arm64) Debian 10 based image.

As I'm experiencing very slow download/upload speeds to an AP less than 1 meter away from my dongle, I would like trying to disable HW encryption/enable SW encryption to check the impact of both options.

Problem is, from the code it seems there's no way to do that without recompiling the rtl8188eu module. Am I missing something? How can I set something like "swenc" parameter or rtw_software_encrypt/rtw_software_decrypt variables, which seem to be the equivalent to "swenc"?

Thanks.

lwfinger, 2020-02-09:
It is unlikely that turning off hardware encryption will help performance. In fact, if you have a moderately slow CPU, it will probably hurt. That said, you will need to rewrite the driver to add that functionality.

I would suggest that you use branch v5.2.2.4. That one is the most likely to have the best performance.

ZupoLlask, 2020-02-09:
Hi Larry!

Thank you for the quick feedback. 

For this matter, do you think that a quad-core ARM Cortex-A53 is slow to the level of capping speed transfer to 3 Mbps (not 3MB/s) up/down?

I'll try that version and I'll report my findings later.

lwfinger, 2020-02-09:
Encryption and packet handling and not multi-threaded, thus the speed of an individual CPU is the important parameter; however, yours should be adequate. The device is N150, thus the maximum total throughput with a single connection will be no better that 75 Mbps. Using my device and v5.2.2.4, I get 43 Mbps download and 40 Mbps upload. Those values are measured against a local copy of speedtest. Using Speedtest.net, I get 14 down and 6 up. The latter number is the speed of my broadband line.

ZupoLlask, 2020-02-11:
Larry,

After compiling and installing, modinfo returns v5.3.9_28540.20180627, not v5.2.2.4, although I'm quite sure I checked out the right version.

Anyway, I'm getting 45Mbps using speedtest.net, the same speed I'm getting using ethernet for this SBC.

So, the question is: why there's no newer version of the driver mainlined? This version has nothing to do with the current mainlined version! 

Great work!

lwfinger, 2020-02-11:
The reason it is not mainlined is that I have no interest in spending the next 6 months or a year converting the Realtek crap code into the form needed to be accepted into the kernel. It takes enough of my time just answering this kind of requeat/comment.

closed 2020-02-11 by lwfinger

ZupoLlask, 2020-02-11:
Hmm... I kind of understand that. 

Again, thank you for your help! 🙂