lwfinger/rtl8188eu #314: can't "make all"

Leonious-Shuvo, 2020-02-18:
$ make all
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.3.0-40-generic/build M=/home/leo/rtl8188eu  modules
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-40-generic'
arch/x86/Makefile:147: CONFIG_X86_X32 enabled but no binutils support
  CC [M]  /home/leo/rtl8188eu/core/rtw_ap.o
/home/leo/rtl8188eu/core/rtw_ap.c:1974:1: fatal error: opening dependency file /home/leo/rtl8188eu/core/.rtw_ap.o.d: Permission denied
 }
 ^
compilation terminated.
scripts/Makefile.build:288: recipe for target '/home/leo/rtl8188eu/core/rtw_ap.o' failed
make[2]: *** [/home/leo/rtl8188eu/core/rtw_ap.o] Error 1
Makefile:1655: recipe for target '_module_/home/leo/rtl8188eu' failed
make[1]: *** [_module_/home/leo/rtl8188eu] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-40-generic'
Makefile:155: recipe for target 'modules' failed
make: *** [modules] Error 2

kelebek333, 2020-02-18:
You can try to install from my PPA for Ubuntu based distro.

```
sudo add-apt-repository ppa:kelebek333/kablosuz
sudo apt-get update
sudo apt install  rtl8188eu-dkms
```

lwfinger, 2020-02-19:
It looks as if you once did a 'sudo make' and are now trying a simple 'make' NEVER, EVER do 'sudo make; You must do 'sudo make install; to install the driver.

Try the following:
sudo make clean
make
sudo make install

Remingtonhill, 2020-09-03:
Hey lwfinger thanks for your hard work on these projects.  I too had some make issues with CentOS, but I worked through them and finally got it compiled, and it was a great learning experience.