lwfinger/rtl8188eu #315: Installed but not working for RTL8188GU

ulanproger, 2020-02-29:
Hi, lwfinger!
I followed the instructions and successfully installed the driver. But Wi-Fi settings still saying: "No adapter found". My usb wifi adaper is RTL8188GU. Is this driver compatible with RTL8188GU?
System information:
Fedora 31
kernel ver. 5.4.15-200.fc31.x86_64
Thank you!

lwfinger, 2020-02-29:
I do not think so. The only way to tell is to get the USB ID using lsusb, and then add that ID to the USB_DEVICE table.

19chang19, 2020-03-01:
this rtlwifi 
vid pid is 
Device 008: ID 0bda:b711 Realtek Semiconductor Corp

not working

lwfinger, 2020-03-01:
If you added that USB ID to the driver, and it still failed, then the driver will NEVER support that device. In case you do not know how to do that, find the line that says 
 static struct usb_device_id rtw_usb_id_tbl[] = {
in file os_dep/usb_intf.c, and add a line that says 
USB_DEVICE(USB_VENDER_ID_REALTEK, 0xb711), .driver_info = RTL8188E},
right after it.

I could not find any Linux drivers on the web.

TOB5894, 2020-03-02:
@19chang19  Does it work now?

cslizhang, 2020-03-02:
I try,it doesn't work for rtl8188gu.

lwfinger, 2020-03-02:
It is likely a totally different device. Good luck finding a driver.

closed 2020-03-02 by lwfinger