lwfinger/rtl8188eu #317: [8188eu] staging driver crash when connect AP in kernel 4.19 (aarch64)

shlee6, 2020-03-05:
Dear beautiful engineer.

I'm trying to connect wifi using TL-WN725N(rtl8188eu) usb dongle in kernel 4.19 aarch64 (xilinx embedded board).
I enabled staging driver in kernel menu config and other config such as cfg80211, wpa_supplicant ... .
(I already finished to use wifi using rtl8188cu and driver(rtlwifi driver in mainline) but it cannot support wifi-direct, so change to rtl 8188eu chipset.)

WiFi network scanning is OK. (command 'iwlist wlan0 scan' is good.)
Device driver in mainline kernel is crashed when try to connect AP with following log.
(After 'iwconfig wlan0 essid "AP essid"' command, then driver seems to be crashed.)

===============================================================
[   70.192378] R8188EU: assoc success
[   70.195863] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   71.043144] Unable to handle kernel paging request at virtual address ffffff800cf00204
[   71.051057] Mem abort info:
[   71.053846]   ESR = 0x96000021
[   71.056893]   Exception class = DABT (current EL), IL = 32 bits
[   71.062806]   SET = 0, FnV = 0
[   71.065850]   EA = 0, S1PTW = 0
[   71.068983] Data abort info:
[   71.071857]   ISV = 0, ISS = 0x00000021
[   71.075685]   CM = 0, WnR = 0
[   71.078647] swapper pgtable: 4k pages, 39-bit VAs, pgdp = 00000000732409ea
[   71.085514] [ffffff800cf00204] pgd=000000004fffe803, pud=000000004fffe803, pmd=00000000075a1003, pte=00e8000040b09f13
[   71.096127] Internal error: Oops: 96000021 [#1] PREEMPT SMP
[   71.096130] Modules linked in: usb_f_rndis uio_ttc(O) kdmabuf(O) devicememory(O) uio_bp(O) r8188eu(C) uio_pdrv_genirq
[   71.096148] CPU: 0 PID: 1829 Comm: irq/51-xhci-hcd Tainted: G         C O      4.19.0-rt5 #1
[   71.096150] Hardware name: xlnx,zynqmp (DT)
[   71.096154] pstate: 20000005 (nzCv daif -PAN -UAO)
[   71.096166] pc : __ll_sc___cmpxchg_case_acq_8+0x4/0x20
[   71.096171] lr : rt_spin_lock+0x34/0x58
[   71.096172] sp : ffffff800ca7baf0
[   71.096174] x29: ffffff800ca7baf0 x28: 0000000000000000
[   71.096179] x27: ffffff800cef7860 x26: ffffff800cf001cc
[   71.096183] x25: ffffff800cf001ec x24: ffffff800ca12988
[   71.096187] x23: ffffff800cf001dc x22: ffffff8000aec000
[   71.096192] x21: ffffff800cef7840 x20: ffffffc04d22a700
[   71.096196] x19: ffffff800cf001ec x18: 0000000000000400
[   71.096200] x17: 0000000000000000 x16: ffffffc04d22a700
[   71.096204] x15: 0000000000001988 x14: 0000000000002278
[   71.096208] x13: 00000000ffffff92 x12: 0000000000000064
[   71.096212] x11: 0000000000000054 x10: 0000000000003c00
[   71.096216] x9 : ffffffc038a90067 x8 : 00000000004e69a6
[   71.096220] x7 : 0000000000000000 x6 : ffffffc02cf1da68
[   71.096224] x5 : 0000000000000000 x4 : 000000000003aaaa
[   71.096228] x3 : 0000000000000000 x2 : ffffffc04d22a700
[   71.096232] x1 : 0000000000000000 x0 : ffffff800cf00204
[   71.096237] Process irq/51-xhci-hcd (pid: 1829, stack limit = 0x00000000ef45eec0)
[   71.096239] Call trace:
[   71.096244]  __ll_sc___cmpxchg_case_acq_8+0x4/0x20
[   71.096273]  recv_func_posthandle+0x190/0x930 [r8188eu]
[   71.096295]  rtw_recv_entry+0xa5c/0x15d8 [r8188eu]
[   71.096316]  rtl8188eu_recv_tasklet+0xa4/0x578 [r8188eu]
[   71.096322]  tasklet_action_common.isra.0+0xa4/0x170
[   71.096325]  tasklet_action+0x20/0x28
[   71.096329]  do_current_softirqs+0x174/0x208
[   71.096332]  __local_bh_enable+0x50/0x80
[   71.096338]  irq_forced_thread_fn+0x64/0x90
[   71.096342]  irq_thread+0x11c/0x1d8
[   71.096346]  kthread+0x124/0x128
[   71.096350]  ret_from_fork+0x10/0x18
[   71.096356] Code: 35ffff91 aa1003e0 d65f03c0 f9800011 (c85ffc10)
[   71.288974] ---[ end trace 0000000000000002 ]---
[   71.288994] genirq: exiting task "irq/51-xhci-hcd" (1829) is an active IRQ thread (irq 51)
[   71.488560] irq 51: nobody cared (try booting with the "irqpoll" option)
[   71.488565] CPU: 0 PID: 1829 Comm: irq/51-xhci-hcd Tainted: G      D  C O      4.19.0-rt5 #1
[   71.488567] Hardware name: xlnx,zynqmp (DT)
[   71.488568] Call trace:
[   71.488574]  dump_backtrace+0x0/0x148
[   71.488578]  show_stack+0x14/0x20
[... 94 lines cut ...]
[   77.555346]  kthread+0x124/0x128
[   77.555349]  ret_from_fork+0x10/0x18
[   77.555355] Code: 35ffff91 aa1003e0 d65f03c0 f9800011 (c85ffc10)
===============================================================

and this is part of kernel starting dmesg
===============================================================
[    8.765066] usb 1-1: New USB device found, idVendor=0bda, idProduct=8179, bcdDevice= 0.00
[    8.773265] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    8.780406] usb 1-1: Product: 802.11n NIC
[    8.784416] usb 1-1: Manufacturer: Realtek
[    8.788513] usb 1-1: SerialNumber: 00E04C0001
[    8.800537] r8188eu: module is from the staging directory, the quality is unknown, you have been warned.
[    8.905229] Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)
===============================================================

I found that mainline driver version(supported max kernel version is  4.4.3 in TP LINK webpage) is not appropriate to my kernel version 4.19.
So I tried to switch device driver download from lwfinger git (branch is kernel-code), but compile is error because header <linux/time.h> is different between git code and kernel 4.19. 
(error : struct timer_list has no member named data).

So I will try to change 8188eu driver source code to match kernel 4.19, but I think it is not good to me...
Do you have any good idea or advice to me?
Is it batter to buy other wifi dongle to match kernel 4.19 and newer version?

Thanks.

lwfinger, 2020-03-05:
Your making arbitrary changes is the reason it is crashing. When something says that kernel up to vx.y is supported, that means that beyond that there are kernel API changes that prevent building.  There is nothing in the hardware that limits the kernel version. On the other hand, I fix those errors, and the drivers in my GitHub repos will build and run on kernels up tp v5.6. You should not have to change anything. Obviously, I cannot support your changes!!

shlee6, 2020-03-05:
Thanks for reply lwfinger.

I think it was a misunderstanding, I don't change any kernel source code when it is crashing.
I just enable staging rtl8188eu driver in mainline using menuconfig.

So I try to compile your repo, branch is kernel_code, but compile is error because <linux/time.h> is different in my kernel 4.19 version.

Do you have any advice for me?

Looking at your github and driver code, you've managed for a long time, I feel you are really an expert. Very impressive to me. Thank you for your service.
I will try to use other wifi dongle and chipset to run wireless AP.
It seems that running AP is more convenient than wifi-direct to customer.

Have a good day!

closed 2020-03-11 by shlee6

shlee6, 2020-03-12:
I got a new information of this problem.
The rtl8188eu driver is broken with Preempt_RT patch and it is known issue...
It is not a problem of rtl8188eu driver.

Detailed information is in under url.
https://github.com/armbian/build/issues/1537

gerhardteduardo, 2023-03-27:
Hello,
I had the same problem. 
Kernel Oops when using the rtl8188eu driver on the real-time system (kernel 5.4).
To solve this, I changed all spin_lock_bh to raw_spin_lock_bh.

I wouldn't want to change a kernel mainline driver.
But I needed to support this device in my project. 
It works perfectly.

Have a good day!

hsiang0100, 2023-09-13:
@gerhardteduardo 
I encountered the same issue while building the rtl8188eu driver.

I received the error:
error: passing argument 1 of 'rt_spin_lock' from incompatible pointer type [-Werror=incompatible-pointer-types]
at this line:
spin_lock_bh(&pmlmepriv->lock);

Like you, I replaced all instances of spin_lock_bh with raw_spin_lock_bh. However, I'm uncertain if this is the ideal solution and am concerned about potential issues it might cause. Have you encountered any problems after making this change? Thank you.