lwfinger/rtl8188eu #319: Error for wireless request "Set Mode" (8B06) :     SET failed on device wlx34c9f0834f2b ; Operation not permitted.

r3d9u11, 2020-03-06:
Hello. I'm trying to plugin USB Wifi based on chip rtl8188cu. But I was faced the trouble: `Error for wireless request "Set Mode" (8B06) : SET failed on device wlx34c9f0834f2b ; Operation not permitted.`

I'm using imx6-based SoM.
I've tried driver which included into kernel 4.1.36 (section "staging") and I've tried to install drvier from github (branch 4.1.8).

here is my log files:

```
root@imx6: iwconfig wlx34c9f0834f2b mode auto
Error for wireless request "Set Mode" (8B06) :
    SET failed on device wlx34c9f0834f2b ; Operation not permitted.
root@imx6: iwconfig wlx34c9f0834f2b mode ad-hoc
Error for wireless request "Set Mode" (8B06) :
    SET failed on device wlx34c9f0834f2b ; Operation not permitted.
root@imx6: iwconfig wlx34c9f0834f2b mode master
Error for wireless request "Set Mode" (8B06) :
    SET failed on device wlx34c9f0834f2b ; Operation not permitted.
root@imx6: iwconfig wlx34c9f0834f2b mode monitor
Error for wireless request "Set Mode" (8B06) :
    SET failed on device wlx34c9f0834f2b ; Operation not permitted.
```

```
#uname -a
Linux imx6 4.1.36-18288-ge95e34f-dirty #17 SMP PREEMPT Fri Mar 6 08:36:57 +03 2020 armv7l armv7l armv7l GNU/Linux
```

```
# iwconfig
wlx34c9f0834f2b  unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Auto  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

```
# ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.48.66  netmask 255.255.255.0  broadcast 192.168.48.255
        inet6 fe80::20d:15ff:fe01:17f8  prefixlen 64  scopeid 0x20<link>
        ether 00:0d:15:01:17:f8  txqueuelen 1000  (Ethernet)
        RX packets 130  bytes 11789 (11.7 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 17  bytes 2322 (2.3 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 0  (Local Loopback)
        RX packets 130  bytes 9190 (9.1 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 130  bytes 9190 (9.1 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

```
# ifconfig -a
can0: flags=128<NOARP>  mtu 16
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 10  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 36  

can1: flags=128<NOARP>  mtu 16
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 10  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 37  

eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.48.66  netmask 255.255.255.0  broadcast 192.168.48.255
        inet6 fe80::20d:15ff:fe01:17f8  prefixlen 64  scopeid 0x20<link>
        ether 00:0d:15:01:17:f8  txqueuelen 1000  (Ethernet)
        RX packets 1300  bytes 114186 (114.1 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 23  bytes 2626 (2.6 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 0  (Local Loopback)
        RX packets 126  bytes 8906 (8.9 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 126  bytes 8906 (8.9 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

sit0: flags=128<NOARP>  mtu 1480
        sit  txqueuelen 0  (IPv6-in-IPv4)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlx34c9f0834f2b: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether 34:c9:f0:83:4f:2b  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

```
# lshw -C network
  *-usb:1 DISABLED
       description: Wireless interface
       product: 802.11n NIC
       vendor: Realtek
       physical id: 2
       bus info: usb@1:1.2
       logical name: wlx34c9f0834f2b
       version: 0.00
       serial: 34:c9:f0:83:4f:2b
       capabilities: usb-2.00 ethernet physical wireless
       configuration: broadcast=yes driver=rtl8188eu maxpower=500mA multicast=yes speed=480Mbit/s wireless=unassociated
  *-network
       description: Ethernet interface
       physical id: 2
       logical name: eth0
       serial: 00:0d:15:01:17:f8
       size: 100Mbit/s
       capacity: 1Gbit/s
       capabilities: ethernet physical tp mii 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=fec driverversion=Revision: 1.0 duplex=full ip=192.168.48.66 link=yes multicast=yes port=MII speed=100Mbit/s
```

```
# lsmod
Module                  Size  Used by
mxc_v4l2_capture       27346  0
ipu_bg_overlay_sdc      5268  1 mxc_v4l2_capture
ipu_still               2296  1 mxc_v4l2_capture
ipu_prp_enc             5896  1 mxc_v4l2_capture
ipu_csi_enc             3861  1 mxc_v4l2_capture
v4l2_int_device         2897  2 ipu_csi_enc,mxc_v4l2_capture
ipu_fg_overlay_sdc      6090  1 mxc_v4l2_capture
8188eu                602932  0
mxc_dcic                6520  0

# ls /lib/firmware/rtlwifi/rtl8188eufw.bin
/lib/firmware/rtlwifi/rtl8188eufw.bin

# ls /etc/modprobe.d
50-8188eu.conf
8188eu.conf
blacklist-ath_pci.conf
blacklist-firewire.conf
blacklist-framebuffer.conf
blacklist-rare-network.conf
blacklist.conf
iwlwifi.conf
```

# modinfo 8188eu: [modinfo_8188eu.txt](https://github.com/lwfinger/rtl8188eu/files/4297229/modinfo_8188eu.txt)

What I'm doing wrong?
Thanks!

lwfinger, 2020-03-06:
Running in ad-hoc mode is not supported by that driver. Use the v5.2.2.4 branch. That version uses nl80211, and should handle ad-hoc connections.

closed 2020-03-06 by lwfinger

r3d9u11, 2020-03-09:
@lwfinger Hello, thanks for your answer. But
1) as you can see, I've tried to set defferent modes: auto, master, monitor, ah-hoc
2) I've just tried v5.2.2.4, nothing changed. I've got same error

lwfinger, 2020-03-09:
OK, iwconfig is totally obsolete, and nl80211 does not implement those commands, thus neither do the drivers. You need to use that appropriate commands of utility iw, or use Network Manager to set the mode you want. Note: NM is a lot easier than iw.

r3d9u11, 2020-03-10:
@lwfinger should I add some needed CONFIGs into my Kernel ?
As I told, I'm using ARM (imx6) platform

lwfinger, 2020-03-10:
No, you should not need any additional CONFIGs set if you are using a standard distro kernel. Do a Google search for "using networkmanager to create ad-hoc connections". There are several good articles.