lwfinger/rtl8188eu #320: Question to support AP mode with rtl8xxxu driver.

shlee6, 2020-03-11:
Dear engineers.

I have a few usb wifi adapter.
One of them is using rtl8188cu (manufacturer is ipTime in South Korea.)

I can see some experience that iptime usb wifi adapter can support as AP mode in windows.
So I tried to make AP using iptime adapter, but it seems that AP doesn't support in my Linux.(kernel 4.19. using mainline driver. "rtl8xxxu" )
Because command 'iw list' shows that there are no 'Supported interface mode : AP'.

Only two modes are supported.
Supported interface modes:
                 * managed
                 * monitor

Can you tell me why this driver can't support AP?
The problem is hostapd? (I used hostapd v2.9.)
or manufacturer? driver? need another customized(?) code in github?

I'm newbie of wireless driver and network SW but I can't understand why mainline driver can't support AP?

Thanks!

lwfinger, 2020-03-11:
Why are you asking about a driver for the RTL8188CU in the RTL8188EU repo? Do you see the difference in the name? The differences in the drivers are a lot bigger.

Since rtl8xxxu is in mainline, the proper place to discuss it would be in linux-wireless@vger.kernel.org. You should know, however, that they are quite likely to tell you that since the code is open-source, you should probably consider adding the extension yourself. Mainline contains a number of drivers that handle AP mode - they will show you how to do it. In my case, I prefer to spend the $35, or so, to buy a device that specializes as an AP and not put that load on my computer.

closed 2020-03-11 by lwfinger

shlee6, 2020-03-11:
@lwfinger I was so curious and frustrated that I wanted to ask you an expert. Thanks for the answer.