lwfinger/rtl8188eu #325: Error with wpa_supplicant > Driver does not support auten...

godlux98, 2020-04-13:
When i use wpa_supplicant (to connect wpa2 network) i got error:
"Driver does not support authenticaton/association or connect commands"

What i am doing wrong? Linux, ubuntu server, 4.15.0-96-generic.

Scanning networks work good.

lwfinger, 2020-04-13:
If you are using the driver built into the kernel, it does not support nl80211 commands. Without knowing how you are controlling wpa_supplicant, I have no idea what is wrong.

The problem is somewhere in your user-space control, not specifically in the driver.

godlux98, 2020-04-13:
I installed driver, just like it descripted in readme. And then, just follow this (https://www.linuxbabe.com/ubuntu/connect-to-wi-fi-from-terminal-on-ubuntu-18-04-19-04-with-wpa-supplicant) steps to connect wifi. Unfortunately, i don't know how does all system works, but at oldest version of kernel this driver had work fine.

So, if problem not in your side, i will close issue. But if you have a bit time to explain me where i can read about "good way to setup wifi" with your driver, maybe i may ask you for little help :)

Sorry for my bad english. Best regards. Godlux.

closed 2020-04-13 by godlux98

lwfinger, 2020-04-13:
Are you using a desktop? If so, install NetworkManager and let it control the wireless.

godlux98, 2020-04-14:
I am using Ubuntu server, so now I have only console access via ssh. I will try installing network-manager. 10Q.