lwfinger/rtl8188eu #327: make doesn't compile anything

Zerogoki00, 2020-05-10:
Hi. I'm trying to compile this driver for my tablet with Allwinner A13 CPU.
I have kernel 4.13 compiled in another directory (linux-4.13)

I compile module this way:
`ayanami@zerodell: ~/sunxi/rtl8188eu$ make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- -C ../linux-4.13/ M=`pwd` modules

make: Entering directory '/home/ayanami/sunxi/linux-4.13'
  Building modules, stage 2.
  MODPOST 0 modules
make: Leaving directory '/home/ayanami/sunxi/linux-4.13'
`
And nothing happens. The actual .ko file with module doesn't get created. What am I doing wrong?

lwfinger, 2020-05-10:
I just push a change that might help. Do a pull and try again.

Zerogoki00, 2020-05-10:
It started compiling but I have these errors 
```
/home/ayanami/sunxi/rtl8188eu/os_dep/os_intfs.c: In function 'rtw_init_netdev':
/home/ayanami/sunxi/rtl8188eu/os_dep/os_intfs.c:757:9: error: 'struct net_device' has no member named 'wireless_handlers'
  757 |  pnetdev->wireless_handlers = (struct iw_handler_def *)&rtw_handlers_def;
      |         ^~
make[1]: *** [scripts/Makefile.build:303: /home/ayanami/sunxi/rtl8188eu/os_dep/os_intfs.o] Error 1
make[1]: *** Waiting for unfinished jobs....
/home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.c:7814:3: error: 'struct iw_handler_def' has no member named 'private'
 7814 |  .private = rtw_private_handler,
      |   ^~~~~~~
/home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.c:7814:13: error: initialization of 'struct iw_statistics * (*)(struct net_device *)' from incompatible pointer type 'int (**)(struct net_device *, struct iw_request_info *, union iwreq_data *, char *)' [-Werror=incompatible-pointer-types]
 7814 |  .private = rtw_private_handler,
      |             ^~~~~~~~~~~~~~~~~~~
/home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.c:7814:13: note: (near initialization for 'rtw_handlers_def.get_wireless_stats')
/home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.c:7815:3: error: 'struct iw_handler_def' has no member named 'private_args'
 7815 |  .private_args = (struct iw_priv_args *)rtw_private_args,
      |   ^~~~~~~~~~~~
/home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.c:7815:18: warning: excess elements in struct initializer
 7815 |  .private_args = (struct iw_priv_args *)rtw_private_args,
      |                  ^
/home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.c:7815:18: note: (near initialization for 'rtw_handlers_def')
/home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.c:7816:3: error: 'struct iw_handler_def' has no member named 'num_private'
 7816 |  .num_private = sizeof(rtw_private_handler) / sizeof(iw_handler),
      |   ^~~~~~~~~~~
/home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.c:7816:17: warning: excess elements in struct initializer
 7816 |  .num_private = sizeof(rtw_private_handler) / sizeof(iw_handler),
      |                 ^~~~~~
/home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.c:7816:17: note: (near initialization for 'rtw_handlers_def')
/home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.c:7817:3: error: 'struct iw_handler_def' has no member named 'num_private_args'
 7817 |  .num_private_args = sizeof(rtw_private_args) / sizeof(struct iw_priv_args),
      |   ^~~~~~~~~~~~~~~~
/home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.c:7817:22: warning: excess elements in struct initializer
 7817 |  .num_private_args = sizeof(rtw_private_args) / sizeof(struct iw_priv_args),
      |                      ^~~~~~
/home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.c:7817:22: note: (near initialization for 'rtw_handlers_def')
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:302: /home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.o] Error 1
make: *** [Makefile:1512: _module_/home/ayanami/sunxi/rtl8188eu] Error 2
make: Leaving directory '/home/ayanami/sunxi/linux-4.13'
```

I also tried it for latest 5.6 kernel and have the same errors.

Zerogoki00, 2020-05-10:
I still have the same result. (tried with 4.13 and latest 5.6 kernels)
I also tried this with master branch and module **started compiling** but I got these errors
```/home/ayanami/sunxi/rtl8188eu/os_dep/os_intfs.c: In function 'rtw_init_netdev':
/home/ayanami/sunxi/rtl8188eu/os_dep/os_intfs.c:757:9: error: 'struct net_device' has no member named 'wireless_handlers'
  757 |  pnetdev->wireless_handlers = (struct iw_handler_def *)&rtw_handlers_def;
      |         ^~
  CC [M]  /home/ayanami/sunxi/rtl8188eu/os_dep/rtw_android.o
  CC [M]  /home/ayanami/sunxi/rtl8188eu/os_dep/usb_intf.o
/home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.c:7814:3: error: 'struct iw_handler_def' has no member named 'private'
 7814 |  .private = rtw_private_handler,
      |   ^~~~~~~
/home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.c:7814:13: error: initialization of 'struct iw_statistics * (*)(struct net_device *)' from incompatible pointer type 'int (**)(struct net_device *, struct iw_request_info *, union iwreq_data *, char *)' [-Werror=incompatible-pointer-types]
 7814 |  .private = rtw_private_handler,
      |             ^~~~~~~~~~~~~~~~~~~
/home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.c:7814:13: note: (near initialization for 'rtw_handlers_def.get_wireless_stats')
/home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.c:7815:3: error: 'struct iw_handler_def' has no member named 'private_args'
 7815 |  .private_args = (struct iw_priv_args *)rtw_private_args,
      |   ^~~~~~~~~~~~
/home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.c:7815:18: warning: excess elements in struct initializer
 7815 |  .private_args = (struct iw_priv_args *)rtw_private_args,
      |                  ^
/home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.c:7815:18: note: (near initialization for 'rtw_handlers_def')
/home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.c:7816:3: error: 'struct iw_handler_def' has no member named 'num_private'
 7816 |  .num_private = sizeof(rtw_private_handler) / sizeof(iw_handler),
      |   ^~~~~~~~~~~
/home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.c:7816:17: warning: excess elements in struct initializer
 7816 |  .num_private = sizeof(rtw_private_handler) / sizeof(iw_handler),
      |                 ^~~~~~
/home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.c:7816:17: note: (near initialization for 'rtw_handlers_def')
make[1]: *** [scripts/Makefile.build:268: /home/ayanami/sunxi/rtl8188eu/os_dep/os_intfs.o] Error 1
/home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.c:7817:3: error: 'struct iw_handler_def' has no member named 'num_private_args'
 7817 |  .num_private_args = sizeof(rtw_private_args) / sizeof(struct iw_priv_args),
      |   ^~~~~~~~~~~~~~~~
make[1]: *** Waiting for unfinished jobs....
/home/ayanami/sunxi/rtl8188eu/os_dep/ioctl_linux.c:7817:22: warning: excess elements in struct initializer
 7817 |  .num_private_args = sizeof(rtw_private_args) / sizeof(struct iw_priv_args),
      |                      ^~~~~~
```

lwfinger, 2020-05-10:
Your kernels need to have CONFIG_WIRELESS_EXT in the configuration.

closed 2020-05-10 by lwfinger