lwfinger/rtl8188eu #329: Getting this work for Ubuntu 20.04 after suspend

ta1bbty, 2020-05-20:
Sorry, if I am asking this again ( I couldn't find it in the issues or in PR ).

## Problem
I've this TPlink TLWN722N which on `lsusb` shows : 
```shell
Bus 001 Device 006: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
```
After I start my laptop from suspend mode, the WiFi gets hardware disabled.
Before suspending the internal wifi and this usb wifi, both work perfectly fine.
But after starting from suspension none of the commands show the usb device except in the network manager and `lsusb`
#### Output of some of the commands
```
❯ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 03f0:0941 HP, Inc X500 Optical Mouse
Bus 001 Device 003: ID 04f2:b50c Chicony Electronics Co., Ltd HP Truevision HD
Bus 001 Device 006: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
[image]

```
❯ iw dev
phy#0
	Interface wlo1
		ifindex 4
		wdev 0x1
		addr d8:5d:e2:05:1a:0f
		type managed
		txpower 20.00 dBm
```
[image]

```
❯ rfkill list
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: yes
```
[image]

Network Manager
[image]

Output of `uname -a`
```
❯ uname -a
Linux ta1bbty 5.4.0-29-generic #33-Ubuntu SMP Wed Apr 29 14:32:27 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

I've tried this repo as well as [quickreflex's one](https://github.com/quickreflex/rtl8188eus) and there's a PR on that repo for `kernel >= 5.2.0` [here](https://github.com/quickreflex/rtl8188eus/pull/40).
I've tried that one too but in vain.
The PR there got my driver to be compiled but loading it using `insmod` or the `dkms` method didn't work.

So, I cloned this and compiled from the master branch, however that also didn't work.
Now going through the issues I found [this one](https://github.com/lwfinger/rtl8188eu/issues/271#issuecomment-441079354).
Here @lwfinger  mentions that 
> All of my recent work is with the v5.2.2.4 branch of this repo. That driver works very well although you do need to blacklist r8188eu and rtl8xxxu.

So I checked out on branch `v5.2.2.4` and the driver did compile.
`make all` worked fine, I haven't done `sudo make install`, cause the last time I did this I had to reinstall ubuntu 20.04. Since it got a bit unstable after that, it was lagging and some other issues but that was with the master branch or the `quickreflex` one, I don't remember exactly :sweat_smile: 

## How to proceed with this ?

Now, in the README , there isn't any mention of blacklisting those two drivers and moreover I don't even know if those exists and if they exist how can I blacklist them ( I would figure that out if that's the way to go ).

Hope that gives a better context of my problem.

Let me know, if you have any ideas as to how to solve this issue or if some more info is required.

wegi1, 2021-04-11:
It wouldbe a HW problem not SW. On my PC machine sometimes this dongle is totally invisible. No any lshw or lsusb can't show this dongle. I can't find any command to wake up this only it work when I eject it from USB port and put to USB port one more time.
rfkill even don't see that as well.

Myguess it would be BIOS problem.

korovkin, 2021-08-22:
i have the same issue, the only thing that works is physically ejecting the dongle from the USB

teja456, 2021-09-06:
I too have the same problem , I had a archer t2u , and it didnot work , so brought this new one .
This is also same ....not working on Ubuntu server

lwfinger, 2021-09-06:
If your system cannot see the device using the lsusb command, that is a HW problem. No change in the SW can fit it.

closed 2021-09-06 by lwfinger