lwfinger/rtl8188eu #330: Compiler failure 5.4.35

berkutta, 2020-05-28:
Hi,

I'm trying to build rtl8188eu within a buildroot environment. I use Kernel 5.4.35 without any patches. `cfg80211 - wireless configuration API` and `cfg80211 wireless extensions compatibility` is enabled in menuconfig. Using the newest commit from master.

```
>>> rtl8188eu ac42434b2ad72887cd9b147780b30978f1c7b728 Building kernel module(s)
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/run/media/vm/buildroot-2020.05-rc2-wifi/output/build/linux-5.4.35'
  CC [M]  /run/media/vm/buildroot-2020.05-rc2-wifi/output/build/rtl8188eu-ac42434b2ad72887cd9b147780b30978f1c7b728/./os_dep/ioctl_linux.o
/run/media/vm/buildroot-2020.05-rc2-wifi/output/build/rtl8188eu-ac42434b2ad72887cd9b147780b30978f1c7b728/./os_dep/ioctl_linux.c:7814:3: error: ‘struct iw_handler_def’ has no member named ‘private’
  .private = rtw_private_handler,
   ^~~~~~~
/run/media/vm/buildroot-2020.05-rc2-wifi/output/build/rtl8188eu-ac42434b2ad72887cd9b147780b30978f1c7b728/./os_dep/ioctl_linux.c:7814:13: error: initialization of ‘struct iw_statistics * (*)(struct net_device *)’ from incompatible pointer type ‘int (**)(struct net_device *, struct iw_request_info *, union iwreq_data *, char *)’ [-Werror=incompatible-pointer-types]
  .private = rtw_private_handler,
             ^~~~~~~~~~~~~~~~~~~
/run/media/vm/buildroot-2020.05-rc2-wifi/output/build/rtl8188eu-ac42434b2ad72887cd9b147780b30978f1c7b728/./os_dep/ioctl_linux.c:7814:13: note: (near initialization for ‘rtw_handlers_def.get_wireless_stats’)
/run/media/vm/buildroot-2020.05-rc2-wifi/output/build/rtl8188eu-ac42434b2ad72887cd9b147780b30978f1c7b728/./os_dep/ioctl_linux.c:7815:3: error: ‘struct iw_handler_def’ has no member named ‘private_args’
  .private_args = (struct iw_priv_args *)rtw_private_args,
   ^~~~~~~~~~~~
/run/media/vm/buildroot-2020.05-rc2-wifi/output/build/rtl8188eu-ac42434b2ad72887cd9b147780b30978f1c7b728/./os_dep/ioctl_linux.c:7815:18: warning: excess elements in struct initializer
  .private_args = (struct iw_priv_args *)rtw_private_args,
                  ^
/run/media/vm/buildroot-2020.05-rc2-wifi/output/build/rtl8188eu-ac42434b2ad72887cd9b147780b30978f1c7b728/./os_dep/ioctl_linux.c:7815:18: note: (near initialization for ‘rtw_handlers_def’)
/run/media/vm/buildroot-2020.05-rc2-wifi/output/build/rtl8188eu-ac42434b2ad72887cd9b147780b30978f1c7b728/./os_dep/ioctl_linux.c:7816:3: error: ‘struct iw_handler_def’ has no member named ‘num_private’
  .num_private = sizeof(rtw_private_handler) / sizeof(iw_handler),
   ^~~~~~~~~~~
/run/media/vm/buildroot-2020.05-rc2-wifi/output/build/rtl8188eu-ac42434b2ad72887cd9b147780b30978f1c7b728/./os_dep/ioctl_linux.c:7816:17: warning: excess elements in struct initializer
  .num_private = sizeof(rtw_private_handler) / sizeof(iw_handler),
                 ^~~~~~
/run/media/vm/buildroot-2020.05-rc2-wifi/output/build/rtl8188eu-ac42434b2ad72887cd9b147780b30978f1c7b728/./os_dep/ioctl_linux.c:7816:17: note: (near initialization for ‘rtw_handlers_def’)
/run/media/vm/buildroot-2020.05-rc2-wifi/output/build/rtl8188eu-ac42434b2ad72887cd9b147780b30978f1c7b728/./os_dep/ioctl_linux.c:7817:3: error: ‘struct iw_handler_def’ has no member named ‘num_private_args’
  .num_private_args = sizeof(rtw_private_args) / sizeof(struct iw_priv_args),
   ^~~~~~~~~~~~~~~~
/run/media/vm/buildroot-2020.05-rc2-wifi/output/build/rtl8188eu-ac42434b2ad72887cd9b147780b30978f1c7b728/./os_dep/ioctl_linux.c:7817:22: warning: excess elements in struct initializer
  .num_private_args = sizeof(rtw_private_args) / sizeof(struct iw_priv_args),
                      ^~~~~~
/run/media/vm/buildroot-2020.05-rc2-wifi/output/build/rtl8188eu-ac42434b2ad72887cd9b147780b30978f1c7b728/./os_dep/ioctl_linux.c:7817:22: note: (near initialization for ‘rtw_handlers_def’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:266: /run/media/vm/buildroot-2020.05-rc2-wifi/output/build/rtl8188eu-ac42434b2ad72887cd9b147780b30978f1c7b728/./os_dep/ioctl_linux.o] Error 1
make[1]: *** [Makefile:1691: /run/media/vm/buildroot-2020.05-rc2-wifi/output/build/rtl8188eu-ac42434b2ad72887cd9b147780b30978f1c7b728/.] Error 2
make[1]: Leaving directory '/run/media/vm/buildroot-2020.05-rc2-wifi/output/build/linux-5.4.35'
make: *** [package/pkg-generic.mk:270: /run/media/vm/buildroot-2020.05-rc2-wifi/output/build/rtl8188eu-ac42434b2ad72887cd9b147780b30978f1c7b728/.stamp_built] Error 2

```

lwfinger, 2020-05-28:
You have a kernel configuration problem if you want to use the master branch as it requires CONFIG_WIRELESS_EXT. If you switch to branch v5.2.2.4, then  having wireless and cfg80211 will be sufficient. Do the following:
git checkout origin/v5.2.2.4 -b v5.2.2.4
git pull

berkutta, 2020-05-29:
Thanks, this worked for me. So my conclusion is correct, that there is no CONFIG_WIRELESS_EXT available any more in newer kernels and drivers need to update to be compatible again?

lwfinger, 2020-05-29:
Absolutely NOT. CONFIG_WIRELESS_EXT is still available. Your DISTRO chose not to enable it.

closed 2020-05-29 by lwfinger

Nileshbhavsar, 2021-05-10:
Hi,

I have same problem. Please check log below.

emdev@emdev-Macmini:~/quectel/rtl8188eu$ sudo make ARCH=arm CROSS_COMPILE=~/quectel/Atreyo_GSM_SOUND_WD_SD_6_5_2021/crossgcc/gcc-linaro-7.3.1-2018.05-x86_64_arm-linux-gnueabihf/bin/arm-linux-gnueabihf- -C ~/quectel/Atreyo_GSM_SOUND_WD_SD_6_5_2021/linux/kernel/ M=/home/emdev/quectel/rtl8188eu  modules
make: Entering directory '/home/emdev/quectel/Atreyo_GSM_SOUND_WD_SD_6_5_2021/linux/kernel'
  CC [M]  /home/emdev/quectel/rtl8188eu/core/rtw_ap.o
  CC [M]  /home/emdev/quectel/rtl8188eu/core/rtw_br_ext.o
  CC [M]  /home/emdev/quectel/rtl8188eu/core/rtw_cmd.o
  CC [M]  /home/emdev/quectel/rtl8188eu/core/rtw_debug.o
  CC [M]  /home/emdev/quectel/rtl8188eu/core/rtw_efuse.o
  CC [M]  /home/emdev/quectel/rtl8188eu/core/rtw_ieee80211.o
  CC [M]  /home/emdev/quectel/rtl8188eu/core/rtw_io.o
  CC [M]  /home/emdev/quectel/rtl8188eu/core/rtw_ioctl_set.o
  CC [M]  /home/emdev/quectel/rtl8188eu/core/rtw_iol.o
  CC [M]  /home/emdev/quectel/rtl8188eu/core/rtw_led.o
  CC [M]  /home/emdev/quectel/rtl8188eu/core/rtw_mlme.o
  CC [M]  /home/emdev/quectel/rtl8188eu/core/rtw_mlme_ext.o
  CC [M]  /home/emdev/quectel/rtl8188eu/core/rtw_mp.o
  CC [M]  /home/emdev/quectel/rtl8188eu/core/rtw_mp_ioctl.o
  CC [M]  /home/emdev/quectel/rtl8188eu/core/rtw_pwrctrl.o
  CC [M]  /home/emdev/quectel/rtl8188eu/core/rtw_p2p.o
  CC [M]  /home/emdev/quectel/rtl8188eu/core/rtw_recv.o
  CC [M]  /home/emdev/quectel/rtl8188eu/core/rtw_rf.o
  CC [M]  /home/emdev/quectel/rtl8188eu/core/rtw_security.o
  CC [M]  /home/emdev/quectel/rtl8188eu/core/rtw_sreset.o
  CC [M]  /home/emdev/quectel/rtl8188eu/core/rtw_sta_mgt.o
  CC [M]  /home/emdev/quectel/rtl8188eu/core/rtw_wlan_util.o
  CC [M]  /home/emdev/quectel/rtl8188eu/core/rtw_xmit.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/HalHWImg8188E_MAC.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/HalHWImg8188E_BB.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/HalHWImg8188E_RF.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/HalPhyRf.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/HalPhyRf_8188e.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/HalPwrSeqCmd.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/Hal8188EPwrSeq.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/Hal8188ERateAdaptive.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/hal_intf.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/hal_com.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/odm.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/odm_debug.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/odm_interface.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/odm_HWConfig.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/odm_RegConfig8188E.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/odm_RTL8188E.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/rtl8188e_cmd.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/rtl8188e_dm.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/rtl8188e_hal_init.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/rtl8188e_mp.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/rtl8188e_phycfg.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/rtl8188e_rf6052.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/rtl8188e_rxdesc.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/rtl8188e_sreset.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/rtl8188e_xmit.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/rtl8188eu_led.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/rtl8188eu_recv.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/rtl8188eu_xmit.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/usb_halinit.o
  CC [M]  /home/emdev/quectel/rtl8188eu/hal/usb_ops_linux.o
  CC [M]  /home/emdev/quectel/rtl8188eu/os_dep/ioctl_linux.o
/home/emdev/quectel/rtl8188eu/os_dep/ioctl_linux.c:7814:3: error: ‘struct iw_handler_def’ has no member named ‘private’
  .private = rtw_private_handler,
   ^~~~~~~
/home/emdev/quectel/rtl8188eu/os_dep/ioctl_linux.c:7814:13: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .private = rtw_private_handler,
             ^~~~~~~~~~~~~~~~~~~
/home/emdev/quectel/rtl8188eu/os_dep/ioctl_linux.c:7814:13: note: (near initialization for ‘rtw_handlers_def.get_wireless_stats’)
/home/emdev/quectel/rtl8188eu/os_dep/ioctl_linux.c:7815:3: error: ‘struct iw_handler_def’ has no member named ‘private_args’
  .private_args = (struct iw_priv_args *)rtw_private_args,
   ^~~~~~~~~~~~
/home/emdev/quectel/rtl8188eu/os_dep/ioctl_linux.c:7815:18: warning: excess elements in struct initializer
  .private_args = (struct iw_priv_args *)rtw_private_args,
                  ^
/home/emdev/quectel/rtl8188eu/os_dep/ioctl_linux.c:7815:18: note: (near initialization for ‘rtw_handlers_def’)
/home/emdev/quectel/rtl8188eu/os_dep/ioctl_linux.c:7816:3: error: ‘struct iw_handler_def’ has no member named ‘num_private’
  .num_private = sizeof(rtw_private_handler) / sizeof(iw_handler),
   ^~~~~~~~~~~
/home/emdev/quectel/rtl8188eu/os_dep/ioctl_linux.c:7816:17: warning: excess elements in struct initializer
  .num_private = sizeof(rtw_private_handler) / sizeof(iw_handler),
                 ^~~~~~
/home/emdev/quectel/rtl8188eu/os_dep/ioctl_linux.c:7816:17: note: (near initialization for ‘rtw_handlers_def’)
/home/emdev/quectel/rtl8188eu/os_dep/ioctl_linux.c:7817:3: error: ‘struct iw_handler_def’ has no member named ‘num_private_args’
  .num_private_args = sizeof(rtw_private_args) / sizeof(struct iw_priv_args),
   ^~~~~~~~~~~~~~~~
/home/emdev/quectel/rtl8188eu/os_dep/ioctl_linux.c:7817:22: warning: excess elements in struct initializer
[... cut ...]
/home/emdev/quectel/rtl8188eu/os_dep/ioctl_linux.c:7817:22: note: (near initialization for ‘rtw_handlers_def’)
cc1: some warnings being treated as errors
scripts/Makefile.build:265: recipe for target '/home/emdev/quectel/rtl8188eu/os_dep/ioctl_linux.o' failed
make[1]: *** [/home/emdev/quectel/rtl8188eu/os_dep/ioctl_linux.o] Error 1
Makefile:1692: recipe for target '/home/emdev/quectel/rtl8188eu' failed
make: *** [/home/emdev/quectel/rtl8188eu] Error 2
make: Leaving directory '/home/emdev/quectel/Atreyo_GSM_SOUND_WD_SD_6_5_2021/linux/kernel'

lwfinger, 2021-05-10:
As you have the same problem, why didn't you use the fix? I repeat it here:
You have a kernel configuration problem if you want to use the master branch as it requires CONFIG_WIRELESS_EXT. If you switch to branch v5.2.2.4, then having wireless and cfg80211 will be sufficient. Do the following:
git checkout origin/v5.2.2.4 -b v5.2.2.4
git pull

64-coreCPU, 2021-08-10:
@lwfinger Please specify in Readme the need to enable the Config_Wireless_Ext option in the kernel - this will greatly simplify life when assembling the module. Thanks in advance.