lwfinger/rtl8188eu #331: RTL8188EU concurrent mode

Quang-vip, 2020-06-09:
I'm wondering if this driver supports concurrent mode? (Running both station and AP at the same time)

I'm planning to make a dirt cheap wifi extender with my Orange Pi using my TP-Link WN722N v2 wifi adapter, which utilizes a _rtl8188eu_

I tried to add the `EXTRA_CFLAGS += -DCONFIG_CONCURRENT_MODE` flag in the 'Make' file and
enable the `CONFIG_CONCURRENT_MODE` definition in the **autoconf.h** header file
but I still got nothing, it still remained as 1 interface only.

lwfinger, 2020-06-09:
If you are running the master branch, it will not run concurrent mode. If you run the v5.2.2.4 branch, then you will have the full capabilities of mac80211 available.

Quang-vip, 2020-06-10:
Hmmm, weird...

Despite having loaded the driver from the v5.2.2.4 branch and properly loaded the module (modinfo shows **8188eu**'s version being v5.2.2.4), it still stays at 1 interface only.

Perhaps I missed something..