lwfinger/rtl8188eu #332: RTL_IOCTL_HOSTAPD can be called with different data length than expected

ardovm, 2020-06-10:
This report is for the master branch.

Commit 02614d02dc6a1da42597ae70823bf62876e475dc changes file `os_dep/ioctl_linux.c`, adding a check that the data passed to ioctl `RTL_IOCTL_HOSTAPD` always have the same size, i.e. the size of a `struct ieee_param`.

However, some functions in file `hostapd-0.8/src/drivers/driver_rtw.c` seem to call the IOCTL using _different_ data sizes: [rtl871x_set_beacon_ops()](https://github.com/lwfinger/rtl8188eu/blob/ac42434b2ad72887cd9b147780b30978f1c7b728/hostapd-0.8/src/drivers/driver_rtw.c#L1361), [rtl871x_set_key_ops](https://github.com/lwfinger/rtl8188eu/blob/ac42434b2ad72887cd9b147780b30978f1c7b728/hostapd-0.8/src/drivers/driver_rtw.c#L1440),
[rtl871x_set_wps_assoc_resp_ie()](https://github.com/lwfinger/rtl8188eu/blob/ac42434b2ad72887cd9b147780b30978f1c7b728/hostapd-0.8/src/drivers/driver_rtw.c#L1575).

Should the check on the IOCTL parameter's be different, then?

ardovm, 2020-06-11:
Thank you very much!

ardovm, 2020-06-11:
I am afraid we declared this problem solved too soon.

The function [rtl871x_set_key_ops](https://github.com/lwfinger/rtl8188eu/blob/71af295db5415e41a70343e90213169ee307e878/hostapd-0.8/src/drivers/driver_rtw.c#L1394) can also send a _bigger_ buffer than `sizeof(struct ieee_param)`.

Maybe the check should be removed at all? Or made particular for each IOCTL type?

Please let me know if I can be of any help. I am not proposing any pull requests because this is not "my cup of tea", but I am available to help if you give me instructions.

lwfinger, 2020-06-11:
How big is the one from rtl871x_set_key_ops?

ardovm, 2020-06-12:
> How big is the one from rtl871x_set_key_ops?

Apparently, it is `sizeof(struct ieee_param)` plus the length of the key.
For what I know, WEP keys can have different sizes, and for WPA keys I am not sure.
There is the possibility to check that the buffer has the correct size, because the key size is put into another field of the struct.
And in fact, the driver does check in function [rtw_set_encryption()](https://github.com/lwfinger/rtl8188eu/blob/ac42434b2ad72887cd9b147780b30978f1c7b728/os_dep/ioctl_linux.c#L4896) that the size of the data matches the expectations, for that particular command (i.e. value of `ieee_param.cmd`).

Other commands are not checked. If your goal is to be sure that all data has the correct size, we should add individual checks for each value of `ieee_param.cmd`. They could be inserted into individual rtw_xxx_() functions in os_dep/ioctl_linux.c.

Other than this, I think that the only other possible solution is to completely remove the check on the size of the data passed to the ioctl, i.e. to revert commit 02614d0.

What do you think?

lwfinger, 2020-06-12:
I am not willing to very much work on this outdated driver. The size of the key should be less than 100 bytes (WEP is up to 40 and WPA is shorter), thus I will change to allow that size.

ardovm, 2020-06-15:
It seems to work perfectly now. Thank you again!