lwfinger/rtl8188eu #334: Cross-compiling for Xilinx Zynq (ARM) board

mikeboich, 2020-07-24:
Thanks for making this code available!  I downloaded (via Git) to my X86 development system.  I'm using kernel 4.9.0 which is created by the Petalinux/Bitbake system.  When I point KSRC to the kernel source (after doing make defconfig and make prepare, it _almost_ builds, but produces the errors below.  Any hints?

thx

Mike

mike@mike-MS-7B48:~/Downloads/rtl8188eu$ make
make ARCH=arm CROSS_COMPILE=arm-none-eabi- -C ~/peta_test/Petalinux-Cora-Z7-10/Cora-Z7-10/build/tmp/work-shared/plnx_arm/kernel-source M=/home/mike/Downloads/rtl8188eu  modules
make[1]: Entering directory '/home/mike/peta_test/Petalinux-Cora-Z7-10/Cora-Z7-10/build/tmp/work-shared/plnx_arm/kernel-source'

  WARNING: Symbol version dump ./Module.symvers
           is missing; modules will have no dependencies and modversions.

  CC [M]  /home/mike/Downloads/rtl8188eu/core/rtw_ap.o
  CC [M]  /home/mike/Downloads/rtl8188eu/core/rtw_br_ext.o
  CC [M]  /home/mike/Downloads/rtl8188eu/core/rtw_cmd.o
  CC [M]  /home/mike/Downloads/rtl8188eu/core/rtw_debug.o
  CC [M]  /home/mike/Downloads/rtl8188eu/core/rtw_efuse.o
  CC [M]  /home/mike/Downloads/rtl8188eu/core/rtw_ieee80211.o
  CC [M]  /home/mike/Downloads/rtl8188eu/core/rtw_io.o
  CC [M]  /home/mike/Downloads/rtl8188eu/core/rtw_ioctl_set.o
  CC [M]  /home/mike/Downloads/rtl8188eu/core/rtw_iol.o
  CC [M]  /home/mike/Downloads/rtl8188eu/core/rtw_led.o
  CC [M]  /home/mike/Downloads/rtl8188eu/core/rtw_mlme.o
  CC [M]  /home/mike/Downloads/rtl8188eu/core/rtw_mlme_ext.o
  CC [M]  /home/mike/Downloads/rtl8188eu/core/rtw_mp.o
  CC [M]  /home/mike/Downloads/rtl8188eu/core/rtw_mp_ioctl.o
  CC [M]  /home/mike/Downloads/rtl8188eu/core/rtw_pwrctrl.o
  CC [M]  /home/mike/Downloads/rtl8188eu/core/rtw_p2p.o
  CC [M]  /home/mike/Downloads/rtl8188eu/core/rtw_recv.o
  CC [M]  /home/mike/Downloads/rtl8188eu/core/rtw_rf.o
  CC [M]  /home/mike/Downloads/rtl8188eu/core/rtw_security.o
  CC [M]  /home/mike/Downloads/rtl8188eu/core/rtw_sreset.o
  CC [M]  /home/mike/Downloads/rtl8188eu/core/rtw_sta_mgt.o
  CC [M]  /home/mike/Downloads/rtl8188eu/core/rtw_wlan_util.o
  CC [M]  /home/mike/Downloads/rtl8188eu/core/rtw_xmit.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/HalHWImg8188E_MAC.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/HalHWImg8188E_BB.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/HalHWImg8188E_RF.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/HalPhyRf.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/HalPhyRf_8188e.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/HalPwrSeqCmd.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/Hal8188EPwrSeq.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/Hal8188ERateAdaptive.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/hal_intf.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/hal_com.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/odm.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/odm_debug.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/odm_interface.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/odm_HWConfig.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/odm_RegConfig8188E.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/odm_RTL8188E.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/rtl8188e_cmd.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/rtl8188e_dm.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/rtl8188e_hal_init.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/rtl8188e_mp.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/rtl8188e_phycfg.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/rtl8188e_rf6052.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/rtl8188e_rxdesc.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/rtl8188e_sreset.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/rtl8188e_xmit.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/rtl8188eu_led.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/rtl8188eu_recv.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/rtl8188eu_xmit.o
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/usb_halinit.o
/home/mike/Downloads/rtl8188eu/hal/usb_halinit.c: In function 'GetHwReg8188EU':
/home/mike/Downloads/rtl8188eu/hal/usb_halinit.c:1926:3: warning: empty declaration
   __attribute__((__fallthrough__));
   ^~~~~~~~~~~~~
/home/mike/Downloads/rtl8188eu/hal/usb_halinit.c:1926:3: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
  CC [M]  /home/mike/Downloads/rtl8188eu/hal/usb_ops_linux.o
/home/mike/Downloads/rtl8188eu/hal/usb_ops_linux.c: In function 'usb_read_port_complete':
/home/mike/Downloads/rtl8188eu/hal/usb_ops_linux.c:556:4: warning: empty declaration
    __attribute__((__fallthrough__));
    ^~~~~~~~~~~~~
/home/mike/Downloads/rtl8188eu/hal/usb_ops_linux.c:556:4: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
  CC [M]  /home/mike/Downloads/rtl8188eu/os_dep/ioctl_linux.o
/home/mike/Downloads/rtl8188eu/os_dep/ioctl_linux.c:7814:2: error: unknown field 'private' specified in initializer
  .private = rtw_private_handler,
  ^
/home/mike/Downloads/rtl8188eu/os_dep/ioctl_linux.c:7814:13: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .private = rtw_private_handler,
             ^~~~~~~~~~~~~~~~~~~
/home/mike/Downloads/rtl8188eu/os_dep/ioctl_linux.c:7814:13: note: (near initialization for 'rtw_handlers_def.get_wireless_stats')
/home/mike/Downloads/rtl8188eu/os_dep/ioctl_linux.c:7815:2: error: unknown field 'private_args' specified in initializer
  .private_args = (struct iw_priv_args *)rtw_private_args,
  ^
/home/mike/Downloads/rtl8188eu/os_dep/ioctl_linux.c:7815:18: warning: excess elements in struct initializer
  .private_args = (struct iw_priv_args *)rtw_private_args,
                  ^
/home/mike/Downloads/rtl8188eu/os_dep/ioctl_linux.c:7815:18: note: (near initialization for 'rtw_handlers_def')
/home/mike/Downloads/rtl8188eu/os_dep/ioctl_linux.c:7816:2: error: unknown field 'num_private' specified in initializer
  .num_private = sizeof(rtw_private_handler) / sizeof(iw_handler),
  ^
/home/mike/Downloads/rtl8188eu/os_dep/ioctl_linux.c:7816:17: warning: excess elements in struct initializer
  .num_private = sizeof(rtw_private_handler) / sizeof(iw_handler),
                 ^~~~~~
/home/mike/Downloads/rtl8188eu/os_dep/ioctl_linux.c:7816:17: note: (near initialization for 'rtw_handlers_def')
/home/mike/Downloads/rtl8188eu/os_dep/ioctl_linux.c:7817:2: error: unknown field 'num_private_args' specified in initializer
  .num_private_args = sizeof(rtw_private_args) / sizeof(struct iw_priv_args),
  ^
/home/mike/Downloads/rtl8188eu/os_dep/ioctl_linux.c:7817:22: warning: excess elements in struct initializer
  .num_private_args = sizeof(rtw_private_args) / sizeof(struct iw_priv_args),
                      ^~~~~~
/home/mike/Downloads/rtl8188eu/os_dep/ioctl_linux.c:7817:22: note: (near initialization for 'rtw_handlers_def')
cc1: some warnings being treated as errors
scripts/Makefile.build:293: recipe for target '/home/mike/Downloads/rtl8188eu/os_dep/ioctl_linux.o' failed
make[2]: *** [/home/mike/Downloads/rtl8188eu/os_dep/ioctl_linux.o] Error 1
Makefile:1490: recipe for target '_module_/home/mike/Downloads/rtl8188eu' failed
make[1]: *** [_module_/home/mike/Downloads/rtl8188eu] Error 2
make[1]: Leaving directory '/home/mike/peta_test/Petalinux-Cora-Z7-10/Cora-Z7-10/build/tmp/work-shared/plnx_arm/kernel-source'
Makefile:155: recipe for target 'modules' failed
make: *** [modules] Error 2

[errs.txt](https://github.com/lwfinger/rtl8188eu/files/4973931/errs.txt)

lwfinger, 2020-07-24:
The fallthrough warnings can be ignored. The errors are because you have not enabled WIRELESS_EXT in your kernel. Actually, you need
CONFIG_WIRELESS_EXT=y
CONFIG_WEXT_CORE=y
CONFIG_WEXT_PROC=y
CONFIG_WEXT_PRIV=y

If you switch to the v5.2.2.4, the driver will use cfg80211 and not need the wireless extensions.

closed 2020-07-24 by lwfinger

mikeboich, 2020-07-24:
Thanks for the quick reply!   Switching kernels is a bit above my pay-grade, as the board support package from the board manufacturer only supports Petalinux 2017_4, which in turn only supports kernel 4.9.0.  I'll do the config changes you supplied and give it a go.

salmanfarisvp, 2022-01-08:
Hi, I'm using the Kria KV260 kit and using the TL-WN725N V3 WiFi module, I'm also getting the same error, can you please point me how can I solve this error? 

Linux version

`Linux kria 5.4.0-1017-xilinx-zynqmp #20-Ubuntu SMP Wed Nov 3 09:57:02 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux`

SajedeNick1999, 2022-02-13:
hello @mikeboich 
can you explain this step more?

> When I point KSRC to the kernel source (after doing make defconfig and make prepare, it almost builds

or give me reference how can I use kernel source?