lwfinger/rtl8188eu #336: not able to make file

chetak123, 2020-08-20:
when ever i try to do sudo make (or  $sudo make clean | make ) it gives me this error. What is the meaning of this error and how can i resolve it?

/rtl8188eu/os_dep/os_intfs.c:701:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]

  .ndo_select_queue = rtw_select_queue,
                      ^~~~~~~~~~~~~~~~

Full output after make command:
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.4.0-42-generic/build M=/home/ayushman/Documents/class_shell/rtl8188eu  modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-42-generic'
  CC [M]  /home/ayushman/Documents/class_shell/rtl8188eu/core/rtw_ap.o
In file included from /home/ayushman/Documents/class_shell/rtl8188eu/include/rtw_ht.h:24:0,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/include/drv_types.h:36,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/core/rtw_ap.c:23:
/home/ayushman/Documents/class_shell/rtl8188eu/include/wifi.h:733:0: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40
 
In file included from /home/ayushman/Documents/class_shell/rtl8188eu/include/wifi.h:23:0,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/include/rtw_ht.h:24,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/include/drv_types.h:36,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/core/rtw_ap.c:23:
./include/linux/ieee80211.h:1460:0: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100
 
  CC [M]  /home/ayushman/Documents/class_shell/rtl8188eu/core/rtw_br_ext.o
In file included from /home/ayushman/Documents/class_shell/rtl8188eu/include/rtw_ht.h:24:0,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/include/drv_types.h:36,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/core/rtw_br_ext.c:29:
/home/ayushman/Documents/class_shell/rtl8188eu/include/wifi.h:733:0: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40
 
In file included from /home/ayushman/Documents/class_shell/rtl8188eu/include/wifi.h:23:0,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/include/rtw_ht.h:24,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/include/drv_types.h:36,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/core/rtw_br_ext.c:29:
./include/linux/ieee80211.h:1460:0: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100
 
  CC [M]  /home/ayushman/Documents/class_shell/rtl8188eu/core/rtw_cmd.o
In file included from /home/ayushman/Documents/class_shell/rtl8188eu/include/rtw_ht.h:24:0,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/include/drv_types.h:36,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/core/rtw_cmd.c:23:
/home/ayushman/Documents/class_shell/rtl8188eu/include/wifi.h:733:0: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40
 
In file included from /home/ayushman/Documents/class_shell/rtl8188eu/include/wifi.h:23:0,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/include/rtw_ht.h:24,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/include/drv_types.h:36,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/core/rtw_cmd.c:23:
./include/linux/ieee80211.h:1460:0: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100
 
  CC [M]  /home/ayushman/Documents/class_shell/rtl8188eu/core/rtw_debug.o
In file included from /home/ayushman/Documents/class_shell/rtl8188eu/include/rtw_ht.h:24:0,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/include/drv_types.h:36,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/include/rtw_debug.h:24,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/core/rtw_debug.c:22:
/home/ayushman/Documents/class_shell/rtl8188eu/include/wifi.h:733:0: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40
 
In file included from /home/ayushman/Documents/class_shell/rtl8188eu/include/wifi.h:23:0,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/include/rtw_ht.h:24,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/include/drv_types.h:36,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/include/rtw_debug.h:24,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/core/rtw_debug.c:22:
./include/linux/ieee80211.h:1460:0: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100
 
  CC [M]  /home/ayushman/Documents/class_shell/rtl8188eu/core/rtw_efuse.o
In file included from /home/ayushman/Documents/class_shell/rtl8188eu/include/rtw_ht.h:24:0,
[... 768 lines cut ...]
In file included from /home/ayushman/Documents/class_shell/rtl8188eu/include/rtw_ht.h:24:0,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/include/drv_types.h:36,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/os_dep/os_intfs.c:24:
/home/ayushman/Documents/class_shell/rtl8188eu/include/wifi.h:733:0: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40
 
In file included from /home/ayushman/Documents/class_shell/rtl8188eu/include/wifi.h:23:0,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/include/rtw_ht.h:24,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/include/drv_types.h:36,
                 from /home/ayushman/Documents/class_shell/rtl8188eu/os_dep/os_intfs.c:24:
./include/linux/ieee80211.h:1460:0: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100
 
/home/ayushman/Documents/class_shell/rtl8188eu/os_dep/os_intfs.c:701:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .ndo_select_queue = rtw_select_queue,
                      ^~~~~~~~~~~~~~~~
/home/ayushman/Documents/class_shell/rtl8188eu/os_dep/os_intfs.c:701:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
cc1: some warnings being treated as errors
scripts/Makefile.build:273: recipe for target '/home/ayushman/Documents/class_shell/rtl8188eu/os_dep/os_intfs.o' failed
make[2]: *** [/home/ayushman/Documents/class_shell/rtl8188eu/os_dep/os_intfs.o] Error 1
Makefile:1731: recipe for target '/home/ayushman/Documents/class_shell/rtl8188eu' failed
make[1]: *** [/home/ayushman/Documents/class_shell/rtl8188eu] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-42-generic'
Makefile:151: recipe for target 'modules' failed
make: *** [modules] Error 2

lwfinger, 2020-08-20:
On my system, both the master and v5.2.2.4 branches build cleanly.
Have you done a 'git pull'? I think your source is stale.
You should never need to use sudo for make, only for 'sudo make install'.

bekhub, 2020-08-24:
seems like I have the same issue
`scripts/Makefile.build:273: recipe for target '/home/neon/wifi/rtl8188eu/core/rtw_ap.o' failed
make[2]: *** [/home/neon/wifi/rtl8188eu/core/rtw_ap.o] Error 1
Makefile:1731: recipe for target '/home/neon/wifi/rtl8188eu' failed
make[1]: *** [/home/neon/wifi/rtl8188eu] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-42-generic'
Makefile:155: recipe for target 'modules' failed
make: *** [modules] Error 2`

kelebek333, 2020-08-24:
You can try to install from PPA (supported Ubuntu 16.04~20.04 bases/Linux Mint 18.x~20.x series) .

```
sudo add-apt-repository ppa:kelebek333/kablosuz
sudo apt-get update
sudo apt install  rtl8188eu-dkms
```

kelebek333, 2020-08-24:
You can try to install from PPA (supported Ubuntu 16.04 ~ 20.04 bases / Linux Mint 18.x ~ 20.x series) .

```
sudo add-apt-repository ppa:kelebek333/kablosuz
sudo apt-get update
sudo apt install  rtl8188eu-dkms
```

bekhub, 2020-08-25:
@kelebek333 
I installed it, but i don't know how to make it work. After installing my wifi adapter didn't start. What should i do after installing?

kelebek333, 2020-08-25:
Are you post outputs of following commands?

```
dkms status
modinfo 8188eu
sudo modprobe 8188eu
mokutil --sb-state
```

bekhub, 2020-08-25:
dkms status-> rtl8188eu, 5.2.2.4~ppa: added
---------------------------------------------------------------------------------------------------------
modinfo 8188eu-> modinfo: ERROR: Module 8188eu not found.
---------------------------------------------------------------------------------------------------------
sudo modprobe 8188eu-> modprobe: FATAL: Module 8188eu not found in directory /lib/modules/5.4.0-42-generic
---------------------------------------------------------------------------------------------------------
dpkg -l | grep rtl8188eu-> ii  rtl8188eu-dkms                                5.2.2.4~bionic                                   all          rtl8188eu driver in DKMS format.
---------------------------------------------------------------------------------------------------------
mokutil --sb-state-> EFI variables are not supported on this system
---------------------------------------------------------------------------------------------------------

kelebek333, 2020-08-25:
Driver added by DKMS but not installed. Can you post outputs of following command?

`sudo dkms install rtl8188eu/5.2.2.4~ppa`

_Note: I succesfully install rtl8188eu driver from PPA to Linux Mint 19.3 (based Ubuntu 18.04.x) and Linux Mint 20 (based Ubuntu 20.04.x)._

bekhub, 2020-08-26:
`sudo dkms install rtl8188eu/5.2.2.4~ppa`:
```
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.4.0-42-generic.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.0-42-generic (x86_64)
Consult /var/lib/dkms/rtl8188eu/5.2.2.4~ppa/build/make.log for more information.
```
`dpkg -l | grep linux-headers-generic`:
`ii  linux-headers-generic-hwe-18.04               5.4.0.42.46~18.04.35                             amd64        Generic Linux kernel headers`

bekhub, 2020-08-26:
I think the problem is due to 5.4.0-42-generic kernel

kelebek333, 2020-08-26:
I downloaded Ubuntu 18.04.5 and I can succesfully installed rtl8188eu driver on Live ISO.

You can try install build-essential package before compiling driver.

`sudo apt install build-essential`

After you can re-try to compile driver.

`sudo dkms install rtl8188eu/5.2.2.4~ppa`

If compiling issue continue, you can share the output of the command below.

`cat /var/lib/dkms/rtl8188eu/5.2.2.4~ppa/build/make.log`

bekhub, 2020-08-28:
`` cat /var/lib/dkms/rtl8188eu/5.2.2.4~ppa/build/make.log `` :

```
./arch/x86/include/asm/bug.h:35:22: error: expected identifier or ‘(’ before string constant
  asm_inline volatile("1:\t" ins "\n"    \
                      ^
./arch/x86/include/asm/bug.h:73:2: note: in expansion of macro ‘_BUG_FLAGS’
  _BUG_FLAGS(ASM_UD2, 0);     \
  ^
./include/asm-generic/bug.h:62:57: note: in expansion of macro ‘BUG’
 #define BUG_ON(condition) do { if (unlikely(condition)) BUG(); } while (0)
                                                         ^
./include/net/cfg80211.h:5000:2: note: in expansion of macro ‘BUG_ON’
  BUG_ON(!wdev);
  ^
In file included from <command-line>:0:0:
././include/linux/compiler_types.h:210:24: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
 #define asm_inline asm __inline
                        ^
./arch/x86/include/asm/bug.h:35:2: note: in expansion of macro ‘asm_inline’
  asm_inline volatile("1:\t" ins "\n"    \
  ^
./arch/x86/include/asm/bug.h:73:2: note: in expansion of macro ‘_BUG_FLAGS’
  _BUG_FLAGS(ASM_UD2, 0);     \
  ^
./include/asm-generic/bug.h:62:57: note: in expansion of macro ‘BUG’
 #define BUG_ON(condition) do { if (unlikely(condition)) BUG(); } while (0)
                                                         ^
./include/net/cfg80211.h:5000:2: note: in expansion of macro ‘BUG_ON’
  BUG_ON(!wdev);
  ^
./include/net/cfg80211.h: In function ‘cfg80211_find_ie_match’:
././include/linux/compiler_types.h:210:24: error: expected ‘(’ before ‘__inline’
 #define asm_inline asm __inline
                        ^
./arch/x86/include/asm/bug.h:35:2: note: in expansion of macro ‘asm_inline’
  asm_inline volatile("1:\t" ins "\n"    \
  ^
./arch/x86/include/asm/bug.h:79:2: note: in expansion of macro ‘_BUG_FLAGS’
  _BUG_FLAGS(ASM_UD2, BUGFLAG_WARNING|(flags));  \
  ^
./include/asm-generic/bug.h:90:19: note: in expansion of macro ‘__WARN_FLAGS’
 #define __WARN()  __WARN_FLAGS(BUGFLAG_TAINT(TAINT_WARN))
                   ^
./include/asm-generic/bug.h:115:3: note: in expansion of macro ‘__WARN’
   __WARN();      \
   ^
./include/net/cfg80211.h:5301:6: note: in expansion of macro ‘WARN_ON’
  if (WARN_ON((match_len && match_offset < 2) ||
      ^
In file included from ./include/linux/bug.h:5:0,
                 from ./include/linux/thread_info.h:12,
                 from ./arch/x86/include/asm/preempt.h:7,
                 from ./include/linux/preempt.h:78,
                 from ./include/linux/spinlock.h:51,
                 from ./include/linux/seqlock.h:36,
                 from ./include/linux/time.h:6,
                 from ./include/linux/stat.h:19,
                 from ./include/linux/module.h:10,
                 from /var/lib/dkms/rtl8188eu/5.2.2.4~ppa/build/include/basic_types.h:13,
                 from /var/lib/dkms/rtl8188eu/5.2.2.4~ppa/build/include/drv_types.h:15,
                 from /var/lib/dkms/rtl8188eu/5.2.2.4~ppa/build/core/rtw_cmd.c:6:
./arch/x86/include/asm/bug.h:35:22: error: expected identifier or ‘(’ before string constant
  asm_inline volatile("1:\t" ins "\n"    \
                      ^
./arch/x86/include/asm/bug.h:79:2: note: in expansion of macro ‘_BUG_FLAGS’
  _BUG_FLAGS(ASM_UD2, BUGFLAG_WARNING|(flags));  \
  ^
./include/asm-generic/bug.h:90:19: note: in expansion of macro ‘__WARN_FLAGS’
 #define __WARN()  __WARN_FLAGS(BUGFLAG_TAINT(TAINT_WARN))
[... 509 lines cut ...]
  asm_inline volatile("1:\t" ins "\n"    \
  ^
./arch/x86/include/asm/bug.h:79:2: note: in expansion of macro ‘_BUG_FLAGS’
  _BUG_FLAGS(ASM_UD2, BUGFLAG_WARNING|(flags));  \
  ^
./include/asm-generic/bug.h:90:19: note: in expansion of macro ‘__WARN_FLAGS’
 #define __WARN()  __WARN_FLAGS(BUGFLAG_TAINT(TAINT_WARN))
                   ^
./include/asm-generic/bug.h:115:3: note: in expansion of macro ‘__WARN’
   __WARN();      \
   ^
/var/lib/dkms/rtl8188eu/5.2.2.4~ppa/build/include/osdep_service.h:305:32: note: in expansion of macro ‘WARN_ON’
 #define rtw_warn_on(condition) WARN_ON(condition)
                                ^
/var/lib/dkms/rtl8188eu/5.2.2.4~ppa/build/core/rtw_cmd.c:4445:4: note: in expansion of macro ‘rtw_warn_on’
    rtw_warn_on(1);
    ^
scripts/Makefile.build:273: recipe for target '/var/lib/dkms/rtl8188eu/5.2.2.4~ppa/build/core/rtw_cmd.o' failed
make[2]: *** [/var/lib/dkms/rtl8188eu/5.2.2.4~ppa/build/core/rtw_cmd.o] Error 1
Makefile:1731: recipe for target '/var/lib/dkms/rtl8188eu/5.2.2.4~ppa/build' failed
make[1]: *** [/var/lib/dkms/rtl8188eu/5.2.2.4~ppa/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-42-generic'
Makefile:482: recipe for target 'modules' failed
make: *** [modules] Error 2
```

lwfinger, 2020-08-28:
Your compiler or kernel headers are completely screwed up. Those errors do not come from the source for this driver.

bekhub, 2020-08-29:
@lwfinger @kelebek333 thank you very much, I updated my kde neon to latest version and driver started to work

lwfinger, 2020-08-29:
It was a distro problem, as I expected.

closed 2020-08-29 by lwfinger