lwfinger/rtl8188eu #338: Cannot compile driver due to error: redefinition of 'struct sha256_state'

dedifferentiator, 2020-08-29:
Cannot compile driver due to `error: redefinition of 'struct sha256_state'`

Building on master db29c33e180e630280d74c629473e4297b6af7ad
Compiler version: gcc (GCC) 10.2.0
Error: 
```
$ make all
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.8.5-arch1-1/build M=/home/bj0rr/utils/rtl8188eu  modules
make[1]: Entering directory '/usr/lib/modules/5.8.5-arch1-1/build'
  CC [M]  /home/bj0rr/utils/rtl8188eu/core/rtw_br_ext.o
In file included from /home/bj0rr/utils/rtl8188eu/include/drv_types.h:43,
                 from /home/bj0rr/utils/rtl8188eu/core/rtw_br_ext.c:29:
/home/bj0rr/utils/rtl8188eu/include/rtw_security.h:171:8: error: redefinition of 'struct sha256_state'
  171 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from ./include/linux/filter.h:23,
                 from ./include/net/sock.h:59,
                 from ./include/net/inet_sock.h:22,
                 from ./include/net/ip.h:27,
                 from /home/bj0rr/utils/rtl8188eu/core/rtw_br_ext.c:23:
./include/crypto/sha.h:84:8: note: originally defined here
   84 | struct sha256_state {
      |        ^~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:281: /home/bj0rr/utils/rtl8188eu/core/rtw_br_ext.o] Error 1
make[1]: *** [Makefile:1756: /home/bj0rr/utils/rtl8188eu] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.8.5-arch1-1/build'
make: *** [Makefile:155: modules] Error 2
```

But at the same time driver compiles fine and works on [v5.2.2.4](https://github.com/lwfinger/rtl8188eu/tree/v5.2.2.4)

dedifferentiator, 2020-08-29:
Also it doesn't compile on commits, which were compiling before, like something around here 1ea03fb6dea3d412d063c69e0dcd42e99ff6260b, cannot say commit-hash for sure. Maybe it's something with gcc, I have updated it recently, from 10.1.0 I guess.

lwfinger, 2020-08-29:
It was a new kernel, not a new compiler. I pushed the fix.

dedifferentiator, 2020-08-29:
@lwfinger Thanks, works as intended!

closed 2020-08-29 by dedifferentiator