lwfinger/rtl8188eu #339: Issue with kernel 4.15.0-115 (x86_64, Ubuntu)

rostyslav80, 2020-09-02:
On a kernel 4.15.0-111 a dongle works well but after Ubuntu update it has stopped working (WiFi adapter is not found). When I run `lsmod | egrep "8188|8xxxu"` on the kernel 4.15.0-111, I get  `8188eu                729088  0`, but the same command on kernel 4.15.0-112 and kernel 4.15.0-115 shows nothing. 

This is what I'm using:
```
~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.5 LTS
Release:	18.04
Codename:	bionic
```
```
~$ uname -a
Linux username 4.15.0-115-generic #112-Ubuntu SMP Thu Jul 9 20:32:34 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```
```
~$ lsusb
Bus 002 Device 002: ID 8087:8000 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
Bus 003 Device 004: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 003 Device 003: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 003 Device 002: ID 046d:0a44 Logitech, Inc. Headset H390
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Content of `/etc/modprobe.d/50-rtl8188eu.conf` is `blacklist r8188eu`.
The driver is compiled and installed from the `master` branch.

rostyslav80, 2020-09-02:
It looks as if I once did a `sudo make all`. I fixed my issue with the following commands:
```
sudo make clean
make all
sudo make install
```

closed 2020-09-02 by rostyslav80

lwfinger, 2020-09-02:
With an external driver, YOU WILL NEED TO REINSTALL THE DRIVER AFTER EVERY KERNEL CHANGE. There are no exceptions!!!!