lwfinger/rtl8188eu #342: TL-WN725N activity LED indicator doesn't work

rostyslav80, 2020-09-15:
The activity LED indicator doesn't work at all on the `v5.2.2.4` branch. When I'm switching to the `master` branch, compile and install the driver, the indicator starts flashing.

MrEngman, 2020-09-15:
You probably need to uncomment lline 151 in file include/autoconf.h

/* #define CONFIG_LED */    <<<<<<======
#ifdef CONFIG_LED
	#define CONFIG_SW_LED
	#ifdef CONFIG_SW_LED
		/* #define CONFIG_LED_HANDLED_BY_CMD_THREAD */
	#endif
#endif /* CONFIG_LED */

MrEngman

rostyslav80, 2020-09-18:
@MrEngman No, after uncommenting of the lline 151 in file include/autoconf.h the status LED just lights all time and doesn't blinking.

@lwfinger could you explain why the LED is disabled by default on this branch and how can I enable it?

lwfinger, 2020-09-18:
It is disabled in this branch because it doesn't work.

You would need to debug what is happening in the LED routines. My 8188eu does not have any LEDS, thus I cannot tell if it works or not.

dubhater, 2022-12-19:
A simple alternative to the disabled LED code is hardware-controlled blinking. You just need to set bits 1 and 5 of REG_LEDCFG2. Then the chip will make the LED blink as needed. This works with my TP-Link TL-WN725N.

rostyslav80, 2022-12-20:
@dubhater Could you please explain exactly where and how to make these changes?

dubhater, 2022-12-20:
Like this, for example:
```diff
diff --git a/usb_halinit.c b/usb_halinit.c
index 3e037e5..1fab065 100644
--- a/usb_halinit.c
+++ b/usb_halinit.c
@@ -1339,6 +1339,12 @@ static u32 rtl8188eu_hal_init(PADAPTER Adapter)
 	_InitHWLed(Adapter);
 #endif /* CONFIG_LED */
 
+	if (Adapter->registrypriv.led_enable) {
+		value8 = rtw_read8(Adapter, REG_LEDCFG2);
+		value8 |= BIT(5) | BIT(1);
+		rtw_write8(Adapter, REG_LEDCFG2, value8);
+	}
+
 	/*  */
 	/* Joseph Note: Keep RfRegChnlVal for later use. */
 	/*  */
```

The proper way would be to put those lines into _InitHWLed() where the "to do ...." comment is, but then you also have to make sure  CONFIG_LED is defined and LedStrategy is HW_LED.

rostyslav80, 2022-12-20:
@dubhater Thanks, it works.

lwfinger, 2022-12-20:
I have added this change to the source. Anyone that does not like the change will need to load the module  with the option led_enable=0.

closed 2022-12-20 by lwfinger