lwfinger/rtl8188eu #343: rtl8188eu doesn´t work on ubuntu

Aicios, 2020-10-14:
Hi lwfinger, hoping no to bother you, but I've been trying to configure an usb wifi adapter chip rtl8188eu for a month, however I haven`t been able to achieve it.  I've done everything you post on read me, and eve-n in some other forums about this driver, clone the git, install headers, compile it by make and install, copy the firmware, create the 80-wifi.conf file, tried the dkms way, try the rtl8188fu instead, try on different computers and even distributions, ubuntu 20.04, ubuntu 18.04, linux mint 19.1, linux mint 20.4 and is still doesnt worked, could you help me fix this? the system can recognize the adapter its connect, however it doesn´t worked, to be honest, I don´t know what else to do . when I put lsmod | grep 8188 , it shows 
8188eu             729088      0

Aicios, 2020-10-14:
With iwconfig, I get

wlx00e04c818802  unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Auto  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=0/100  Signal level=0 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

enp5s0    no wireless extensions.

lo        no wireless extensions.

Aicios, 2020-10-14:
I've tried to turn it on with 
Assus-PC:~$ sudo ifconfig wlx00e04c818802 up
SIOCSIFFLAGS: Operación no permitida

In other forum, I also found this option 
Assus-PC:~$ sudo iw dev wlx00e04c818802 set type managed
[sudo] contraseña:
command failed: No such device (-19)

But as you can see. none of em worked, any suggestions?

lwfinger, 2020-10-14:
iwconfig is obsolete. That name looks really strange. What does 'iw dev' show?

Aicios, 2020-10-14:
Nothing
 Assus-PC:~$ iw dev
Assus-PC:~$

lwfinger, 2020-10-14:
In that case, you need to read the output of 'dmesg | less' to see why the wireless device is not being created.