lwfinger/rtl8188eu #344: Immediate disconnect in AP mode: reason codes 4 and 8

G-S-E, 2020-10-24:
Hello.

I am using connman and driver from branch v.4.1. I run:
Linux 131F204C5254324B 4.19.78-linux4sam-6.2-00073-geb5797c91e44 #37 PREEMPT Sat Oct 24 08:39:48 MSK 2020 armv7l armv7l armv7l GNU/Linux
~# wpa_supplicant -v
wpa_supplicant v2.7
~# connmand -v
1.36

When I am trying to connect a client to the AP which I run on this device I get immediate client disconnects with reason 4 and 8. Adding rtw_ips_mode=0 did not help.

```
2020-10-24T17:21:56.574261+00:00 avahi-daemon[230]: [info] Joining mDNS multicast group on interface wlan0.IPv6 with address fe80::5
ef3:70ff:fe5d:32f5.
2020-10-24T17:21:56.575934+00:00 avahi-daemon[230]: [info] New relevant interface wlan0.IPv6 for mDNS.
2020-10-24T17:21:56.577477+00:00 systemd-networkd[165]: [info] wlan0: Gained IPv6LL
2020-10-24T17:21:56.578764+00:00 avahi-daemon[230]: [info] Registering new address record for fe80::5ef3:70ff:fe5d:32f5 on wlan0.*.
2020-10-24T17:21:58.074979+00:00 rsyslogd: [warning] omelasticsearch: checkConn failed after 1 attempts. [v8.1903.0 try https://www.
rsyslog.com/e/2007 ]
2020-10-24T17:22:05.150912+00:00 kernel: [warning] R8188EU: set group key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4) keyid:1
2020-10-24T17:22:05.159839+00:00 kernel: [warning] R8188EU: send eapol packet
2020-10-24T17:22:05.165375+00:00 wpa_supplicant[351]: [info] wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
2020-10-24T17:22:05.174882+00:00 kernel: [warning] R8188EU: send eapol packet
2020-10-24T17:22:05.189899+00:00 kernel: [warning] R8188EU: set pairwise key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4) camid:4
2020-10-24T17:22:05.196432+00:00 wpa_supplicant[351]: [info] wlan0: AP-STA-CONNECTED a0:a4:c5:22:51:1e
2020-10-24T17:22:05.762967+00:00 kernel: [warning] R8188EU: ap recv deauth reason code(4) sta:a0:a4:c5:22:51:1e
2020-10-24T17:22:05.763120+00:00 kernel: [warning] R8188EU: set pairwise key to hw: alg:0(WEP40-1 WEP104-5 TKIP-2 AES-4) camid:4
2020-10-24T17:22:05.767331+00:00 wpa_supplicant[351]: [info] wlan0: AP-STA-DISCONNECTED a0:a4:c5:22:51:1e
2020-10-24T17:22:10.500087+00:00 rsyslogd: [warning] omelasticsearch: checkConn failed after 1 attempts. [v8.1903.0 try https://www.
rsyslog.com/e/2007 ]
2020-10-24T17:22:11.477634+00:00 kernel: [warning] R8188EU: send eapol packet
2020-10-24T17:22:11.482686+00:00 wpa_supplicant[351]: [info] wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
2020-10-24T17:22:11.507655+00:00 kernel: [warning] R8188EU: send eapol packet
2020-10-24T17:22:11.532691+00:00 kernel: [warning] R8188EU: set pairwise key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4) camid:4
2020-10-24T17:22:11.542514+00:00 wpa_supplicant[351]: [info] wlan0: AP-STA-CONNECTED a0:a4:c5:22:51:1e
2020-10-24T17:22:11.958256+00:00 connmand[828]: [info] DHCP server: Received DISCOVER
2020-10-24T17:22:11.960387+00:00 connmand[828]: [info] DHCP server: find yiaddr 33597632
2020-10-24T17:22:11.961979+00:00 connmand[828]: [info] DHCP server: lease_mac (nil) lease_nip (nil)
2020-10-24T17:22:11.963270+00:00 connmand[828]: [info] DHCP server: Sending OFFER of 192.168.0.2
2020-10-24T17:22:11.964599+00:00 connmand[828]: [info] DHCP server: Broadcasting packet to client
2020-10-24T17:22:11.969965+00:00 connmand[828]: [info] DHCP server: Received REQUEST NIP -1062731774
2020-10-24T17:22:11.971441+00:00 connmand[828]: [info] DHCP server: Sending ACK
2020-10-24T17:22:11.974020+00:00 connmand[828]: [info] DHCP server: Sending ACK to 192.168.0.2
2020-10-24T17:22:11.975324+00:00 connmand[828]: [info] DHCP server: Broadcasting packet to client
2020-10-24T17:22:11.976636+00:00 connmand[828]: [info] DHCP server: lease_mac 0x5976d8 lease_nip 0x5976d8
2020-10-24T17:22:12.080767+00:00 kernel: [warning] R8188EU: ap recv deauth reason code(4) sta:a0:a4:c5:22:51:1e
2020-10-24T17:22:12.080914+00:00 kernel: [warning] R8188EU: set pairwise key to hw: alg:0(WEP40-1 WEP104-5 TKIP-2 AES-4) camid:4
2020-10-24T17:22:12.091399+00:00 wpa_supplicant[351]: [info] wlan0: AP-STA-DISCONNECTED a0:a4:c5:22:51:1e
2020-10-24T17:22:30.389996+00:00 kernel: [warning] R8188EU: send eapol packet
2020-10-24T17:22:30.395059+00:00 wpa_supplicant[351]: [info] wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
2020-10-24T17:22:30.403011+00:00 kernel: [warning] R8188EU: send eapol packet
2020-10-24T17:22:30.416073+00:00 kernel: [warning] R8188EU: set pairwise key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4) camid:4
2020-10-24T17:22:30.424317+00:00 wpa_supplicant[351]: [info] wlan0: AP-STA-CONNECTED 76:3c:fe:06:22:94
2020-10-24T17:22:30.703261+00:00 connmand[828]: [info] DHCP server: Received REQUEST NIP -1062731773
2020-10-24T17:22:30.705465+00:00 connmand[828]: [info] DHCP server: Sending NAK
2020-10-24T17:22:30.707367+00:00 connmand[828]: [info] DHCP server: Sending NAK
2020-10-24T17:22:31.862820+00:00 connmand[828]: [info] DHCP server: Received REQUEST NIP -1062731773
2020-10-24T17:22:31.865168+00:00 connmand[828]: [info] DHCP server: Sending NAK
2020-10-24T17:22:31.866527+00:00 connmand[828]: [info] DHCP server: Sending NAK
2020-10-24T17:22:31.884534+00:00 connmand[828]: [info] DHCP server: Received DISCOVER
2020-10-24T17:22:31.887489+00:00 connmand[828]: [info] DHCP server: find yiaddr 50374848
2020-10-24T17:22:31.888873+00:00 connmand[828]: [info] DHCP server: lease_mac (nil) lease_nip (nil)
2020-10-24T17:22:31.891808+00:00 connmand[828]: [info] DHCP server: Sending OFFER of 192.168.0.3
2020-10-24T17:22:31.894193+00:00 connmand[828]: [info] DHCP server: Broadcasting packet to client
2020-10-24T17:22:33.454518+00:00 connmand[828]: [info] DHCP server: Received DISCOVER
2020-10-24T17:22:33.455967+00:00 connmand[828]: [info] DHCP server: find yiaddr 50374848
2020-10-24T17:22:33.457959+00:00 connmand[828]: [info] DHCP server: lease_mac 0x58cce0 lease_nip 0x58cce0
2020-10-24T17:22:33.459560+00:00 connmand[828]: [info] DHCP server: Sending OFFER of 192.168.0.3
2020-10-24T17:22:33.460860+00:00 connmand[828]: [info] DHCP server: Broadcasting packet to client
2020-10-24T17:22:33.633386+00:00 kernel: [warning] R8188EU: ap recv disassoc reason code(8) sta:76:3c:fe:06:22:94
2020-10-24T17:22:33.633538+00:00 kernel: [warning] R8188EU: set pairwise key to hw: alg:0(WEP40-1 WEP104-5 TKIP-2 AES-4) camid:4
2020-10-24T17:22:33.658335+00:00 wpa_supplicant[351]: [info] wlan0: AP-STA-DISCONNECTED 76:3c:fe:06:22:94
2020-10-24T17:22:38.083414+00:00 rsyslogd: [warning] omelasticsearch: checkConn failed after 1 attempts. [v8.1903.0 try https://www.rsyslog.com/e/2007 ]
2020-10-24T17:22:50.506910+00:00 rsyslogd: [warning] omelasticsearch: checkConn failed after 1 attempts. [v8.1903.0 try https://www.rsyslog.com/e/2007 ]
```

Please help.
BTW how can I donate to the project?

G-S-E, 2020-10-24:
When I use master branch I get:
```
root@131F204C5254324B:~# Configuration file: /etc/hostapd.conf
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlan0 wasn't started
```
Client functionality works fine as with the v.4.1. branch.

G-S-E, 2020-10-25:
Also I get this while compilling branch 'v4.1.8_9499':
```
/home/sergei/git/rtl8188eu/os_dep/ioctl_cfg80211.c: In function ‘rtw_cfg80211_inform_bss’:
/home/sergei/git/rtl8188eu/os_dep/ioctl_cfg80211.c:502:1: warning: the frame size of 1072 bytes is larger than 1024 bytes [-Wframe-larger-than=]
 }
 ^
```

G-S-E, 2020-10-26:
I also tried version from v5.2 branch but wpa_supplicant returned that AP mode is not supported by driver.

G-S-E, 2020-10-26:
When try to enable AP I get the following:
```
RTW: survey done event(1c) band:0 for wlan0
RTW: rtw_indicate_scan_done(wlan0)
RTW: ==>rtw_ps_processor .fw_state(8)
RTW: ==>ips_enter cnts:2
RTW: nolinked power save enter
RTW: ===> rtw_ips_pwr_down...................
RTW: ====> rtw_ips_dev_unload...
RTW: usb_read_port_cancel
RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
RTW: usb_write_port_cancel
RTW: ==> rtl8188eu_hal_deinit
RTW: bkeepfwalive(0)
RTW: SetHwReg: bMacPwrCtrlOn=0
RTW: <=== rtw_ips_pwr_down..................... in 280ms
Error enabling wifi tethering: Not supported
```

lwfinger, 2020-10-26:
I do not support AP mode or packet injection on any out the Realtek devices with drivers outside the kernel. If I want an AP, I spend the $35 to get one. If I were to contract with someone to implement this, the first hour would cost more than said AP. If you must have the AP on your laptop, then buy a USB stick from a different vendor.

closed 2020-10-26 by lwfinger

G-S-E, 2020-10-26:
Thank you for your reply.
Could you elaborate a littele bit more on: "I do not support AP mode or packet injection on any out the Realtek devices with drivers outside the kernel." please or give a link about the topic? Is it some thing different then hostapd do? I saw that this chipset is used as AP with hostapd. Is the any difference between hostapd and wpa_supplicant. I cannot buy AP because I need bring up an AP with this chipset which is already incorporated in the embedded ARM based device....
Any help will be appreciated.

G-S-E, 2020-10-26:
iw list shows me that AP is supported mode for the driver. I am confused by your reply. Please clarify a little bit.

lwfinger, 2020-10-26:
I have no knowledge of the chip internals, and only cursory understanding of the driver, which I did not write. For me to support anything that does not work "out of the box" would take a lot of time. At one time, I would have volunteered to do it, but at age 80, I am now more careful with my time. As you need to make it work, you are going to need to do the time in debugging it yourself.