lwfinger/rtl8188eu #348: device not ready

anshumeena1947, 2020-11-13:
i install your driver in my zorin os 15.3 (based of ubuntu 18.04) all process are work successfully but after installing the driver and reboot it show that "device not find" i read your FAQ The network manager says: "Device is not ready"!

Make sure you copied the firmware (rtl8188eufw.bin) to /lib/firmware/rtlwifi/ 
 i done this also but it not working it show the same problem
device not ready

lwfinger, 2020-11-16:
For me, running openSUSE Leap 15.2 and using NetworkManager, it just works. That implies something is wrong with your user-space code. Is it killed? Check 'rfkill list'.

anshumeena1947, 2020-11-27:
when i check give command to terminal "rfkill list " then there is not output in this but command work successfully and aligable to give another commands

lwfinger, 2020-11-27:
If rfkill list does not list any output, then you have no wifi or bluetooth devices registered.

Please post the output of 'lsusb' with your device plugged in.

anshumeena1947, 2020-12-01:
no when i type "lsusb" then it show
Bus 001 Device 003: ID 0bda:f179 Realtek Semiconductor Corp. 

i plug in this

lwfinger, 2020-12-02:
0bda:f179 is NOT an RTL8188EU, it is an RTL8188FU. As far as I know, there are no Linux drivers for that device. I tested the Windows drivers to see if they would work with NDISWRAPPER; however, there were two missing globals. One was easily replaced, but the other was fatal.

closed 2020-12-02 by lwfinger

Calistex, 2021-11-19:
I too have been misled to believe that 0bda:f179 was RTL8188EU, because the CD that was bundled with the WiFi adapter contained this driver, and a clean install of RTL8188FU finally made it work. Thank you for the clarification.