lwfinger/rtl8188eu #349: Ubuntu 20.10 no ssid for 8188eu usb dongle

csselo, 2020-11-19:
Hi

I installed it succesfully but there is no ssid in wifi scan.
What can I do ?

```

*-network ETKİSİZLEŞTİRİLDİ
       açıklama: Kablosuz arayüz
       physical id: 2
       bus info: usb@1:7
       mantıksal ad: wlx00e020301740
       serial: 00:e0:20:30:17:40
       kapasiteler: eternet physical wireless
       yapılandırma: broadcast=yes driver=r8188eu driverversion=5.8.0-29-generic multicast=yes wireless=unassociated
WARNING: output may be incomplete or inaccurate, you should run this program as super-user.

lo        no wireless extensions.

enp7s0    no wireless extensions.

wlx00e020301740  unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Auto  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=0/100  Signal level=0 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

[   89.276146] kernel: usbcore: registered new interface driver r8188eu
[   89.279808] kernel: r8188eu 1-8:1.0 wlx00e020301740: renamed from wlan0
[   89.300047] kernel: cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   89.300274] kernel: cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   89.316483] kernel: R8188EU: Firmware Version 11, SubVersion 1, Signature 0x88e1
[   90.706425] kernel: nouveau 0000:01:00.0: Direct firmware load for nouveau/nva8_fuc084 failed with error -2
[   90.706434] kernel: nouveau 0000:01:00.0: Direct firmware load for nouveau/nva8_fuc084d failed with error -2
[   90.706435] kernel: nouveau 0000:01:00.0: msvld: unable to load firmware data
[   90.706445] kernel: nouveau 0000:01:00.0: msvld: init failed, -19
[   92.312947] kernel: rfkill: input handler disabled
```

lwfinger, 2020-11-19:
The only thing I see in the logs is that line about rfkill. It could be blocking your connection. What does 'rfkill list' show?

csselo, 2020-11-20:
selo@selo-Z170-HD3P:~$ rfkill list
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
selo@selo-Z170-HD3P:~$

lwfinger, 2020-11-20:
Your wifi is NOT in that list.

You are probably running the "master" branch. That one uses some obsolete kernel methods that are not supported by your kernel.

Check out the v5.2.2.4 branch by using:
git checkout origin:v5.2.2.4 -b v5.2.2.4

Then rebuild and try again.

If you are using the zip format rather than git to download the driver, you are on your own. I stopped supporting that a long time ago. It was too much effort for me. Remember, I'm a cranky 80-year old. Kind of like your grandfather!

csselo, 2020-11-21:
there is only one branch (master) I tried your command.
how can I get ur other brach?

csselo, 2020-11-21:
Directory .git does not exist indicating that you downloaded the source as a zip file. Only the 'git clone' method is now supported.
make: *** [Makefile:479: test] Error 1

for zip

lwfinger, 2020-11-21:
This has been posted many, many times, but here goes for you once again!
cd ~
rm -rf rtl8188eu
git clone http://github.com/lwfinger/rtl8188eu.git -b extended
cd rtl8188eu
make
sudo make install
sudo reboot

csselo, 2020-11-22:
`selo@selo-Z170-HD3P:~/Belgeler$ git clone http://github.com/lwfinger/rtl8188eu.git -b extended
Klonlama konumu: 'rtl8188eu'...
warning: şuraya yeniden yönlendiriliyor: https://github.com/lwfinger/rtl8188eu.git/
fatal: extended uzak dalı origin üstkaynağında bulunamadı
`

origin not found

tariellaelius, 2020-11-23:
Just replace **extended** with **v5.2.2.4**

csselo, 2020-11-23:
It clones now but there is no wifi adapter in iwconfig or lsusb and also gnome.

lwfinger, 2020-11-23:
Sorry about the typo.

If you still do not see the adapter in lsusb, then how do you expect to find it in any other software? The kernel scans the hardware just like lsusb, searches for a driver that supports that USB ID, and loads it. You are failing step 1!

csselo, 2020-11-24:
that means "the driver doesn t support my kernel" ??

git clone http://github.com/lwfinger/rtl8188eu.git -b v5.2.2.4 is correct ? or how can I understand which version was downloaded?

lwfinger, 2020-11-24:
You need to read the git manual. The command you want is 'git branch'. Your kernel is supported; however, your distro may have selected configuration parameters that are not compatible.

Please post the output of the 'lsusb' command.

csselo, 2020-11-25:
correct branch * v5.2.2.4

`
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 03f0:0024 HP, Inc KU-0316 Keyboard
Bus 001 Device 003: ID 0bda:f179 Realtek Semiconductor Corp. 802.11n
Bus 001 Device 002: ID 047d:1002 Kensington Expert Mouse Pro
Bus 001 Device 011: ID 214b:7250 Huasheng Electronics 
Bus 001 Device 010: ID 214b:7250 Huasheng Electronics USB2.0 HUB
Bus 001 Device 005: ID 058f:6362 Alcor Micro Corp. Flash Card Reader/Writer
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

`

lwfinger, 2020-11-25:
The line with ID 0bda:f179 is your device.

That device is an RTL8188FU. The driver for the RTL8188EU will not work for you.

To my knowledge, Realtek has not released a Linux driver for the RTL8188FU. A few months ago, we tested a couple of options, neither worked. We also tried a Windows driver using ndiswrapper; however, it used two Windows global routines that are not implemented in ndiswrapper. As a result, there is no current solution that works.

closed 2020-11-25 by lwfinger