lwfinger/rtl8188eu #354: device not ready

alekione, 2020-12-24:
hi...i installed the driver on kali linux and it was successful. But when i check my wireless connection it is reading device not ready. 

# ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.103  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::20c:29ff:feaa:65c1  prefixlen 64  scopeid 0x20<link>
        ether 00:0c:29:aa:65:c1  txqueuelen 1000  (Ethernet)
        RX packets 35861  bytes 20302070 (19.3 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 18489  bytes 2620166 (2.4 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 12  bytes 556 (556.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 12  bytes 556 (556.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0lo        no wireless extensions.

# iwconfig

eth0      no wireless extensions.

wlan0     unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Auto  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=0/100  Signal level=0 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

# rfkill list

0: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
where is the error...im still new on kali linux

anshumeena1947, 2020-12-29:
hi, i read your problem but i don't say anything properly 
my WIFI USB stick is 0bda:f179 Realtek Semiconductor Corp. (after typing lsusb command in terminal)
and i also downlaod this driver(rtl8188eu) it install successfully but it show that device not ready
i found so many things and then i delete this driver(rtl8188eu) and install new driver([RTL8188FU](https://github.com/kelebek333/rtl8188fu)) from this [repository](https://github.com/kelebek333/rtl8188fu) i want to give a suggestion first delete this driver(rtl8188eu) by DKMS and install this driver[(RTL8188FU)](https://github.com/kelebek333/rtl8188fu)

1. then just restart your pc
2. than your problem solve

if problem is not solve then just replay i wait for your replay

lwfinger, 2020-12-29:
Driver rtl8188eu will never work on an RTL8188FU device. I have no idea if the kelebeck333 repo will work. I was not aware that any Linux drivers had been released. If it does work, then good.

closed 2020-12-29 by lwfinger

alekione, 2021-01-03:
> hi, i read your problem but i don't say anything properly
> my WIFI USB stick is 0bda:f179 Realtek Semiconductor Corp. (after typing lsusb command in terminal)
> and i also downlaod this [...]

i tried rtl8188fu and it worked..thanks