lwfinger/rtl8188eu #355: not work on linux 5.4/5.8/5.10.2

Kxuan, 2020-12-24:
hostapd report `ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported`(EOPNOTSUPP) and failed to start.

I use systemtap to trace the syscall in kernel. 

The EOPNOTSUPP is returned from `wireless_process_ioctl`.
And in `wireless_process_ioctl`, it tries to get handler using `get_handler`, but `get_handler` returned NULL. Because the follow line is a `NULL` value.
https://github.com/lwfinger/rtl8188eu/blob/b02c92bc92e22fd8b51c9acd6d602637cfce7256/os_dep/ioctl_linux.c#L7782

This looks like a bug. Could you please fix it?

lwfinger, 2020-12-24:
It is not a bug. The feature is deliberately missing. What branch are you using?

Kxuan, 2020-12-25:
> It is not a bug. The feature is deliberately missing. What branch are you using?

Thank you for your quick reply.

I used the `master` yesterday.

But I replaced a new device and used the `v5.2.2.4` branch. Now it works.

Thanks for your help!

closed 2020-12-25 by Kxuan