lwfinger/rtl8188eu #356: cannot see wifi-ssids below is the dmesg result

openOrganizationPakistan, 2020-12-27:
[   78.470301] =======================================================
[   78.470301] ==== Launching Wi-Fi driver! (Powered by Rockchip) ====
[   78.470302] =======================================================
[   78.470302] Realtek 8188FU USB WiFi driver (Powered by Rockchip) init.
[   78.470302] RTL871X: module init start
[   78.470303] RTL871X: rtl8188fu v4.3.23.6_20964.20170110
[   78.470346] RTL871X: 
               usb_endpoint_descriptor(0):
[   78.470347] RTL871X: bLength=7
[   78.470347] RTL871X: bDescriptorType=5
[   78.470347] RTL871X: bEndpointAddress=81
[   78.470348] RTL871X: wMaxPacketSize=512
[   78.470348] RTL871X: bInterval=0
[   78.470348] RTL871X: RT_usb_endpoint_is_bulk_in = 1
[   78.470349] RTL871X: 
               usb_endpoint_descriptor(1):
[   78.470349] RTL871X: bLength=7
[   78.470349] RTL871X: bDescriptorType=5
[   78.470350] RTL871X: bEndpointAddress=2
[   78.470350] RTL871X: wMaxPacketSize=512
[   78.470350] RTL871X: bInterval=0
[   78.470351] RTL871X: RT_usb_endpoint_is_bulk_out = 2
[   78.470351] RTL871X: 
               usb_endpoint_descriptor(2):
[   78.470351] RTL871X: bLength=7
[   78.470352] RTL871X: bDescriptorType=5
[   78.470352] RTL871X: bEndpointAddress=3
[   78.470352] RTL871X: wMaxPacketSize=512
[   78.470353] RTL871X: bInterval=0
[   78.470353] RTL871X: RT_usb_endpoint_is_bulk_out = 3
[   78.470353] RTL871X: nr_endpoint=3, in_num=1, out_num=2

[   78.470354] RTL871X: USB_SPEED_HIGH
[   78.470354] RTL871X: CHIP TYPE: RTL8188FU
[   78.470608] RTL871X: rtw_hal_config_rftype RF_Type is 3 TotalTxPath is 1 
[   78.470610] RTL871X: Chip Version Info: CHIP_8188F_Normal_Chip_SMIC_B_CUT_1T1R_RomVer(0)
[   78.470611] RTL871X: _ConfigChipOutEP_8188F OutEpQueueSel(0x05), OutEpNumber(2)
[   78.470816] RTL871X: EEPROM type is E-FUSE
[   78.471185] RTL871X: Boot from EFUSE, Autoload OK !
[   78.471733] RTL871X: hal_EfuseSwitchToBank: Efuse switch bank to 0
[   78.600074] RTL871X: hal_ReadEFuse_WiFi: data end at address=0x68
[   78.600194] RTL871X: EEPROM VID = 0x bda
[   78.600195] RTL871X: EEPROM PID = 0xf179
[   78.600199] RTL871X: hal_com_config_channel_plan chplan:0x20
[   78.600201] RTL871X: Hal_EfuseParsePowerSavingMode_8188F...bHWPwrPindetect(0)-bHWPowerdown(0) ,bSupportRemoteWakeup(1)
[   78.600202] RTL871X: ### PS params=>  power_mgnt(0),usbss_enable(0) ###
[   78.602522] RTL871X: kfree Pwr Trim flag:1
[   78.602523] RTL871X: bb_gain:-1
[   79.270332] RTL871X: ERROR HalPwrSeqCmdParsing: Fail to polling Offset[0x5]=01
[   79.780743] RTL871X: reg 0x1a0, usb write 1 fail, status:-110 value=0x8ae380fd, vendorreq_times:1
[   80.284729] RTL871X: reg 0x1a0, usb write 1 fail, status:-110 value=0x8ae380fd, vendorreq_times:2
[   80.788746] RTL871X: reg 0x1a0, usb write 1 fail, status:-110 value=0x8ae380fd, vendorreq_times:3
[   81.292738] RTL871X: reg 0x1a0, usb write 1 fail, status:-110 value=0x8ae380fd, vendorreq_times:4
[   81.796751] RTL871X: reg 0x1a0, usb write 1 fail, status:-110 value=0x8ae380fd, vendorreq_times:5
[   81.796756] RTL871X: [dvobj:000000001824c07f][ERROR] continual_io_error:5 > 4
[   81.796759] RTL871X: rtl8188f_FirmwareDownload((null)) tmp_ps=0
[   81.796763] usb 2-3: request firmware rtlwifi/rtl8188fufw.bin
[   81.797631] usb 2-3: request firmware rtlwifi/rtl8188fufw.bin loaded
[   81.797634] RTL871X: rtl8188f_FirmwareDownload: fw_ver=4 fw_subver=0000 sig=0x88f1, Month=08, Date=22, Hour=17, Minute=36
[   81.797635] RTL871X: rtl8188f_FirmwareDownload(): Shift for fw header!
[   81.797636] RTL871X: rtl8188f_FirmwareDownload by IO write!
[   82.006505] RTL871X: _FWDownloadEnable: !!!!!!!!Write 0x80 Fail!: count = 101
[   82.006508] RTL871X: _8051Reset8188: Finish
[   82.006510] RTL871X: _FWFreeToGo: Polling FW ready Fail! (1, 0ms), REG_MCUFWDL:0x00000000
[   82.006512] RTL871X: rtl8188f_FirmwareDownload fail. write_fw:0, 209ms
[   82.006518] RTL871X:  <=== rtl8188f_FirmwareDownload()
[   82.006519] RTL871X: CardDisableRTL8188FU
[   82.006526] RTL871X: hal_read_mac_hidden_rpt Fail! (0, 3404ms), fwdl:0, id:0xfd
[   82.006528] RTL871X: InitAdapterVariablesByPROM_8188FU(): REPLACEMENT = 0
[   82.006529] RTL871X: rtw_hal_read_chip_info in 3536 ms
[... 228 lines cut ...]
[ 1038.157556] RTL871X: cfg80211_rtw_get_txpower
[ 1038.158131] RTL871X: cfg80211_rtw_get_txpower
[ 1038.158167] RTL871X: +871x_drv - drv_open, bup=0
[ 1039.133790] RTL871X: ERROR HalPwrSeqCmdParsing: Fail to polling Offset[0x5]=01
[ 1039.133793] RTL871X: rtl8188fu_hal_init in 976ms
[ 1039.133794] RTL871X: rtw_hal_init: hal__init fail
[ 1039.133795] RTL871X: -871x_drv - drv_open fail, bup=0
[ 1039.133903] RTL871X: cfg80211_rtw_get_txpower
[ 1039.134225] RTL871X: +871x_drv - drv_open, bup=0
[ 1040.114537] RTL871X: ERROR HalPwrSeqCmdParsing: Fail to polling Offset[0x5]=01
[ 1040.114540] RTL871X: rtl8188fu_hal_init in 980ms
[ 1040.114541] RTL871X: rtw_hal_init: hal__init fail
[ 1040.114542] RTL871X: -871x_drv - drv_open fail, bup=0
[ 1065.421409] RTL871X: +871x_drv - drv_open, bup=0
[ 1066.345964] RTL871X: ERROR HalPwrSeqCmdParsing: Fail to polling Offset[0x5]=01
[ 1066.345967] RTL871X: rtl8188fu_hal_init in 925ms
[ 1066.345968] RTL871X: rtw_hal_init: hal__init fail
[ 1066.345970] RTL871X: -871x_drv - drv_open fail, bup=0
[ 1066.347031] RTL871X: +871x_drv - drv_open, bup=0
[ 1067.126366] RTL871X: ERROR HalPwrSeqCmdParsing: Fail to polling Offset[0x5]=01
[ 1067.126369] RTL871X: rtl8188fu_hal_init in 779ms
[ 1067.126370] RTL871X: rtw_hal_init: hal__init fail
[ 1067.126372] RTL871X: -871x_drv - drv_open fail, bup=0

I'll provide any help you might need about the issue ASAP...

lwfinger, 2020-12-28:
What device do you have? Please post the output of 'lsusb'.

openOrganizationPakistan, 2020-12-29:
Sorry issue solved yesterday. I just reinstall OS and drivers and this seems to solve the issue.
My device by the way is rtl8188eu

closed 2020-12-29 by lwfinger