lwfinger/rtl8188eu #357: redefinition of 'struct iovec'

aep, 2021-01-08:
branch v5.2.2.4  60cb0b does not appear to compile against opernwrts backport

master works fine

```
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/aep/proj/captif/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include/sys/socket.h:18:0,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/include/linux/wireless.h:12,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/include/osdep_service_linux.h:37,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/include/osdep_service.h:16,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/include/drv_types.h:16,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/core/rtw_cmd.c:6:
/home/aep/proj/captif/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include/bits/alltypes.h:372:8: error: redefinition of 'struct iovec'
 struct iovec { void *iov_base; size_t iov_len; };
        ^~~~~
In file included from /home/aep/proj/captif/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport/linux/fs.h:3:0,
                 from ./arch/mips/include/asm/elf.h:12,
                 from ./include/linux/elf.h:5,
                 from ./include/linux/module.h:15,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/include/basic_types.h:13,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/include/drv_types.h:15,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/core/rtw_cmd.c:6:
./include/linux/fs.h:47:8: note: originally defined here
 struct iovec;
        ^~~~~
In file included from /home/aep/proj/captif/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include/sys/socket.h:20:0,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/include/linux/wireless.h:12,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/include/osdep_service_linux.h:37,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/include/osdep_service.h:16,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/include/drv_types.h:16,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/core/rtw_cmd.c:6:
/home/aep/proj/captif/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include/bits/socket.h:1:8: error: redefinition of 'struct msghdr'
 struct msghdr {
        ^~~~~~
In file included from /home/aep/proj/captif/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport/linux/socket.h:3:0,
                 from ./include/linux/skbuff.h:23,
                 from /home/aep/proj/captif/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport/linux/skbuff.h:3,
                 from ./include/linux/if_ether.h:23,
                 from /home/aep/proj/captif/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport/linux/if_ether.h:3,
                 from ./include/uapi/linux/ethtool.h:19,
                 from ./include/linux/ethtool.h:18,
                 from /home/aep/proj/captif/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport/linux/ethtool.h:3,
                 from ./include/linux/netdevice.h:41,
                 from /home/aep/proj/captif/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport/linux/netdevice.h:3,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/include/osdep_service_linux.h:22,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/include/osdep_service.h:16,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/include/drv_types.h:16,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/core/rtw_cmd.c:6:
./include/linux/socket.h:48:8: note: originally defined here
 struct msghdr {
        ^~~~~~
In file included from /home/aep/proj/captif/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include/sys/socket.h:20:0,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/include/linux/wireless.h:12,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/include/osdep_service_linux.h:37,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/include/osdep_service.h:16,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/include/drv_types.h:16,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/core/rtw_cmd.c:6:
/home/aep/proj/captif/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include/bits/socket.h:11:8: error: redefinition of 'struct cmsghdr'
 struct cmsghdr {
        ^~~~~~~
In file included from /home/aep/proj/captif/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport/linux/socket.h:3:0,
                 from ./include/linux/skbuff.h:23,
                 from /home/aep/proj/captif/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport/linux/skbuff.h:3,
                 from ./include/linux/if_ether.h:23,
                 from /home/aep/proj/captif/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport/linux/if_ether.h:3,
                 from ./include/uapi/linux/ethtool.h:19,
                 from ./include/linux/ethtool.h:18,
                 from /home/aep/proj/captif/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport/linux/ethtool.h:3,
                 from ./include/linux/netdevice.h:41,
                 from /home/aep/proj/captif/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport/linux/netdevice.h:3,
[... 37 lines cut ...]
/home/aep/proj/captif/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include/sys/socket.h:316:8: error: redefinition of 'struct sockaddr'
 struct sockaddr {
        ^~~~~~~~
In file included from ./include/linux/socket.h:11:0,
                 from /home/aep/proj/captif/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport/linux/socket.h:3,
                 from ./include/linux/skbuff.h:23,
                 from /home/aep/proj/captif/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport/linux/skbuff.h:3,
                 from ./include/linux/if_ether.h:23,
                 from /home/aep/proj/captif/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport/linux/if_ether.h:3,
                 from ./include/uapi/linux/ethtool.h:19,
                 from ./include/linux/ethtool.h:18,
                 from /home/aep/proj/captif/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport/linux/ethtool.h:3,
                 from ./include/linux/netdevice.h:41,
                 from /home/aep/proj/captif/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport/linux/netdevice.h:3,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/include/osdep_service_linux.h:22,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/include/osdep_service.h:16,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/include/drv_types.h:16,
                 from /home/aep/proj/captif/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/rtl8188eufw-1/core/rtw_cmd.c:6:
./include/uapi/linux/socket.h:9:46: note: originally defined here
 #define _K_SS_ALIGNSIZE (__alignof__ (struct sockaddr *))
                                              ^
./include/uapi/linux/socket.h:20:27: note: in expansion of macro '_K_SS_ALIGNSIZE'
 } __attribute__ ((aligned(_K_SS_ALIGNSIZE))); /* force desired alignment */

```

aep, 2021-01-08:
possibly related #266

closed 2021-01-08 by aep

dausruddin, 2021-01-14:
How do we actually install this driver in openwrt? I am using snapshot.

aep, 2021-01-14:
You cannot. It doesn't work

dausruddin, 2021-01-14:
Are you using any other USB adapter now? There is not a lot of people using USB adapter, so it is hard to find a recommendation.