lwfinger/rtl8188eu #358: where I can find the documentation? :)

threeturn, 2021-01-16:
Hi,

first of all, I want to thank you for the valuable work the contributors are doing, without them my USB dongle would be totally useless :)

I'm writing to raise the issue of documentation that is really lacking. I'm trying to collect info to contribute maybe a document myself that clarifies how to compile and install this module and how to use the RTL8188EU with wpa_supplicant and in AP mode with hostap.

1) from what I've seen the "official" module in the staging directory of the kernel tree shares portions of code present in this repository and some contributors but does not receive changes for just over a year while here the development is more active.  I wonder, what are the functional differences between the two modules and why don't the contributors here directly update the official repo?

2) Compiling and installing the module is definitely trivial, but then when it comes to using it with wpa_supplicant and hostap the documentation is almost non-existent or when there is it is fragmentary or outdated. Can you tell me where I can find some documentation to help me get started?

Use case is pretty simple: I have one raspberry pi 1b and I would like to create an AP.

Thank you very much.

Kind regards,
Andrea.

githubtobi, 2021-01-19:
Hi Andrea,

I just did it with this documentation on an pi 1b (the article refers this repository):
https://blog.samat.org/2014/12/15/realtek-8188eu-based-wi-fi-adapters-on-the-raspberry-pi/

Don't forget this step at the end "Make sure you copied the firmware (rtl8188eufw.bin) to /lib/firmware/rtlwifi/" and reboot.

There is another issue then with dhcpcd, see the latest post how to do a temp fix and get wifi finally working:
https://raspberrypi.stackexchange.com/questions/112062/raspbian-buster-lite-couldnt-communicate-with-wpa-supplicant

But I think this shouldn't be part of this issue. In general, it's working fine with the documentation from samat.org

wegi1, 2021-04-11:
I had information that wpa_supplicant is not good idea, its a bit buggy sometimes. If is possible try to use nmcli with nmtui which is more stabile and usefull.

lwfinger, 2021-04-11:
All software is a bit buggy sometimes; however, I have has no problems with wpa_supplicant for many years. Are you aware that NetworkManager uses wpa_supplicant to handle encryption? For me, I use NM in the GUI that openSUSE provides. It works without problem.

wegi1, 2021-04-11:
Don't remember where, probably on the askubuntu.com someone mention about this or on any YT video. My problem was do the configuration static IP for this dongle on Ubuntu server. Yes I know in the GUI on Ubuntu desktop it work flawlesly, but I just tried do it on server without success. (I'm a newbie in Linux) From nmcli especially from nmtui did it without any problem.

Using this occasion @lwfinger  I want say to you mega thanks for you great job over all of this drivers. Its really much much impressive and helpfull - THANKS