lwfinger/rtl8188eu #361: TL-WN727n problem: Direct firmware load for regulatory.db failed with error -2

Zaryob, 2021-02-18:
Hi,
I'm using TP-Link TL-WN727n adaptor. In the documentation it says that, USB WiFi dongle comes with rtl8188eus chip. I found its source code provided by Archer, it wouldn't work then I found that repository.

I have built kernel module and modprobed. It seems no problem after modprobe. However, I connected to USB and it come with that error.  I'm gonna sharing screenshot of it.

Zaryob, 2021-02-18:
[image]

Zaryob, 2021-02-18:
This is not about the RHEL. Im using various systems (Fedora, RHEL, Ubuntu) and all of them not worked this dongle as well. 

Im using also kernel-ml 5.11.0 with RHEL.

Zaryob, 2021-02-18:
Addition: ID of bus device output provied by `lsusb` is 
```
ID 2357:0111 TP-Link
```

lwfinger, 2021-02-18:
The missing file is regulatory.db which is not a part of this driver. You should be able to get that from your distro.

closed 2021-02-18 by lwfinger