lwfinger/rtl8188eu #362: Raspberry Pi 4: scanning interface, turning on hostapd, wpa_supplicant all give -19 (No such device)

ramast, 2021-02-28:
Raspberry Pi 4: scanning interface, turning on hostapd, wpa_supplicant all fail.

-19 (No such device).

Kernel doesn't load `8188eu.ko` by  default and I am experriancing same behavior whether I `modprobe 8188eu.ko` or not. I have also experienced same behavior even before installing this.

```
# lsusb 
Bus 001 Device 004: ID 2357:010c TP-Link TL-WN722N v2
```

```
# uname -a
Linux rpi4-20201112 5.9.0-0.bpo.5-arm64 #1 SMP Debian 5.9.15-1~bpo10+1 (2020-12-31) aarch64 GNU/Linux
```

```
# dmesg
[ 1320.197554] usbcore: registered new interface driver r8188eu
[ 1320.226037] r8188eu 1-1.3:1.0: firmware: direct-loading firmware rtlwifi/rtl8188eufw.bin
[ 1320.234636] R8188EU: Firmware Version 11, SubVersion 1, Signature 0x88e1
[ 1320.636909] MAC Address = d0:37:45:e8:d5:3c
```

```
# ifconfig wlan1
wlan1: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether d0:37:45:e8:d5:3c  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

```
# ifconfig wlan1 up
```

```
# iwconfig wlan1
wlan1     unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Auto  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=0/100  Signal level=0 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed
```

```
iw dev wlan1 scan
command failed: No such device (-19)
```

```
# hostapd hostapd/hostapd.conf 
Configuration file: hostapd/hostapd.conf
Line 55: DEPRECATED: 'dump_file' configuration variable is not used anymore
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan1 disabled_11b_rates=0
nl80211 driver initialization failed.
```

```
# cat /etc/debian_version 
10.8
```

```
root@rpi4-20201112:~/Downloads/rtl8188eu/hostapd-0.8# git log -1
commit b02c92bc92e22fd8b51c9acd6d602637cfce7256 (HEAD -> master, origin/master, origin/HEAD)
Merge: c928027 6672b42
Author: lwfinger <Larry.Finger@lwfinger.net>
Date:   Tue Dec 1 12:09:25 2020 -0600

    Merge pull request #350 from sim0nx/addsupportfor_edimax_EW_7811Un_v2
    
    Add support for Edimax EW-7811Un V2
```

closed 2021-02-28 by ramast

cecchisandrone, 2021-12-14:
Hey @ramast how did you solve this?

ramast, 2021-12-14:
I did! Since my last edit above

>  Edit3: Switching to branch v5.2.2.4 git checkout v5.2.2.4, then compiling and installing that have solved all my problems.

cecchisandrone, 2021-12-14:
I'm on bullseye on armv6l and I can't find linux headers, make fails with the following error:
```
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.10.83+/build M=/home/pi/rtl8188eu  modules
make[1]: *** /lib/modules/5.10.83+/build: No such file or directory.  Stop.
make: *** [Makefile:155: modules] Error 2
```
I can't find kernel headers for my kernel, how can it be?
```
sudo apt install linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-headers-5.10.83
E: Couldn't find any package by glob 'linux-headers-5.10.83'
```

ramast, 2021-12-14:
How to install Linux headers on Debian is not related to this github repo.
I don't know what is causing the problem but I'd just google "Couldn't find any package by glob 'linux-headers'"

lwfinger, 2021-12-14:
@ramast: Thank you.

ramast, 2021-12-16:
I needed to install the driver on a raspberry pi running Raspbian yesterday (previously I've used Debian) and ran into same problem u had.
I've installed `apt install raspberrypi-kernel-headers` and that did solve the problem.
I did upgrade first to debian 11.1 (bullseye) not sure if that helped or not.

cecchisandrone, 2021-12-17:
@ramast Yes, I confirm that installing headers with `apt install raspberrypi-kernel-headers` makes it. My problem was about having too recent firmware so I needed to downgrade it via `rpi-update` to match downloaded headers.