lwfinger/rtl8188eu #365: LED not blinking after installing the driver for TP-Link TL-WN722N v2.

d4rk-c1ph3r, 2021-03-14:
Monitor mode and packet injection working perfectly. Chipset is rtl8188eu

wegi1, 2021-04-11:
Did you check in any other device where LED is blinking in exactly this one dongle?
Maybe basically LED is burned?

gr33nm0nk2802, 2021-04-11:
LED works fine on windows. Also mt7601u drivers support led blinking

wegi1, 2021-04-11:
> LED works fine on windows. Also mt7601u drivers support led blinking

On my dongle working as well on the Linux and WIN10 booth 64bits.

But you can't have d4rk-c1ph3r dongle where can exist any problem like:
- LED is damaged
- LED doesn't exist
- Code version had disable support of LED blinking
- ... any other possible driver or hardware errors

That's way I'd ask d4rk-c1ph3r hi is sure in his dongle LED exist.

gr33nm0nk2802, 2021-04-11:
The problem lies i  the code itself. I have the same adapter

wegi1, 2021-04-11:
> The problem lies in the code itself. I have the same adapter .
> 
> LED working on windows hence not broken.
> LED detected first time when connected before installing driver on ubuntu

Possible. You can't be sure you have this same adapter like d4rk-c1ph3r. As lwfinger mentioned in #342 they are medels without LED.

BUT as you said in last words you have right - most probably the code is here the problem.

gr33nm0nk2802, 2021-04-11:
Yes its the same adapter v2/3 having same chipset and a green LED which blinks initially and after driver installation everything works fine except LED blinking

wegi1, 2021-04-11:
So resolve is in #342
Uncomment LED blinking code.

gr33nm0nk2802, 2021-04-11:
I don't see similar code in this version could you mention the version.

lwfinger, 2021-04-11:
My 8188eu device does not have a LED, thus you are on your own to get it working! That is the beauty of open source.

d4rk-c1ph3r, 2021-04-13:
@lwfinger can you help me with some resources to understand and write my own code. I have never written code about drivers

lwfinger, 2021-04-13:
That is well above my pay grade. If you checkout branch v4.1.8_9499, your LED should work.

closed 2021-04-13 by lwfinger