lwfinger/rtl8188eu #366: Can't connect throught wifi in HiveOS (802.11n dongle)

delucca, 2021-03-19:
Hi everyone.

I'm trying to connect my HiveOS rig using the 802.11n dongle ([this guy](https://produto.mercadolivre.com.br/MLB-1631534578-antena-wi-fi-adaptador-wireless-1200mbs-usb-pc-notebook-_JM?quantity=1)) but without success. I've read almost every thread in the wi-fi setup (also, usually the 802.11n seems to work) but I was not able to make it work.

The problem is the following:

In HiveOS we can setup the wifi with the following command:
```
wifi
```
In my rig's terminal, I got the following error:
```
=== WiFi setup ===
Failed to bring up wlan0: SIOCSIFFLAGS: Operation not permitted

wlan0    unassociated ESSID="" Nickname:"<WIFI@REALTEK>"
         Mode: Auto  Frequency=2.412GHz  Access Point: Not-Associated
         Sensitivity:0/0
         Retry:off  RTS the:off  Fragment  the:off
         Encryption key:off
         Power Management:off
         Link Quality=0/100  Signal level=0 dBm  Noise level=0 dBm
         Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
         Tx excessive retries:0  Invalid misc:0  Missed beacon:0
● 3: wlan0
        Link File: /lib/systemd/network/99-default.link
     Network File: /etc/systemd/network/30-wireless.network
             Type: wlan
            State: off (configuring)
             Path: pci-0000:00:14.0-usb-0:9:1.0
           Driver: r8188eu
           Vendor: Realtek Semiconductor Corp.
            Model: 802.11n
       HW Address: 00:e0:4c:81:88:02 (REALTEK SEMICONDUCTOR CORP.)
...
```

The `wlan0` appears in my `iwconfig` but not in `ifconfig`

I've already tried to reinstall the `r8188eu` driver, but with no success.

The same error happens if I try to run:
```
ifconfig wlan0 up
```
This is the output of that command:
```
SIOCSIFFLAGS: Operation not permitted
```

I've tried to do a complete reinstall following this repository README steps. The installation went smoothly, but the issue persists.

Any ideas how to fix this?

lwfinger, 2021-03-19:
You are using the wrong branch. The master branch is built for the wireless extensions. To use the cfg80211/nl80211 commands, you need to use the v5.2.2.4 branch. From the directory with the source code, use the command

git checkout origin/v5.2.2.4 -b v5.2.2.4

The make and install.

delucca, 2021-03-19:
> You are using the wrong branch. The master branch is built for the wireless extensions. To use the cfg80211/nl80211 commands, you need to use the v5.2.2.4 branch. From the directory with the source code, use the command
> 
> git checkout origin/v5.2.2.4 -b v5.2.2.4
> 
> The make and install.

Thanks for your quick response!

So, I've done the steps you suggested (moving to the given branch) and rebooted my rig afterwards, and now the USB dongle is not recognized anymore.

It appears as `Unknown` in my `lsusb` command, but It does not appear in neither `iwconfig` nor `ifconfig`. Also, the previous `wifi` command (specific for HiveOS) can't recognize it either.

I've already tried multiple USB ports (both 3.0 and 2.0), and I've also tried to reboot it without the Ethernet cable (some folks say it can "deactivate" the wifi interface, don't know if that is true)

When I installed the v5.2.2.4, I didn't remove the previous installation. Do I need to do so?

lwfinger, 2021-03-20:
No, installing this one overwrites the other. One possibility is that the USB ID for your device is in the tables for the master branch, but not in v5.2.2.4. To check that, please post the output of lsusb.

delucca, 2021-03-20:
> No, installing this one overwrites the other. One possibility is that the USB ID for your device is in the tables for the master branch, but not in v5.2.2.4. To check that, please post the output of lsusb.

Sure, here it is:

```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0bda:f179 Realtek Semiconductor Corp. 
Bus 001 Device 003: ID abcd:1234 Unknown 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

The `Unknown` device is the one. The ID is: `abcd:1234`

lwfinger, 2021-03-20:
No, that USB ID is a bogus one. That piece of hardware is probably broken.

Code 0bda:f179 is a Realtek RTL8188FU. The driver was missing a critical piece for that device. Do a 'git pull', 'make', and 'sudo make install'. It should work better now.

delucca, 2021-03-20:
> No, that USB ID is a bogus one. That piece of hardware is probably broken.
> 
> Code 0bda:f179 is a Realtek RTL8188FU. The driver was missing a critical piece for that device. Do a 'git pull', 'make', and 'sudo make install'. It should work better now.

Ok!

So, I've done as you suggested (in the v5.2.2.4 branch) but it is still not being recognized (neither by iwconfig, nor ifconfig or event the wifi command) :/

delucca, 2021-03-20:
> Ok!
>
> So, I've done as you suggested (in the v5.2.2.4 branch) but it is still not being recognized (neither by iwconfig, nor ifconfig or event the wifi command) :/

Just to remind you. In the master branch it is recognized by iwconfig and the wifi command, but that weird error happens.

delucca, 2021-03-24:
@lwfinger sorry for bumping this. Can you give me a hand? I really don't know what to do :/

lwfinger, 2021-03-24:
I do not either, and I do not have a device like yours.

I have already spent entirely too much money equipping myself with lots of USB sticks, and I'm not doing that any more.

navarobot, 2021-07-12:
I have [this](https://aliexpress.com/item/32962776973.html) adapter and it contains rtl8188ftv chip (i paid around 1$). 
Information:
- ArchLinux 5.12.14-zen1-1-zen which doesn't support rtl8188fu by default
- Your [driver](https://github.com/lwfinger/rtl8188eu), branch v5.2.2.4 installed by dkms

Journalctl:
```
kernel: Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_B_CUT_1T1R_RomVer(0)
kernel: EEPROM ID(0x3031) is invalid!!
kernel: EEPROM ID = 0x3031
NetworkManager[403]: <info>  [1626111456.4015] wifi-wext: (wlan1): driver supports SSID scans (scan_capa 0x3F).
NetworkManager[403]: <info>  [1626111456.4016] wifi-wext: (wlan1): using WEXT for Wi-Fi device control
NetworkManager[403]: <info>  [1626111456.4069] manager: (wlan1): new 802.11 Wi-Fi device (/org/freedesktop/NetworkManager/Devices/23)
NetworkManager[403]: <warn>  [1626111456.6654] platform-linux: do-change-link[19]: failure changing link: failure 1 (Operation not permitted)
NetworkManager[403]: <warn>  [1626111456.7658] device (wlan1): set-hw-addr: new MAC address 86:6B:89:F8:20:BF not successfully set (scanning)
NetworkManager[403]: <warn>  [1626111456.9224] platform-linux: do-change-link[19]: failure changing link: failure 1 (Operation not permitted)
wpa_supplicant[425]: nl80211: Could not configure driver mode
wpa_supplicant[425]: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
wpa_supplicant[425]: rfkill: Cannot get wiphy information
wpa_supplicant[425]: Could not set interface wlan1 flags (UP): Operation not permitted
wpa_supplicant[425]: WEXT: Could not set interface 'wlan1' UP
wpa_supplicant[425]: wlan1: Failed to initialize driver interface
NetworkManager[403]: <error> [1626111457.1002] device (wlan1): Couldn't initialize supplicant interface: GDBus.Error:fi.w1.wpa_supplicant1.UnknownError: wpa_supplicant couldn't grab this interface.
```
Journatlctl with debug level 8 - [journalctl_8188eu.txt](https://github.com/lwfinger/rtl8188eu/files/6803449/journalctl_8188eu.txt)

[lsusb.txt](https://github.com/lwfinger/rtl8188eu/files/6803164/lsusb.txt)

lwfinger, 2021-07-12:
I added this USB ID based on what someone told me, but I could not test. Try the driver at https://github.com/kelebek333/rtl8188fu. That driver has worked for others.

navarobot, 2021-07-28:
Sorry for misinformation, in my previous post i've used master branch.
With v5.2.2.4 branch i get errors about chip_version, as i understand your and [this](https://github.com/aircrack-ng/rtl8188eus) driver doesn't have HAL for 8188fu.
[journalctl_v5.2.2.4.txt](https://github.com/lwfinger/rtl8188eu/files/6895359/journalctl_v5.2.2.4.txt)

Anyway, i installed [this](https://github.com/ulli-kroll/rtl8188fu) driver and it didn't work too, when i tried to set wlan1 link state to UP, i got `INFO: task ip:5056 blocked for more than 122 seconds.` with interesting call trace:
```
1. __schedule
2. schedule_preempt_disabled
3. __mutex_lock.constprop.0
4. wiphy_apply_custom_regulatory [cfg80211]
5. rtw_regd_init [rtl8188fu]
6. rtw_cfg80211_init_wiphy [rtl8188fu]
7. _netdev_open [rtl8188fu]
8. netdev_open [rtl8188fu]
9. __dev_open
10. ...
```
Then i found [this](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/staging/rtl8723bs?h=v5.13.6&id=81f153faacd04c049e5482d6ff33daddc30ed44e) patch and with it but a little bit modified, my adapter started working.
[iw_phy.txt](https://github.com/lwfinger/rtl8188eu/files/6895514/iw_phy.txt)
[journalctl_ulli-kroll_rtl8188fu.txt](https://github.com/lwfinger/rtl8188eu/files/6895545/journalctl_ulli-kroll_rtl8188fu.txt)