lwfinger/rtl8188eu #371: device uses r8188eu despite blacklist

Rick7221, 2021-05-16:
I am using the TL-WN722N v2 wifi dongle on a Raspberry Pi Zero and it is still loading the driver despite it being blacklisted in modprobe. 

```
pi@blaubeere:~ $ cat /etc/modprobe.d/blacklist.conf
blacklist r8188eu

pi@blaubeere:~ $ cat /etc/modprobe.d/8188eu.conf
options 8188eu rtw_power_mgnt=0 rtw_enusbss=0 rtw_hwpwrp_detect=0 rtw_low_power=0 rtw_ips_mode=1
```

```
pi@blaubeere:~ $ lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Vendor Specific Class, Driver=r8188eu, 480M
```

lwfinger, 2021-05-16:
The presence of r8188eu in that listing is not proof of r8188eu being loaded. What does 'lsmod | grep 8188' show?

Rick7221, 2021-05-16:
This is the output:
```
pi@blaubeere:~ $ lsmod | grep 8188
8188eu                761856  0
```

I used this driver before and it worked, but had to setup the pi fresh and now it doesn't work anymore. Only connects wlan0 and not through wlan1 (the dongle). Previously it was enough to disable internal wifi and then it would connect through the dongle.

lwfinger, 2021-05-16:
OK, r8188 is blacklisted. When you set up the pi, did anything else change? The master branch does not work with newer versions of wireless management tools. Try branch v5.2.2.4. It uses cfg80211, rather than the old WEXT methods.

Rick7221, 2021-05-16:
It's working now, many thanks! Switching to v5.2.2.4 worked.
Now i hope it does not drop wifi randomly after a few hours like earlier! :)

closed 2021-05-16 by Rick7221