lwfinger/rtl8188eu #373: Wifi USB Adapter can scan networks, can connect, but can't browse Internet

achede22, 2021-05-22:
https://github.com/feren-OS/Bug-Reporting-Center/issues/226

Package Name or Section of Feren OS where this bug resides
Where does this bug happen? (e.g.: feren-themer-cinnamon) ___
Wifi Network
Wifi Network works on Windows 10, same PC
Ethernet works in FerenOS with no problems

Describe the bug
A clear and concise description of what the bug is.
Wifi USB Adapter can scan networks, can connect to the router SSID, but can't browse Internet (DNS error)

Trying to fix it I mannualy setup DNS address and that not worked

Steps to Reproduce

Connect a Wifi USB Adapter
Connect to a network
Try to browse the web (Error) try with any browser, any network, same results
Expected behavior
FerenOS can browse internet with an USB Wifi Adapter antenna.

Screenshots
Firefox message:
We can’t connect to the server at www.google.com.

If that address is correct, here are three other things you can try:

Try again later.
Check your network connection.
If you are connected but behind a firewall, check that Firefox has permission to access the Web.
Additional context

Already tried

Manually setup DNS address
sudo dnf copr enable dcaratti/wpa_supplicant ; sudo dnf update ; reboot

```
inxi -Fxz

Network: Device-1: Broadcom and subsidiaries NetLink BCM5784M Gigabit Ethernet PCIe vendor: Foxconn driver: tg3
v: kernel port: e000 bus ID: 02:00.0
IF: enp2s0 state: up speed: 100 Mbps duplex: full mac:

      Device-2: Realtek RTL8188EUS 802.11n Wireless Network Adapter type: USB driver: r8188eu bus ID: 1-1:2  
      IF: wlx00e02d0d95f0 state: down mac: <filter> 
```
      
```
lsusb
Bus 001 Device 002: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
```

```
git clone https://github.com/lwfinger/rtl8188eu
cd rtl8188eu
make all
make install
systemctl restart NetworkManager
```
#Same DNS issue

lwfinger, 2021-05-23:
You are using the master branch, which is an older version of the software. I preserve it for those people that need it, but I recommend that you use the v5.2.2.4 branch. Replace your command sequence above with the following:

cd ~
rm -rf rtl8188eu/
git clone https://github.com/lwfinger/rtl8188eu -b v5.2.2.4
cd rtl8188eu/
make
sudo make install
sudo modprobe -rv 8188eu
sudo modprobe -v 8188eu
sudo systemctl restart NetworkManager

Your sequence of commands indicate that you are building the driver as root, which is bad practice as a bug in a Makefile can destroy a system. That happened a few years ago with the kernel build sequence. As a normal user, there was a non-fatal message logged that was quickly lost in the normal output of a kernel build. If you ran the build as root, it deleted /dev/null, and made the system unbootable. The Makefile here does not contain any code that would harm your system, but using root privilege only when it is needed is still good practice.

If the new version of the driver still fails, please run the following command and post the output:

nslookup google.com

achede22, 2021-05-25:
#Fixed 
#Not a bug 

Thanks for the explanation, from now on I will check the branch for newer versions, and also thanks for show me the consequences of the root user privileges, I did all the recommended step and "for me" the issue is the same, DNS failure. (The driver or kernel module is OK)

I can Scan Wifi networks, I can connect to it, but I can't browse internet

Chrome error:

This site can’t be reached www.primevideo.com ’s server IP address could not be found.
Try:

Checking the proxy, firewall, and DNS configuration
DNS_PROBE_FINISHED_BAD_CONFIG

FerenOS: Ticket https://github.com/feren-OS/Bug-Reporting-Center/issues/226

closed 2021-05-25 by achede22