lwfinger/rtl8188eu #375: Error: Connection activation failed: (7) Secrets were required, but not provided

rlelchev, 2021-08-03:
I tried to install this driver and I got it successfully installed, but NetworkManager gives me this error message. Any help?

lwfinger, 2021-08-03:
Are you sure you gave the secret correctly?

Which branch are you using?  Branch v4.1.8_9499 is currently being added to the kernel, and it works correctly.

rlelchev, 2021-12-18:
I don't know the fix for this issue, because when I reinstalled my OS everything got fixed. It might have been just me being too much of a noob on Linux to type out the secrets. I was a Linux beginner back then and I didn't know what I was doing. Arch Linux and other minimal Linux distros do not come preinstalled with NetworkManager and everything it requires and that makes me need to install drivers and other things to make everything work. Thank you for trying to help, even though I was such a noob back then.

closed 2021-12-18 by rlelchev

bhardwajvanshika-eaton, 2025-03-25:
i am facing the same error on stm32mp157f after migration, from dunfell to kirkstone.