lwfinger/rtl8188eu #378: Wpa_supplicant not working

rlelchev, 2021-09-24:
I am trying to connect to the network with wpa_supplicant, but it's not working. Can I connect using NetworkManager?

lwfinger, 2021-09-24:
That is how I connect. I run a KDE desktop and use the associated applet to control connections. Note that NM uses wpa_supplicant. It just connects through D-Bus rather than a supplicant control file.

rlelchev, 2021-09-25:
How can i connect using D-Bus from tty?

rlelchev, 2021-09-25:
Wpa_supplicant gives me "operation not permitted" errors and "rfkill: cannot get wirpy information". How can I fix that?

lwfinger, 2021-09-25:
Without knowing what operation is not permitted, it would be hard to comment.

matiasdlf, 2021-09-30:
I'm having the same problem with wpa_supplicant with a Bus 002 Device 002: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
This is the journalctl -xe log:
`sep 30 11:45:15 fedora NetworkManager[1961]: <info>  [1633013115.0913] device (wlp0s19f2u1): state change: need-auth -> prepare (reason 'none', sys-iface-state: 'managed')
sep 30 11:45:15 fedora NetworkManager[1961]: <info>  [1633013115.0931] device (wlp0s19f2u1): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
sep 30 11:45:15 fedora NetworkManager[1961]: <info>  [1633013115.0939] device (wlp0s19f2u1): Activation: (wifi) connection 'NaranjaMecanica_2,4' has security, and secrets exist.  No new secr>
sep 30 11:45:15 fedora NetworkManager[1961]: <info>  [1633013115.0940] Config: added 'ssid' value 'NaranjaMecanica_2,4'
sep 30 11:45:15 fedora NetworkManager[1961]: <info>  [1633013115.0940] Config: added 'scan_ssid' value '1'
sep 30 11:45:15 fedora NetworkManager[1961]: <info>  [1633013115.0940] Config: added 'bgscan' value 'simple:30:-70:86400'
sep 30 11:45:15 fedora NetworkManager[1961]: <info>  [1633013115.0940] Config: added 'key_mgmt' value 'WPA-PSK WPA-PSK-SHA256'
sep 30 11:45:15 fedora NetworkManager[1961]: <info>  [1633013115.0941] Config: added 'auth_alg' value 'OPEN'
sep 30 11:45:15 fedora NetworkManager[1961]: <info>  [1633013115.0941] Config: added 'psk' value '<hidden>'
sep 30 11:45:15 fedora wpa_supplicant[2089]: wlp0s19f2u1: Trying to associate with 04:a2:22:73:58:68 (SSID='NaranjaMecanica_2,4' freq=2437 MHz)
sep 30 11:45:15 fedora wpa_supplicant[2089]: Failed to add supported operating classes IE
sep 30 11:45:15 fedora wpa_supplicant[2089]: wlp0s19f2u1: Association request to the driver failed
sep 30 11:45:15 fedora NetworkManager[1961]: <info>  [1633013115.5320] device (wlp0s19f2u1): supplicant interface state: disconnected -> associating
sep 30 11:45:17 fedora wpa_supplicant[2089]: wlp0s19f2u1: CTRL-EVENT-DISCONNECTED bssid=04:a2:22:73:58:68 reason=0
sep 30 11:45:17 fedora kernel: R8188EU: INFO indicate disassoc
sep 30 11:45:17 fedora NetworkManager[1961]: <info>  [1633013117.1707] device (wlp0s19f2u1): supplicant interface state: associating -> disconnected
sep 30 11:45:17 fedora NetworkManager[1961]: <info>  [1633013117.2708] device (wlp0s19f2u1): supplicant interface state: disconnected -> scanning
`
I Have linux fedora 5.13.19-200.fc34.x86_64,
Tried even with the KDE Applet to manage connections but it does not working.
The strange thing it's sometimes connects but in a few minutes a got "limited or null connection"

Any idea?

rlelchev, 2021-12-18:
The reason I was getting those errors was because I wasn't running wpa_supplicant with root and I just ignore the "cannot get wirpy information" errors, because the network is working even if I get these errors. I consider wpa_supplicant more stable than NetworkManager, but starting it at startup is a pain. I think wpa_supplicant is really good for livecd Linux distros, because it comes with the livecd. Thank you for the help and sorry for not replying with the fix for a whole month.

closed 2021-12-18 by rlelchev

lwfinger, 2021-12-18:
Well, I have used NetworkManager exclusively for more than 5 years  without a problem.