lwfinger/rtl8188eu #380: Operation not permitted

ibratabian17, 2021-09-25:
so, i use wifi-menu to connect wifi, bu it show this
`root@ubuntu-ibra:/home/ibra# wifi-menu wlx00e04c818802
Scanning for networks... Could not set interface wlx00e04c818802 flags (UP): Operation not permitted
WEXT: Could not set interface 'wlx00e04c818802' UP
wlx00e04c818802: Failed to initialize driver interface
failed
No networks found
`
and i use `sudo ifconfig wlx00e04c818802 up` it show this

`ibra@ubuntu-ibra:~$ sudo ifconfig wlx00e04c818802 up
SIOCSIFFLAGS: Operation not permitted
`

lwfinger, 2021-09-25:
You configured your kernel without selecting WEXT. The driver will not work without it. A number of us are working on a full CFG80211 version that will not need WEXT, but that is still a work in progress.

ibratabian17, 2021-09-25:
How to selecting WEXT? I'm currently updating the kernel

closed 2021-09-25 by ibratabian17