lwfinger/rtl8188eu #383: The network manager says: "Device is not ready"

Kurdes38, 2021-11-04:
I am using a Wi-Fi USB dongle TL-WN725N(EU) Ver:3.0.
My Linux system is:
`  $ hostnamectl
   ...
   Operating System: Simply Linux 9.1 (Destiny)
   Kernel: Linux 5.10.74-un-def-alt1
   Architecture: x86-64
   ...
`
8188eu module:
` $ sudo modinfo 8188eu
  filename:       /lib/modules/5.10.74-un-def-alt1/kernel/drivers/net/wireless/**8188eu.ko**
  version:        v4.1.4_6773.20130222
  author:         Realtek Semiconductor Corp.
  description:    Realtek Wireless Lan Driver
  ...
`
It works well on Windows 10 and Linux separately, but when I reboot my Windows machine to Linux, a problem arises: the network manager says: "The device is not ready." 
Only reconnecting the dongle helps.
The firmware (rtl8188eufw.bin) is indeed located in /lib/firmware/rtlwifi/:
`$ ls -l
  ...
  -rw-r--r-- 1 root root  11216 мая 25 03:14 rtl8188efw.bin
  **-rw-r--r-- 1 root root  13904 ноя  4 00:33 rtl8188eufw.bin**
  -rw-r--r-- 1 root root  16192 мая 25 03:14 rtl8192cfw.bin
  -rw-r--r-- 1 root root  16332 мая 25 03:14 rtl8192cfwU_B.bin
  ...
`
Part of the syslog (journalctl) when the device is not ready:
  `ноя 04 18:27:42 DESKTOP-P6TDM7G NetworkManager[2870]: <warn>  [1636036062.9817] device (wlan1): re-acquiring supplicant interface (#3).
  ноя 04 18:27:42 DESKTOP-P6TDM7G wpa_supplicant[2989]: nl80211: Driver does not support authentication/association or connect commands
  ноя 04 18:27:42 DESKTOP-P6TDM7G wpa_supplicant[2989]: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
  ноя 04 18:27:42 DESKTOP-P6TDM7G wpa_supplicant[2989]: rfkill: Cannot get wiphy information
  ноя 04 18:27:43 DESKTOP-P6TDM7G **wpa_supplicant[2989]: Could not set interface wlan1 flags (UP): Operation not permitted**
  ноя 04 18:27:43 DESKTOP-P6TDM7G wpa_supplicant[2989]: WEXT: Could not set interface 'wlan1' UP
  ноя 04 18:27:43 DESKTOP-P6TDM7G wpa_supplicant[2989]: wlan1: Failed to initialize driver interface
  ноя 04 18:27:43 DESKTOP-P6TDM7G NetworkManager[2870]: <error> [1636036063.0377] sup-iface[0x55efe48f4340,wlan1]: **error adding interface: wpa_supplicant couldn't grab this interface.**
  ноя 04 18:27:43 DESKTOP-P6TDM7G NetworkManager[2870]: <info>  [1636036063.0378] device (wlan1): supplicant interface state: starting -> down
  ноя 04 18:27:52 DESKTOP-P6TDM7G NetworkManager[2870]: <warn>  [1636036072.9890] device (wlan1): re-acquiring supplicant interface (#4).
  ноя 04 18:27:52 DESKTOP-P6TDM7G wpa_supplicant[2989]: nl80211: Driver does not support authentication/association or connect commands
  ноя 04 18:27:52 DESKTOP-P6TDM7G wpa_supplicant[2989]: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
  ноя 04 18:27:52 DESKTOP-P6TDM7G wpa_supplicant[2989]: rfkill: Cannot get wiphy information
  ноя 04 18:27:53 DESKTOP-P6TDM7G **wpa_supplicant[2989]: Could not set interface wlan1 flags (UP): Operation not permitted**
  ноя 04 18:27:53 DESKTOP-P6TDM7G wpa_supplicant[2989]: WEXT: Could not set interface 'wlan1' UP
  ноя 04 18:27:53 DESKTOP-P6TDM7G wpa_supplicant[2989]: wlan1: Failed to initialize driver interface
  ноя 04 18:27:53 DESKTOP-P6TDM7G NetworkManager[2870]: <error> [1636036073.0452] sup-iface[0x55efe48f4420,wlan1]: **error adding interface: wpa_supplicant couldn't grab this interface.**
  ноя 04 18:27:53 DESKTOP-P6TDM7G NetworkManager[2870]: <info>  [1636036073.0453] device (wlan1): supplicant interface state: starting -> down
`
How to make it work after restarting machine from Windows 10 to Linux without reconnecting the USB dongle?

lwfinger, 2021-11-04:
Does your Windows instance use the quick boot feature? That should/must be disabled if you are going to boot into Linux.

See https://www.howtogeek.com/243901/the-pros-and-cons-of-windows-10s-fast-startup-mode/ for a description of what it does, and how to disable it.

closed 2021-11-04 by lwfinger

Kurdes38, 2021-11-07:
Yes, my Windows 10 OS used the Fast Boot feature. I have disabled quick boot feature in Windows and in BIOS / UEFI settings.
But that didn't solve the problem. When I reboot my machine from Windows OS to Linux OS, the network manager says, "The device is not ready."
In my case, only two things help (when the network manager correctly detects Wi-Fi networks):
- shutting down the Windows / Linux OS, and then loading the Linux instance;
- reconnecting the USB dongle after booting Linux.

lwfinger, 2021-11-08:
Obviously, Windows is not power-cycling the USB device. It probably has Wake-on-Wlan enabled. If you want to avoid having to plug/unplug the device, then Shut Down Windows. That should work. I think that is what you meant in bullet point 1.

Kurdes38, 2021-11-08:
OK, I thought about it. As I understand it, "Wake-on-Wlan" may be enabled by setting "wake up" parameter for corresponding device. On my Linux instance, Wi-Fi USB dongle has a "wakeup" parameter value:
```
cat /sys/bus/usb/devices/1-7/power/wakeup
disabled
```
I tried to change this parameter manually to "enabled", but it didn't take some effect.
Perhaps setting this parameter at boot time will help?

lwfinger, 2021-11-08:
I think you want to turn that "feature" off in Windows, but I have no idea if that is even possible.

Kurdes38, 2021-11-09:
OK. It might not be possible to disable this behavior under Windows, but is it possible to start a Linux instance (after restarting Windows) by enabling the wake up USB key option without reconnecting it?
Because restarting the Linux instance to Windows does not require the key to be reconnected.

lwfinger, 2021-11-09:
Are you talking about a Linux VM? If so, the USB stuff will still be under control of Windows - only the passthrough mechanism gets to Linux. Whatever the cause, Windows is not powering down the dongle on shutdown.

Kurdes38, 2021-11-09:
No, it isn't VM. I have dual-boot systems.

lwfinger, 2021-11-09:
Whatever the problem, it is Windows that is leaving the USB port in a bad state. The only thing I have found is in https://helpdeskgeek.com/how-to/prevent-windows-from-powering-off-usb-device/. That article is telling you how to keep nubs from going to sleep, but make sure that the "Allow the device to wake the computer" box is NOT checked. If non of your hubs have this enabled, I have no further help.

RadoslavL, 2022-03-31:
Going to the BIOS/UEFI and turning off wake-on-wlan there for USB devices should work. Sorry for replying after five months, but I wanted to share my opinion.