lwfinger/rtl8188eu #385: EEPROM ID(0xffff) is invalid!!

kelppo, 2021-11-14:
I just updated the kernel to 5.15.2-arch1-1 and noticed that the module r8188eu started matching my USB wifi adapter (0bda:f179) that previously had to be driven by an out of tree module (https://github.com/ulli-kroll/rtl8188fu) but after uninstalling it I find that r8188eu doesn't work, I can see wlan0 in `ip a` but iwd couldn't find and configure the interface. Blacklisting r8188eu and reinstalling the out of tree module works though.

```
$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: enp7s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 70:85:c2:c6:3b:6c brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.186/24 metric 1024 brd 192.168.0.255 scope global dynamic enp7s0
       valid_lft 85832sec preferred_lft 85832sec
    inet6 fd01::7285:c2ff:fec6:3b6c/64 scope global dynamic mngtmpaddr noprefixroute
       valid_lft 251sec preferred_lft 251sec
    inet6 fe80::7285:c2ff:fec6:3b6c/64 scope link
       valid_lft forever preferred_lft forever
5: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 00:e0:09:2f:7c:61 brd ff:ff:ff:ff:ff:f
```
```
$ iwctl device list
                                    Devices
--------------------------------------------------------------------------------
  Name                Address             Powered   Adapter   Mode
--------------------------------------------------------------------------------
```
```
$ iwctl station list
                            Devices in Station Mode
--------------------------------------------------------------------------------
  Name                State          Scanning
--------------------------------------------------------------------------------
No devices in Station mode available.
```
Here is my dmesg (also with debug=9):
[dmesg.txt](https://github.com/lwfinger/rtl8188eu/files/7533167/dmesg.txt)

rlelchev, 2021-11-17:
rtl8188eu use the WEXT driver and iwd and iw don't support it. Use NetworkManager or wpa_supplicant with that adapter.
Example: sudo pacman -S networkmanager && sudo systemctl enable NetworkManager && sudo systemctl start NetworkManager && nmtui (nmcli)

rlelchev, 2021-11-17:
@lwfinger Please mark this as awaiting verification

kelppo, 2021-11-17:
>rtl8188eu use the WEXT driver

Sorry for the confusion but I'm having trouble with the new r8188eu found in driver/staging on the 5.15 release. I tried using wpa_supplicant but it still won't work.
```
$ sudo wpa_supplicant -c /etc/wpa_supplicant/wpa_supplicant-wlan0.conf -i wlan0 -D wext
Successfully initialized wpa_supplicant
rfkill: Cannot get wiphy information
Could not set interface wlan0 flags (UP): Operation not permitted
WEXT: Could not set interface 'wlan0' UP
wlan0: Failed to initialize driver interface
```
```
$ sudo wpa_supplicant -c /etc/wpa_supplicant/wpa_supplicant-wlan0.conf -i wlan0
Successfully initialized wpa_supplicant
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
wlan0: Failed to initialize driver interface
```
Feel free to close this issue if this is not the correct place to report this.

lwfinger, 2021-11-17:
I verify that the driver does not support nl80211. You need to use a D-bus implementation such as Network Manager.

kelppo, 2021-11-18:
NetworkManager does not work either.
```
Starting Network Manager...
<info>  [1637204784.7068] NetworkManager (version 1.32.12-1) is starting... (for the first time)
<info>  [1637204784.7068] Read config: /etc/NetworkManager/NetworkManager.conf (lib: 20-connectivity.conf)
Started Network Manager.
<info>  [1637204784.7084] bus-manager: acquired D-Bus service "org.freedesktop.NetworkManager"
<info>  [1637204784.7091] wifi-wext: (wlan0): driver supports SSID scans (scan_capa 0x3F).
<info>  [1637204784.7091] wifi-wext: (wlan0): using WEXT for Wi-Fi device control
<info>  [1637204784.7127] manager[0x564320c9a060]: monitoring kernel firmware directory '/lib/firmware'.
<info>  [1637204784.7975] hostname: hostname: using hostnamed
<info>  [1637204784.7975] hostname: hostname changed from (none) to "mainframe"
<info>  [1637204784.7981] dns-mgr[0x564320c5b250]: init: dns=default,systemd-resolved rc-manager=symlink
<info>  [1637204784.7983] manager[0x564320c9a060]: rfkill: Wi-Fi hardware radio set enabled
<info>  [1637204784.7984] manager[0x564320c9a060]: rfkill: WWAN hardware radio set enabled
<info>  [1637204784.7999] Loaded device plugin: NMAtmManager (/usr/lib/NetworkManager/1.32.12-1/libnm-device-plugin-adsl.so)
<info>  [1637204784.8011] Loaded device plugin: NMBluezManager (/usr/lib/NetworkManager/1.32.12-1/libnm-device-plugin-bluetooth.so)
<info>  [1637204784.8013] Loaded device plugin: NMOvsFactory (/usr/lib/NetworkManager/1.32.12-1/libnm-device-plugin-ovs.so)
<info>  [1637204784.8046] Loaded device plugin: NMTeamFactory (/usr/lib/NetworkManager/1.32.12-1/libnm-device-plugin-team.so)
<info>  [1637204784.8048] Loaded device plugin: NMWifiFactory (/usr/lib/NetworkManager/1.32.12-1/libnm-device-plugin-wifi.so)
<info>  [1637204784.8049] Loaded device plugin: NMWwanFactory (/usr/lib/NetworkManager/1.32.12-1/libnm-device-plugin-wwan.so)
<info>  [1637204784.8051] manager: rfkill: Wi-Fi enabled by radio killswitch; enabled by state file
<info>  [1637204784.8052] manager: rfkill: WWAN enabled by radio killswitch; enabled by state file
<info>  [1637204784.8052] manager: Networking is enabled by state file
<info>  [1637204784.8052] dhcp-init: Using DHCP client 'internal'
<info>  [1637204784.8053] settings: Loaded settings plugin: keyfile (internal)
<info>  [1637204784.8056] device (lo): carrier: link connected
<info>  [1637204784.8057] manager: (lo): new Generic device (/org/freedesktop/NetworkManager/Devices/1)
<info>  [1637204784.8061] device (enp7s0): carrier: link connected
<info>  [1637204784.8064] manager: (enp7s0): new Ethernet device (/org/freedesktop/NetworkManager/Devices/2)
<info>  [1637204784.8076] settings: (enp7s0): created default wired connection 'Wired connection 1'
<info>  [1637204784.8087] device (enp7s0): state change: unmanaged -> unavailable (reason 'connection-assumed', sys-iface-state: 'external')
<info>  [1637204784.8091] device (enp7s0): state change: unavailable -> disconnected (reason 'connection-assumed', sys-iface-state: 'external')
<info>  [1637204784.8096] device (enp7s0): Activation: starting connection 'enp7s0' (7b66f408-00bd-4c6b-bb3a-38cba757b2bf)
<info>  [1637204784.8101] manager: (wlan0): new 802.11 Wi-Fi device (/org/freedesktop/NetworkManager/Devices/3)
<info>  [1637204784.8105] device (wlan0): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
<warn>  [1637204788.5338] platform-linux: do-change-link[3]: failure changing link: failure 1 (Operation not permitted)
<warn>  [1637204788.6341] device (wlan0): set-hw-addr: new MAC address 2A:46:DD:67:CD:10 not successfully set (scanning)
<warn>  [1637204792.3573] platform-linux: do-change-link[3]: failure changing link: failure 1 (Operation not permitted)
<info>  [1637204792.3603] ovsdb: Could not connect: No such file or directory
<info>  [1637204792.3604] ovsdb: disconnected from ovsdb
<info>  [1637204792.3616] device (enp7s0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'external')
<info>  [1637204792.3620] device (enp7s0): state change: prepare -> config (reason 'none', sys-iface-state: 'external')
<info>  [1637204792.3625] device (enp7s0): state change: config -> ip-config (reason 'none', sys-iface-state: 'external')
<info>  [1637204792.3628] device (enp7s0): state change: ip-config -> ip-check (reason 'none', sys-iface-state: 'external')
<info>  [1637204792.3657] device (enp7s0): state change: ip-check -> secondaries (reason 'none', sys-iface-state: 'external')
<info>  [1637204792.3659] device (enp7s0): state change: secondaries -> activated (reason 'none', sys-iface-state: 'external')
<info>  [1637204792.3663] manager: NetworkManager state is now CONNECTED_LOCAL
<info>  [1637204792.3667] device (enp7s0): Activation: successful, device activated.
<error> [1637204796.1361] device (wlan0): Couldn't initialize supplicant interface: GDBus.Error:fi.w1.wpa_supplicant1.UnknownError: wpa_supplicant couldn't grab this interface.
<info>  [1637204796.1362] manager: startup complete
<warn>  [1637204812.3835] device (wlan0): re-acquiring supplicant interface (#1).
<error> [1637204816.1072] device (wlan0): Couldn't initialize supplicant interface: GDBus.Error:fi.w1.wpa_supplicant1.UnknownError: wpa_supplicant couldn't grab this interface.
<warn>  [1637204826.1969] device (wlan0): re-acquiring supplicant interface (#2).
[... cut ...]
rfkill: Cannot get wiphy information
Could not set interface wlan0 flags (UP): Operation not permitted
WEXT: Could not set interface 'wlan0' UP
wlan0: Failed to initialize driver interface
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
rfkill: Cannot get wiphy information
Could not set interface wlan0 flags (UP): Operation not permitted
WEXT: Could not set interface 'wlan0' UP
wlan0: Failed to initialize driver interface
```

lwfinger, 2021-11-18:
Here is what I get:
[ 8505.889136] usb 2-6: new high-speed USB device number 6 using xhci_hcd
[ 8506.037701] usb 2-6: New USB device found, idVendor=0bda, idProduct=8179, bcdDevice= 0.00
[ 8506.037708] usb 2-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8506.037710] usb 2-6: Product: 802.11n NIC
[ 8506.037712] usb 2-6: Manufacturer: Realtek
[ 8506.037713] usb 2-6: SerialNumber: 00E04C0001
[ 8506.057436] usb 2-6: USB disconnect, device number 6
[ 8506.365155] usb 2-6: new high-speed USB device number 7 using xhci_hcd
[ 8506.513710] usb 2-6: New USB device found, idVendor=0bda, idProduct=8179, bcdDevice= 0.00
[ 8506.513715] usb 2-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8506.513717] usb 2-6: Product: 802.11n NIC
[ 8506.513719] usb 2-6: Manufacturer: Realtek
[ 8506.513720] usb 2-6: SerialNumber: 00E04C0001
[ 8552.255883] r8188eu: module is from the staging directory, the quality is unknown, you have been warned.
[ 8552.377266] Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_A_CUT_1T1R_RomVer(0)
[ 8552.399358] EEPROM ID = 0x8129
[ 8552.401841] usbcore: registered new interface driver r8188eu
[ 8552.410085] r8188eu 2-6:1.0 wlp0s20u6: renamed from wlan0
[ 8552.456708] R8188EU: Firmware Version 11, SubVersion 1, Signature 0x88e1
[ 8552.837377] MAC Address = a0:f3:c1:15:41:cf
[ 8552.841648] R8188EU: INFO indicate disassoc
[ 8554.091959] R8188EU: INFO indicate disassoc
[ 8555.488163] R8188EU: INFO assoc success

I do not know what is wrong with your setup, but mine just works. This system is running openSUSE Tumbleweed with kernel 5.16.0-rc1, and NetworkManager 1.13.12 with a KDE desktop.

kelppo, 2021-11-18:
The out-of-tree module still works for me, that's why I thought that the newly merged module is broken for my device.

lwfinger, 2021-11-18:
If you think it is broken, then you might try bisecting it.

kelppo, 2021-11-18:
I tried building commit db29c33e180e630280d74c629473e4297b6af7ad (where you added 8188fu USB ID) with 373d98d270880554839136308b8a1861f80bd289 applied so that the module builds on a 5.15 kernel and it doesn't work, I also looked around and found others with similar device apparently found this driver incompatible too (https://github.com/lwfinger/rtl8188eu/issues/366#issuecomment-888511731).

kelppo, 2021-11-18:
Should I send a patch to the staging mailing list to remove the 8188fu USB ID so I and other owners of this device don't have to blacklist the module?

lwfinger, 2021-11-18:
The 8188fu device does not work with this driver. You should send a patch. Have you found a driver that works? I never did.

kelppo, 2021-11-18:
>The 8188fu device does not work with this driver. You should send a patch.
Ok.
>Have you found a driver that works? I never did.
This one does: https://github.com/ulli-kroll/rtl8188fu but it does need fixes to make it work with the latest kernel.

closed 2021-11-18 by kelppo

navarobot, 2021-11-20:
[r8188fu.txt](https://github.com/lwfinger/rtl8188eu/files/7574884/r8188fu.txt) patch for r8188eu driver in 5.15.2 kernel fixes firmware loading somehow, but scan results in wpa_cli are empty after scan was triggered (i have 0bda:f179).
And rtl8188fufw.bin is 21020 bytes long in comparison to rtl8188eufw.bin which 13904 bytes long (strangely).

navarobot, 2021-11-21:
What is more interesting is that 2017 [rtl8188eus](https://github.com/aircrack-ng/rtl8188eus/) driver from realtek with some patches for hacking has code to support rtl8188fu, but the hal folders for rtl8188fu are missing. Does realtek just hide hal for this chip? I think i know the answer: yes, because it is so cheap, it is not profitable to release and support driver for this chip.